raspberrypi/bookworm-feedback #174: vnc completly broken after todays update

sgalke, 2023-12-14:
vnc is no longer operational, neither real-vnc *nor* wayvnc.
bookworm 32-bit  *and*  64-bit, hardware  4b, 5b

raspi-config reports red-colored errors :
Failed to stop wayvnc.service: Unit wayvnc.service not loaded.
Failed to enable unit: Unit file wayvnc.service does not exist.

fresh install did not help. same desaster after update....

sgalke, 2023-12-14:
"todays update" was:

gir1.2-gtk-3.0/stable 1:3.24.38-2~deb12u1+rpt3+rpi1 arm64 [aktualisierbar von: 1:3.24.38-2~deb12u1+rpt1+rpi1]
gir1.2-handy-1/stable 1.8.1-1+rpt3 arm64 [aktualisierbar von: 1.8.1-1+rpt1]
gtk-update-icon-cache/stable 1:3.24.38-2~deb12u1+rpt3+rpi1 arm64 [aktualisierbar von: 1:3.24.38-2~deb12u1+rpt1+rpi1]
libgtk-3-0/stable 1:3.24.38-2~deb12u1+rpt3+rpi1 arm64 [aktualisierbar von: 1:3.24.38-2~deb12u1+rpt1+rpi1]
libgtk-3-common/stable,stable 1:3.24.38-2~deb12u1+rpt3+rpi1 all [aktualisierbar von: 1:3.24.38-2~deb12u1+rpt1+rpi1]
libhandy-1-0/stable 1.8.1-1+rpt3 arm64 [aktualisierbar von: 1.8.1-1+rpt1]
pixflat-icons/stable,stable 0.23 all [aktualisierbar von: 0.21]
raspi-config/stable,stable 20231214 all [aktualisierbar von: 20231108]
xserver-common/stable,stable 2:21.1.7-3+rpt1+deb12u3 all [aktualisierbar von: 2:21.1.7-3+rpt1+deb12u2]
xserver-xorg-core/stable 2:21.1.7-3+rpt1+deb12u3 arm64 [aktualisierbar von: 2:21.1.7-3+rpt1+deb12u2]

themicro1, 2023-12-15:
Same here. 
Solution for now:
Create file /etc/xdg/autostart/wayvnc.desktop

`[Desktop Entry]
Type=Application
Name=wayvnc
Comment=Start wayvnc
NoDisplay=true
Exec=/usr/bin/wayvnc --render-cursor --keyboard=de
OnlyShowIn=wayfire
`

Don't use raspi-config until this ist fixed ;)

sgalke, 2023-12-15:
@themicro1:
folder  `/home/pi/.config/wayvnc` is also missing after fresh installation. 
i copied it from a not-already-updated system. 
with your (partly) solution and the missing folder (and its contents) vnc is up again. 
thank you!

nithyaatrify, 2023-12-15:
Yep, VNC not operational.
Any updates? wayvnc.service does not exist is the error

XECDesign, 2023-12-15:
I think the issue is that we shipped https://github.com/RPi-Distro/raspi-config/pull/226 without having the pre-requisites for it in place.

@spl237 Is that something that's ready to go, or do we need to revert that PR merge?

nithyaatrify, 2023-12-15:
> Yep, VNC not operational. Any updates? wayvnc.service does not exist is the error

The legacy version seems to be working fine

sgalke, 2023-12-15:
> The legacy version seems to be working fine

that does not help for rpi models 5.

spl237, 2023-12-15:
> I think the issue is that we shipped [RPi-Distro/raspi-config#226](https://github.com/RPi-Distro/raspi-config/pull/226) without having the pre-requisites for it in place.
> 
> @spl237 Is that something that's ready to go, or do we need to revert that PR merge?

As far as I know, those changes aren't ready yet, so that change will need to be reverted for now. (I didn't realise you were going to release it... ;) )

apulache, 2023-12-15:
I can confirmed that the solution proposed here didn't work on Raspberry Pi 5, any instructions to remove the packages causing the issues or revert back?
Thanks in advance!

apulache, 2023-12-15:
Here is a temporal fix for Raspberry Pi 5 using Raspberry Pi OS, it worked for me.

- Re-install manually the previous version for raspi-config
- I've got it from the repos: https://archive.raspberrypi.org/debian/pool/main/r/raspi-config/raspi-config_20231108_all.deb
- Enable VNC using the GUI : Menu -> Preferences -> Raspberry Pi Configuration -> Interfaces
- Just in case reboot
- Then VNC should be working again

Be aware not to update raspi-config at least we know this version cause the problems
[raspi-config_20231214_all.deb](https://archive.raspberrypi.org/debian/pool/main/r/raspi-config/raspi-config_20231214_all.deb)	2023-12-14 14:40	29K

sgalke, 2023-12-15:
@nithyaatrify: 

tested with fresh install & recomended update:
vnc under legacy wayland does not work. the same missing files / directory. other (minor?) wayland problems: desktop appearance not costomizable, etc ... (no fun, unusable, i am tired now, please test yourself, sorry...)

vnc with wayland turned off under legacy (bullseye) works.

OwenTea11, 2023-12-15:
> Same here. Solution for now: Create file /etc/xdg/autostart/wayvnc.desktop
> 
> ```
> [Desktop Entry]
> Type=Application
> Name=wayvnc
> Comment=Start wayvnc
> NoDisplay=true
> Exec=/usr/bin/wayvnc --render-cursor 
> OnlyShowIn=wayfire
> ```
> 
> Don't use raspi-config until this ist fixed ;)

Apologies for the noob question but where should I create this file? And is it just a text file?

Thank you!

StefanSchot, 2023-12-15:
> * https://archive.raspberrypi.org/debian/pool/main/r/raspi-config/raspi-config_20231108_all.deb

Thanks this worked: (new pi5)

- Downloaded it with: `wget https://archive.raspberrypi.org/debian/pool/main/r/raspi-config/raspi-config_20231108_all.deb`
- Installed it with `sudo apt install ./raspi-config_20231108_all.deb`
- Restarted `sudo reboot now`
- Started vnc, set autologin for desktop  `sudo raspi-config`
- Started service `systemctl start vncserver-x11-serviced.servic`

hlipnick, 2023-12-16:
> Here is a temporal fix for Raspberry Pi 5 using Raspberry Pi OS, it worked for me.
>
> * Re-install manually the previous version for raspi-config
> * I've got it from the repos: [...]

This worked perfectly on a new Raspberry Pi 5.  Thank you!

Also just adding for anyone not familiar with installing .deb files, this worked for me:

1. Open the temp directory: `cd /tmp`
2. Download the .deb file for the old raspi-config: `curl -OJL "https://archive.raspberrypi.org/debian/pool/main/r/raspi-config/raspi-config_20231108_all.deb"`
3. Install the old version: `sudo apt install ./raspi-config_20231108_all.deb`

sgalke, 2023-12-16:
the latest upgrade  raspi_config_20231216 restores at least the status quo ante - vnc is operational again.
there are some minor error left - but no longer completly broken.
thanks!!
issue closed.

closed 2023-12-16 by sgalke

kventil, 2024-01-23:
Ran in the same issue with  raspi_config_20240108. The Solution provided by @themicro1 worked for me to get vnc up and running again.

> Same here. Solution for now: Create file /etc/xdg/autostart/wayvnc.desktop
> 
> ```
> [Desktop Entry]
> Type=Application
> Name=wayvnc
> Comment=Start wayvnc
> NoDisplay=true
> Exec=/usr/bin/wayvnc --render-cursor 
> OnlyShowIn=wayfire
> ```
> 
> Don't use raspi-config until this ist fixed ;)

lurch, 2024-01-24:
@kventil Have you also updated the `wayvnc` package? Since the last comment in this issue, it has now been updated to be managed by systemd, rather than XDG autostart files. 
So could you please try:
1. `sudo rm /etc/xdg/autostart/wayvnc.desktop`
2. `sudo reboot`
3. `dpkg -l wayvnc`
4. `systemctl status wayvnc`

Thanks.

jimmusson, 2024-01-24:
The legacy version works fine on pi5 (64 bit)

kventil, 2024-01-25:
> @kventil Have you also updated the `wayvnc` package? Since the last comment in this issue, it has now been updated to be managed by systemd, rather than XDG autostart files. So could you please try:
> 
>     1. `sudo rm /etc/xdg/autostart/wayvnc.desktop`
> 
>     2. `sudo reboot`
> 
>     3. `dpkg -l wayvnc`
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-================================================
ii  wayvnc         0.8.0-rc0-2  arm64        VNC server for wlroots-based Wayland compositors
```

>     4. `systemctl status wayvnc`

```
● wayvnc.service - VNC Server
     Loaded: loaded (/lib/systemd/system/wayvnc.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-01-25 09:39:09 CET; 2s ago
       Docs: man:wayvnc
   Main PID: 6410 (sh)
      Tasks: 6 (limit: 4444)
        CPU: 144ms
     CGroup: /system.slice/wayvnc.service
             ├─6410 /bin/sh /usr/sbin/wayvnc-run.sh
             └─6414 wayvnc --render-cursor --detached --config /etc/wayvnc/config --socket /tmp/wayvnc/wayvncctl.sock

Jan 25 09:39:08 diveramonitor systemd[1]: Starting wayvnc.service - VNC Server...
Jan 25 09:39:09 diveramonitor systemd[1]: Started wayvnc.service - VNC Server.
```

Looking okay to me. (Have now another issue with wayland but that's unreladted to vnc)

salasidis, 2024-04-20:
> @kventil Have you also updated the `wayvnc` package? Since the last comment in this issue, it has now been updated to be managed by systemd, rather than XDG autostart files. So could you please [...]

I needed to run

`systemctl` status wayvnc`

Then VNC connections were possible