raspberrypi/bookworm-feedback #175: New update segmentation fault.

steveiliop56, 2023-12-15:
Hello,

After the new update libc-bin fails. Digging into it I found out ldconfig fails with segmentation fault. The image is brand new raspberry pi os 64 bit.

XECDesign, 2023-12-15:
Seems okay on this end. Is there any more information that might help trigger the problem.

steveiliop56, 2023-12-15:
It's just the newest and greatest raspberry pi os 64bit. Just tried to update...

lurch, 2023-12-15:
What model of Raspberry Pi are you running on? Are you booting from SD card? Do you have any additional hardware connected to your Pi? Are you using one of the official PSUs?

steveiliop56, 2023-12-15:
So booting of a USB2 SD card reader with a 32gb SD card (yeah I know USB2 but works fine) attached I have a dsi screen, a camera and a pimoroni fanshim (screen not official, camera module v3) and no low voltage warnings of any kind. I was just trying to test bookworm out before switching to it.

steveiliop56, 2023-12-15:
Hmm worked the third try after I reinstalled the os. I think it may have been that stupid usb2 adaptor. Anyway closing...

closed 2023-12-15 by steveiliop56