raspberrypi/bookworm-feedback #176: App properties crashing.

steveiliop56, 2023-12-15:
Hello,

When I get to this screen:

[image]

And try to click the icon to change it the whole desktop just restarts.

lurch, 2023-12-18:
Can you provide more details please, e.g. exact reproduction steps starting from a freshly-flashed Bookworm image?

I just tested this myself by copying `/usr/share/applications/rp-bookshelf.desktop` to my home directory, and then in the File Manager I was able to right-click on the "Bookshelf" file in my home directory, change the icon, clicked okay, and nothing went wrong.

steveiliop56, 2023-12-18:
Hello!

Here is a video of the issue, the image is very recent I formatted it like 2 days ago but I have some updates to do, will try updating and post here if something is changed. Also this is a connection via WayVNC because I am not able to access my raspberry pi right now but I can confirm it has the same issue physically too.

https://github.com/raspberrypi/bookworm-feedback/assets/106091011/e4860b25-c481-41eb-a5c1-455c61db78b9

lurch, 2023-12-19:
Ahhh, thanks for the detailed reproduction. I can confirm that if I right-click on one of the menu entries and select Properties, then the Properties window just vanishes when I click on the program-icon in that window, and the taskbar also restarts.

Note that it's _only_ the taskbar (`wf-panel-pi`) that restarts, not "the whole desktop" - any other applications that you had running at the same time will continue to keep running.

JamesH65, 2023-12-19:
Yes, definitely can confirm this.

privet971, 2024-09-25:
Hello,

Could you use a terminal window to launch wf-panel-pi and retry to see error messages ?
I have the same issue with another wf-pane-pi item (Add/remove plugin) and I have the following message:
wf-panel-pi: ../../../../src/cairo-surface.c :955 : cairo_surface_destroy:  l'assertion « CAIRO_REFERENCE_COUNT_HAS_REFERENCE (&surface->ref_count) » a échoué