raspberrypi/bookworm-feedback #179: CE0 and CE1 no longer usable on the Pi 5 in Python

makermelissa, 2023-12-19:
I'm not 100% sure this is the correct place, but since the Python error is an OSError, it seems to make sense. 

When trying to access CE0 or CE1 via Python libgpiod bindings, I get `OSError: [Errno 16] Device or resource busy`. It works fine on a Pi 4 using the exact same SD card, so it seems to be a combination of OS and Hardware. If I try using gpiozero, I get "GPIO busy", which is essentially the same error.

I am just trying to run a simple script to toggle the GPIO on and off.

I have no device tree overlays loaded and using the lite version of the OS. SPI is enabled, but not actively being used.

makermelissa, 2023-12-19:
I forgot to mention I have SPI enabled and can access the GPIO when it is disabled, but then SPI isn't available.

makermelissa, 2023-12-19:
I did find a potential solution:
1. In /boot/firmware/config.txt I set dtparam=spi=off
2. I added dtoverlay=spi0-0cs to the bottom
3. I rebooted
After that code worked.

holmesha, 2023-12-31:
@makermelissa Is that all you did? I have the same error when trying to use an Adafruit EYESPI beret for a display and still get the same error for the CE0. Nothing else could be using that pin...But when I run gpioinfo I see CE0 and CE1 are in use...

Tried your fix/reboot and still get the error...

holmesha, 2023-12-31:
oooh I wonder if [this](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices#reassigning-or-disabling-the-spi-chip-enable-lines-3097985) is the solution? I haven't done this before so am worried I will mess up my gpio pins...

makermelissa, 2023-12-31:
Yep, I updated that script so you can disable their usage by the OS. All it does is place the correct dtoverlay into the config along with parameters if needed.

holmesha, 2023-12-31:
Got it, thank you. Unfortunately that line alone in the config.txt file didn't seem to solve the issue for me.

holmesha, 2024-01-01:
Actually never mind, I reloaded the Blinka library and it worked like a charm! Thank you for posting this Melissa! I thought I was going crazy...

lurch, 2024-07-02:
Sounds like this has been solved / worked-around? So closing...

closed 2024-07-02 by lurch