raspberrypi/bookworm-feedback #180: Scratch 3 GPIO Extension not working on Raspberry Pi 4 Modell B 32bit Bookworm

marcocaflisch, 2023-12-20:
Hello together,

I am using Scratch 3 (3.30.6) on a Raspberry Pi 4 Modell B with current full Pi OS Bookworm 32bit Version installation.
When Trying to add the 'Raspberry Pi GPIO' extensions no block group is added in the block groups list.
Same with the 'Raspberry Pi Simple Electronics' extension.
All other extensions work as expected, adding a new block group when selected.

Can anyone recommend a Scratch 3 / Raspbian Combination that works  for using these extensions on Pi 4 Modell B?

btw: I am new on raspberry, new on scratch 3, new on github and new on linux. Any 'for dummies' comments are welcome... :-)

lurch, 2023-12-21:
Duplicate of (or at least related to) #148 ?

marcocaflisch, 2023-12-21:
> Duplicate of (or at least related to) #148 ?

I think not a duplicate. Block's are not even appearing. in #148 they appear but don't work.

JonTheNiceGuy, 2023-12-21:
Confirmed; same issue with a Raspberry Pi 4 8GB on 64bit RaspberryPi OS Bookworm.

[Video in Gist](https://gist.github.com/JonTheNiceGuy/edbfe8b3523b4b3e13323470af881f6e)

marcocaflisch, 2023-12-23:
Solution: The 64bit Version works. :-)

JonTheNiceGuy, 2023-12-23:
Is that the same version of scratch, @marcocaflisch ?

mweb, 2023-12-24:
I got the same issue with a Raspberry Pi 4 8GB on 64bit RaspberryPi OS Bookworm.

mweb, 2023-12-24:
I got the following error within the logs:
Scratch Version is 3.30.6

    react-dom.production.min.js?ca5d:52 Uncaught Error: Cannot find module '/opt/Scratch 3/resources/static/gpiolib.node'
    Require stack:
    /opt/Scratch 3/resources/app.asar/index.html
        at Module._resolveFilename (node:internal/modules/cjs/loader:1002:15)
        at Function.o._resolveFilename (node:electron/js2c/renderer_init:2:3879)
        at Module._load (node:internal/modules/cjs/loader:848:27)
        at Function.f._load (node:electron/js2c/asar_bundle:2:13330)
        at Function.o._load (node:electron/js2c/renderer_init:2:3109)
        at Module.require (node:internal/modules/cjs/loader:1068:19)
        at require (node:internal/modules/cjs/helpers:103:18)
        at eval (index.js?2f37:6:1)
        at Object../node_modules/scratch-vm/src/extensions/scratch3_pigpio/index.js (1.bundle.js:8039:116)
        at n (bootstrap:63:22)

appaaron, 2023-12-24:
Same problem for me. Everything worked fine until the OS update. Now I can not get the Raspberry Pi extension working.

marcocaflisch, 2023-12-26:
> Is that the same version of scratch, @marcocaflisch ?

I simply downloaded the current Image "Raspberry Pi OS bookworm full 64bit" with the "raspberry pi downloader" or so from raspberrypi.com and the installed Scratch3 worked. Unfortunately I don't have my pi with me for the next few days to check the version.

Other hints: user=pi language=English\GB

Bengo, 2023-12-28:
Same problem for me 
and same log
Cannot find module '/opt/Scratch 3/resources/static/gpiolib.node'
an another discussion for this problem can be found in https://forums.raspberrypi.com/viewtopic.php?t=361350

I think the last Scratch 3 version (3.30.6) installed does not have the gpiolib dependencie.

grega, 2024-02-22:
The latest 3.30.8 release fixes issues with loading the GPIO and Simple Electronic extensions:

64-bit: https://archive.raspberrypi.org/debian/pool/main/s/scratch3/scratch3_3.30.8_arm64.deb

32-bit: https://archive.raspberrypi.org/debian/pool/main/s/scratch3/scratch3_3.30.8_armhf.deb