raspberrypi/bookworm-feedback #181: Third party desktop environments require the installation of "gldriver-test" package

sudo-splinter-cell, 2023-12-21:
Device: RPI-5
OS: RPIOS Lite Bookworm (with latest updates)

If you install a third party desktop environment on top of a lite installation of RPIOS (in my case it is KDE), even after selecting the "autologin to desktop" option using rpi-config, the desktop manager never starts (Lightdm or SDDM in my case) unless you install the "gldriver-test" package from the debian repo. I have no idea what this package does or adds to my system that it enables the start of the desktop managers. But it does.

This wasn't a requirement in Raspberry Pi-4 and somehow it is only necessary on a RPI-5. If this particular package (gldriver-test) ceases to exist at some point in the future, i do not know how we will be able to start desktop environments on a RPIOS other than the default one. So i consider this to be a bug which requires fixing.

sudo-splinter-cell, 2023-12-22:
[https://forums.raspberrypi.com/viewtopic.php?t=358853#p2152049](url)

`If you're trying to build up an X11 GUI-ful on Bookworm Lite, you may have already discovered that it doesn't work, while the exact same disk popped into a Pi4 works as it always has, and the GUI comes up.

The error in /var/log/Xorg.0.log is: Cannot run in framebuffer mode. Please specify busIDs

Simple fix: Install the package gldriver-test. This package adds two services that prepare the X11 graphics configuration for use on a Pi5. It appears that the rp1-test service is the important one. The second service (glamor-test) isn't needed unless you're using glamor (I'm not), but it appears to be harmless to leave it enabled.`

cmitu, 2024-01-14:
I don't think it's just 3rd party - installing even the standard Xorg desktop env over RaspiOS Lite is going to yield the same error without adding `gldriver-test`.

The issue IMHI (not a bug per-se) is that `gldriver-test` is not pulled in (as a dependency) by any other xorg(-server) packages, so unless it's manually installed no Xorg based session will start correctly. Adding it as a dependency or rolling the scripts contained into another RPT xorg-related package would make it easier for users that start with the Lite image and then want to install an Xorg desktop environment.

andrum993, 2024-01-18:
This is to be fixed by adding gldriver-test as  dependency to Xorg under Raspberry Pi OS - see https://forums.raspberrypi.com/viewtopic.php?p=2182262#p2182209 👍

cmitu, 2024-01-24:
I see now that `gldriver-test` is now _recommended_ when installing `xserver-xorg-core` (version  _21.1.7-3+rpt2+deb12u4_), this change should take care of installing the necessary package on RaspiOS.

Change is also mentioned in the changelog(!):
> xorg-server (2:21.1.7-3+rpt2+deb12u4) bookworm; urgency=medium
>  * Add gldriver-test to recommended packages
> -- Serge Schneider <serge@raspberrypi.com>  Thu, 18 Jan 2024 14:32:09 +0000

Thank you. I guess this issue can be closed, @sudo-splinter-cell ?

sudo-splinter-cell, 2024-01-24:
All right.

closed 2024-01-24 by sudo-splinter-cell