raspberrypi/bookworm-feedback #182: Selected keyboard layout only applies to current user and login screen

libewa, 2023-12-29:
When changing the keyboard layout using `raspi-config` or Keyboard & Mouse settings, the change only seems to apply to the current user.

This is probably because keyboard layout changes lie in multiple places: `/etc/default/keyboard` and `$HOME/.config/wayfire.ini`.
The Wayfire keyboard layout is set to 105pc, gb when creating a new user, and is only changed for the current user by `raspi-config`. While `/etc/default/keyboard` is also changed, it only applies to the console (and maybe the login screen?) and not Wayfire. Other users, whose config has not been changed, still use the default British keyboard, and, because they may have no admin rights, have to edit `wayfire.ini` manually.

This is annoying, as some keyboard layouts are very different from the QWERTY keyboard, and depending how many characters are changed, the user might not even be able to enter `raspi-config` or edit configuration files, as they may not be able to type the password they set on an account with the right layout.

lurch, 2023-12-30:
Is this the same issue as #128 which is fixed in the latest Bookworm downloads, or is this something else?

I'm not sure exactly what your issue is, so could you please provide a detailed set of reproduction steps, and note what actually happened, and what you expected to happen? Thanks.

(And if you're curious, you can look at https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L558-L629 to see what `raspi-config` is actually changing under the hood.)

libewa, 2023-12-31:
no, the problem is that non-admin users are not able to change their keyboard layout.

lurch, 2024-01-11:
> no, the problem is that non-admin users are not able to change their keyboard layout.

They **are** able to change their keyboard layout, it just involves hand-editing `$HOME/.config/wayfire.ini` (which admittedly isn't as user-friendly as using `raspi-config`).

@spl237 I've done a bit of digging, and I _think_ what is happening is that:
1. User uses `raspi-config` to change their default keyboard setting (e.g. from `gb` to `fr`) which automatically changes a bunch of files - and they expect this change to be system-wide
2. User creates a **new** user account
3. User logs in with their new user account and is surprised that the keyboard has changed back from `fr` to `gb`

Is that an accurate summary of your experience / expectation @libewa ?

I think this is because the `$HOME/.config/wayfire.ini` for the second user gets created from `/etc/wayfire/template.ini` (see https://github.com/raspberrypi-ui/raspberrypi-ui-mods/blob/master/etc/wayfire/template.ini ) and this says `xkb_layout = gb`.
So _maybe_ if the user changes their keyboard layout with `raspi-config`, it should _also_ change the keyboard in `/etc/wayfire/template.ini` so that any subsequently-added users automatically inherit the same keyboard layout? (I guess it's pretty unlikely that different users will be plugging and unplugging different USB keyboards with different layouts from a particular Raspberry Pi :laughing: )

spl237, 2024-01-11:
This is intended behaviour . Just because one user wants a particular keyboard layout, it shouldn't be assumed that all others want the same, and it isn't as if changing the layout is that difficult with rc-gui or lxinput. A new user account will be set to defaults, and that includes the UK keyboard as the default.

Editing template.ini would be a bad idea for numerous reasons - first, you'd need sudo access, but more importantly, that is a file which is installed with the package, so if you then subsequently update the package having edited it, the updater will notice and will ask you difficult questions...

closed 2024-01-11 by spl237

libewa, 2024-01-17:
@spl237 raspi-config and rc-gui need admin privileges, and lxinput i think does too.

libewa, 2024-01-17:
@lurch yes, but the second user wasn't able to change their kb layout (with ui tools).

spl237, 2024-01-17:
> @spl237 raspi-config and rc-gui need admin privileges, and lxinput i think does too.

lxinput doesn't need admin privileges to edit user keyboard layout, and nor does rc_gui.

lurch, 2024-01-24:
Myself and Simon have just dug into the source-code for `lxinput` (which is the same thing as the "Mouse and Keyboard Settings" in the "Preferences" menu), and under-the-hood it makes calls into `raspi-config`; which means that lxinput **does** need admin privileges too.
So I'm afraid the only way for non-admin users to change their keyboard layout is by hand-editing the `xkb_layout` setting in their `~/.config/wayfire.ini` file.