raspberrypi/bookworm-feedback #183: XDG_CURRENT_DESKTOP needs to be set

Botspot, 2024-01-02:
`XDG_CURRENT_DESKTOP=Wayfire` should be set on PiOS in order for programs relying on xdg-desktop-portal-wlr to work properly.
Currently, this var is unset.

This is discussed on the wayfire repo here: https://github.com/WayfireWM/wayfire/issues/1995
I have confirmed this affects RustDesk (an open source teamviewer/anydesk/realvnc alternative)
It likely affects all other applications that make use of xdg-desktop-portal-wlr, including other remote desktop software and screensharing on other web browsers.
To keep all info on one place, this has already been reported on the bookworm feedback forum [here](https://forums.raspberrypi.com/viewtopic.php?t=357697) with no response so far, and was mentioned on [this pi-apps issue](https://github.com/Botspot/pi-apps/issues/2083)
Tagging @theofficialgman

ghollingworth, 2024-01-05:
Looks like this is an upstream wayfire issue and will be fixed there?

theofficialgman, 2024-01-05:
Even if it is fixed upstream, [piOS wayfire](https://github.com/raspberrypi-ui/wayfire) is forked from 0.7.5 (December 1st 2022 release) and any changes upstream won't automatically propagate.

Botspot, 2024-02-20:
Asking the Pi devs: at what place in the boot sequence would be the most acceptable place to set an appropriate value of `XDG_CURRENT_DESKTOP`, ensuring that it detects X11, Wayland, and other compositors reliably?