raspberrypi/bookworm-feedback #184: Getting thrown out of my desktop session every day or so

sfatula, 2024-01-02:
I'll be sitting here on my Pi 5 kernel 6.1.0-rpi7-rpi-2712 and all of the sudden I am logged out and anything I was working on and not saved, gone of course. Just starting to load my new PI 5, so, not much loaded yet. Here's the log file from journalctl from the latest time I was thrown out of my session. Not sure what to do here but suspect a bug. 

`Jan 02 01:51:35 StevePI5 dbus-daemon[730]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 02 01:51:35 StevePI5 systemd[1]: Started systemd-hostnamed.service - Hostname Service.
Jan 02 01:52:05 StevePI5 systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Jan 02 01:52:53 StevePI5 dbus-daemon[730]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.174' (uid=1000 pid=40259 comm="/usr/lib/thunderbird/thunderbird")
Jan 02 01:52:53 StevePI5 systemd[1]: Starting systemd-hostnamed.service - Hostname Service...
Jan 02 01:52:53 StevePI5 dbus-daemon[730]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 02 01:52:53 StevePI5 systemd[1]: Started systemd-hostnamed.service - Hostname Service.
Jan 02 01:52:58 StevePI5 xdg-desktop-por[1746]: Error reading events from display: Broken pipe
Jan 02 01:52:58 StevePI5 polkitd[734]: Unregistered Authentication Agent for unix-session:2 (system bus name :1.50, object path /org/mate/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected from bus)
Jan 02 01:52:58 StevePI5 systemd[1388]: xdg-desktop-portal-gtk.service: Main process exited, code=exited, status=1/FAILURE
Jan 02 01:52:58 StevePI5 systemd[1388]: xdg-desktop-portal-gtk.service: Failed with result 'exit-code'.
Jan 02 01:52:58 StevePI5 systemd[1388]: xdg-desktop-portal-gtk.service: Consumed 1.423s CPU time.
Jan 02 01:52:58 StevePI5 systemd[1388]: app-flatpak-org.keepassxc.KeePassXC-9278.scope: Consumed 4.117s CPU time.
Jan 02 01:52:58 StevePI5 lightdm[1241]: pam_unix(lightdm:session): session closed for user sfatula
Jan 02 01:52:58 StevePI5 systemd-logind[751]: Session 2 logged out. Waiting for processes to exit.
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/ldac
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSink/aptx_hd
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/aptx_hd
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSink/aptx
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/aptx
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSink/sbc_xq
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/sbc_xq
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/faststream
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSink/opus_05
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/opus_05
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Jan 02 01:52:58 StevePI5 bluetoothd[865]: Endpoint unregistered: sender=:1.58 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Jan 02 01:52:58 StevePI5 lightdm[46480]: pam_unix(lightdm-greeter:session): session opened for user lightdm(uid=106) by (uid=0)
Jan 02 01:52:58 StevePI5 systemd[1]: Created slice user-106.slice - User Slice of UID 106.
Jan 02 01:52:58 StevePI5 systemd[1]: Starting user-runtime-dir@106.service - User Runtime Directory /run/user/106...
Jan 02 01:52:58 StevePI5 systemd-logind[751]: New session c2 of user lightdm.
Jan 02 01:52:58 StevePI5 systemd[1]: Finished user-runtime-dir@106.service - User Runtime Directory /run/user/106.
Jan 02 01:52:58 StevePI5 systemd[1]: Starting user@106.service - User Manager for UID 106...
Jan 02 01:52:58 StevePI5 (systemd)[46484]: pam_unix(systemd-user:session): session opened for user lightdm(uid=106) by (uid=0)
Jan 02 01:52:59 StevePI5 systemd[46484]: Queued start job for default target default.target.
Jan 02 01:52:59 StevePI5 systemd[46484]: Created slice app.slice - User Application Slice.
Jan 02 01:52:59 StevePI5 systemd[46484]: Created slice session.slice - User Core Session Slice.
Jan 02 01:52:59 StevePI5 systemd[46484]: Reached target paths.target - Paths.
Jan 02 01:52:59 StevePI5 systemd[46484]: Reached target timers.target - Timers.
Jan 02 01:52:59 StevePI5 systemd[46484]: Starting dbus.socket - D-Bus User Message Bus Socket...
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on dirmngr.socket - GnuPG network certificate management daemon.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on gcr-ssh-agent.socket - GCR ssh-agent wrapper.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on gnome-keyring-daemon.socket - GNOME Keyring daemon.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on gpg-agent-browser.socket - GnuPG cryptographic agent and passphrase cache (access for web browsers).
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on gpg-agent-extra.socket - GnuPG cryptographic agent and passphrase cache (restricted).
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on gpg-agent-ssh.socket - GnuPG cryptographic agent (ssh-agent emulation).
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on gpg-agent.socket - GnuPG cryptographic agent and passphrase cache.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on pipewire-pulse.socket - PipeWire PulseAudio.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on pipewire.socket - PipeWire Multimedia System Socket.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on pk-debconf-helper.socket - debconf communication socket.
Jan 02 01:52:59 StevePI5 systemd[46484]: Listening on dbus.socket - D-Bus User Message Bus Socket.
Jan 02 01:52:59 StevePI5 systemd[46484]: Reached target sockets.target - Sockets.
Jan 02 01:52:59 StevePI5 systemd[46484]: Reached target basic.target - Basic System.
Jan 02 01:52:59 StevePI5 systemd[1]: Started user@106.service - User Manager for UID 106.
Jan 02 01:52:59 StevePI5 systemd[46484]: Started pipewire.service - PipeWire Multimedia Service.
Jan 02 01:52:59 StevePI5 systemd[1]: Started session-c2.scope - Session c2 of User lightdm.
Jan 02 01:52:59 StevePI5 systemd[46484]: Started wireplumber.service - Multimedia Service Session Manager.
Jan 02 01:52:59 StevePI5 systemd[46484]: Started pipewire-pulse.service - PipeWire PulseAudio.
Jan 02 01:52:59 StevePI5 systemd[46484]: Reached target default.target - Main User Target.
Jan 02 01:52:59 StevePI5 systemd[46484]: Startup finished in 148ms.
Jan 02 01:52:59 StevePI5 systemd[46484]: Starting dbus.service - D-Bus User Message Bus...
[... 190 lines cut ...]
Jan 02 01:53:14 StevePI5 systemd[46484]: Closed dbus.socket - D-Bus User Message Bus Socket.
Jan 02 01:53:14 StevePI5 systemd[46484]: Closed dirmngr.socket - GnuPG network certificate management daemon.
Jan 02 01:53:14 StevePI5 systemd[46484]: Closed gcr-ssh-agent.socket - GCR ssh-agent wrapper.
Jan 02 01:53:14 StevePI5 systemd[46484]: Closed gnome-keyring-daemon.socket - GNOME Keyring daemon.
Jan 02 01:53:14 StevePI5 systemd[46484]: Closed gpg-agent-browser.socket - GnuPG cryptographic agent and passphrase cache (access for web browsers).
[... cut ...]
Jan 02 01:53:14 StevePI5 systemd[1]: run-user-106.mount: Deactivated successfully.
Jan 02 01:53:14 StevePI5 systemd[1]: user-runtime-dir@106.service: Deactivated successfully.
Jan 02 01:53:14 StevePI5 systemd[1]: Stopped user-runtime-dir@106.service - User Runtime Directory /run/user/106.
Jan 02 01:53:14 StevePI5 systemd[1]: Removed slice user-106.slice - User Slice of UID 106.
Jan 02 01:53:14 StevePI5 systemd[1]: user-106.slice: Consumed 1.341s CPU time.
Jan 02 01:53:23 StevePI5 systemd[1]: systemd-hostnamed.service: Deactivated successfully.
`

lurch, 2024-01-02:
Possibly the same thing as #146 ?

sfatula, 2024-01-02:
Yeah I read that but not sure it's the same. The OP talks about monitor being switched off. I am sitting here typing as it happens with my monitor on (of course). It also talks about pygame a lot, not using it. The log file looks different, not the same messages. 

Now, I do turn off my monitor each night and it could well be that causes my nightly issue, I do have that as each morning I am greeted with a login screen. But it's also happening while using the machine which is different I think. Strangely, I noted the time it happened today (when first using machine) and it was about exactly when I turned the monitor on. 

I would try the vc4.force_hotplug though I am not sure it's still valid on the PI 5 with bookworm? 

It should be noted I am current on all the software updates.

lurch, 2024-01-02:
> I would try the vc4.force_hotplug though I am not sure it's still valid on the PI 5 with bookworm?

It is, see my comments here https://github.com/raspberrypi/bookworm-feedback/issues/67#issuecomment-1863340591

sfatula, 2024-01-03:
So, adding that to config.txt option made no difference as far as the during my use portion of the issue. i.e., was typing away tonight and in the middle of typing was logged out (or it crashed, whatever) and had to log back in. Will see in the morning if it affects the turning the monitor off portion.

6by9, 2024-01-03:
> So, adding that to config.txt option 

It's an option for cmdline.txt, not config.txt

sfatula, 2024-01-03:
Yep, typed the wrong thing, I added it to cmdline.txt, sorry about that. I did wake up today to a machine that did not have the login window, so, it survived all night. Adding it may have resolved the monitor off all night problem. Doesn't help the during use problem though. 

My cmdline.txt looks like this:

console=serial0,115200 console=tty1 root=PARTUUID=cfd930c2-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US vc4.force_hotplug=2

popcornmix, 2024-01-03:
> vc4.force_hotplug=2

so you have one monitor attached and it's connected to HDMI-A-2 (rather than HDMI-A-1)?
It would be more usual to use the other socket (closest to power connector) if only using one monitor.

sfatula, 2024-01-03:
Correct, am using the furthest away. Can I not use it? It makes it easier to plug in a second one if/when I want to as I can then see the place to plug it in.

popcornmix, 2024-01-03:
Some fullscreen apps (like kodi in fullscreen mode) only support the first hdmi display.
And most help given about configuring displays will assume you are using the first one.

But if you are happy with that, then it should be fine generally.

On Pi4, the recommendation is slightly stronger, as the hardware for the first hdmi port is slightly more capable (it can support higher pixel rates), but on Pi5 the hardware is the same.

sfatula, 2024-01-07:
Still getting thrown out daily, just never with monitor off. I even swapped HDMI cables just in case.