raspberrypi/bookworm-feedback #186: X11 does not work with piOS Bookworm

GYKgamer, 2024-01-02:
Whenever selecting "X11" under "Wayland" in "Advanced Options" and rebooting, the system will come up with a black screen, no mouse, no nothing. Monitor says that no signal has been detected and VNC will just throw an error, SSH still works and you can go back to Wayfire, after rebooting everything goes back into working condition. I have tried with with both 32 bit and 64 bit. Raspberry pi 5 8gb

popcornmix, 2024-01-02:
As someone who is currently running an 8GB Pi5 bookworm with X11 enabled with raspi-config and doesn't have a black screen, we'll need more information.

For you, does a clean install of RPiOS bookworm exhibit the issue if all you do is switch to X11 (without installing any other software or VNC)?

If this works correctly, can you identify at what point the screen goes black when you set it up (e.g. enabling VNC or installing other apps).

GYKgamer, 2024-01-02:
After testing I found out X11 does indeed work after rebooting, but enabling VNC does not, it just says it can't connect to my raspberry pi, even though the raspberry pi is connected to the internet, but on a clean install X11 does work, I don't know how to test if it is running on X11 though.

In all my times trying to get X11 I usually have a program like Wine, or at least a couple of folders, but nothing which should affect X11, no clue. I will test further

GYKgamer, 2024-01-02:
I found the issue, after updating and rebooting, the system will only show a black screen.

lurch, 2024-01-02:
> I don't know how to test if it is running on X11 though.

There's a command [here](https://www.raspberrypi.com/documentation/computers/configuration.html#setting-your-displays-resolution-and-rotation) you can use for that.

GYKgamer, 2024-01-02:
[image]
Yep, when running with X11 and then updating i get no display

lurch, 2024-01-02:
I suspect it'll be much easier for us to help you investigate this if you just use a normal directly-attached HDMI monitor, rather than a "Game Capture" device.

GYKgamer, 2024-01-02:
I can do that, it's just easier for me to hook it up to a capture card.

kolakowj, 2024-01-02:
I have the same issue.  Only way to get a desktop environment is to directly attach an HDMI monitor with Wayland running. 
1. Tried TigerVNC client on Mac but the screen is super tiny (couldn't get it change size - tried a coupe of different ways including maxing out the raspi-config VNC resolution ) and its also completely missing the taskbar on top.  
2. Tried Tiger VnC server on the RPI too but that did not solve anything.  

Need it to run the RPI headless in a remote location. 

BTW, what I really miss about RealVNC is that I could run the RPI desktop on my iPads using the RealVNC IOS app.

SYSTEM DESCRIPTION:
Updated RPI 4B 8GB 64b from Bullseye to Bookworm by changing the sources.list and executing apt full-upgrade. 

cat /proc/cpuinfo >> 
     Revision: d03114
     Model: Raspberry Pi 4 Model B Rev 1.4

cat /proc/version >>
     Linux version 6.1.69-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1711 SMP PREEMPT Thu Dec 21 14:27:49 GMT 2023

Also updated bootloader>>
BOOTLOADER: up to date
   CURRENT: Thu May 11 06:26:03 AM UTC 2023 (1683786363)
    LATEST: Thu May 11 06:26:03 AM UTC 2023 (1683786363)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2711/latest)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0

ps ax | grep [w]ayfire >>
   1173 ?        Ssl    8:37 /usr/bin/wayfire
   1370 ?        Ss     0:00 /usr/bin/ssh-agent /usr/bin/wayfire-pi

-eof-

kolakowj, 2024-01-02:
So I have TightVNC Viewer working on the Mac.  I figured that with the upgrade there might be some conflicting settings from Bullseye. So I went through and commented out video related items in Config.txt (except for the drivers) and specifically set a resolution in the new Wayland config file. Also set the WayVNC Auth to false. Now I have a crisper image on the monitor and have all the functionality over WayVNC server > TigerVNC Viewer via my Mac. Remember to use port 5900 in the connection address.  Still have to test what happens when I reboot with no monitor plugged in but need to go make dinner now.

Here are the details:

sudo nano .config/wayfire.ini
   ADDED:
	[output:HDMI-A-1]
	mode = 1920x1080@60

sudo nano /boot/config.txt

   COMMENTED OUT:
	# uncomment if you get no picture on HDMI for a default "safe" mode
	# hdmi_safe=1

	# uncomment to force a console size. By default it will be display's size minus
	# overscan.
	#framebuffer_width=1280
	#framebuffer_height=720

	# uncomment if hdmi display is not detected and composite is being output
	#hdmi_force_hotplug=1

	# uncomment to force a specific HDMI mode (this will force VGA)
	#hdmi_group=1
	#hdmi_mode=1

	# Disable compensation for displays with overscan
	disable_overscan=1

   UNCOMMENTED:
	# Enable DRM VC4 V3D driver
	dtoverlay=vc4-kms-v3d
	max_framebuffers=2

sudo nano /home/pi/.config/wayvnc/config
   SET TO FALSE:
	enable_auth=false

-30-

lurch, 2024-01-11:
> Updated RPI 4B 8GB 64b from Bullseye to Bookworm by changing the sources.list and executing apt full-upgrade.

We don't recommend nor support direct upgrades from Bullseye to Bookworm - see https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

lurch, 2024-07-18:
Closing as stale. If anyone in this issue is still having problems on an installed-from-fresh and fully-updated Bookworm OS, please add additional comments and I'll re-open.

closed 2024-07-18 by lurch