raspberrypi/bookworm-feedback #189: Performance options for fan, GPU memory, and overclocking removed from raspi-config for Raspberry Pi 3

steinhelten, 2024-01-06:
This is arguably not a bug, but I would certainly call it a mis-feature.  In looking at the code for raspi-config, it appears that someone decided to remove overclocking, GPU memory allocation, and fan control as performance options from raspi-config for Raspberry 3s.  I don't understand this.  My interest is the fan control.  I've tested this with both Buster (unmodified) and Bookworm (I modified the script to be able to call do_fan) and it works precisely as expected on a Raspberry Pi 3 Model B Rev 1.2 with both.  I would include my code, but I only changed the fan part, and I think the ultimate solution is to put all of them back for Raspberry PI 3s (and maybe other models).  The fact that you need a home-brew circuit (such as a 2N2222 transistor and a 1N4001 back emf diode) or a third-party circuit is irrelevant.  The raspberry was, after all, developed as an experimentation and investigation platform.  As for the overclocking and GPU memory - I'm entirely baffled by that.  Why was that removed?  I am willing to participate in testing and/or coding if that would be helpful.

lurch, 2024-01-10:
`gpu_mem` is now a [legacy option](https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#gpu_mem), and is no longer needed in Bookworm, as it now uses the KMS / DRM graphics driver.

( @nathan-contino It looks like https://www.raspberrypi.com/documentation/computers/configuration.html#performance-options and https://www.raspberrypi.com/documentation/computers/configuration.html#memory-split-nonint might need tweaking to match? )

@spl237 might be able to elucidate regarding the fan and overclocking options? (Although https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L3248 says "Set behaviour of GPIO case fan" so maybe that option has been repurposed to **only** refer to https://www.raspberrypi.com/products/raspberry-pi-4-case-fan/ ?)

steinhelten, 2024-01-10:
Thanks for the info!  The GPU stuff is a pure black box to me, so this is good information.  As for the fan... "repurposing" in this way would amount to "taking away" for all RPI3 owners, so I think this needs some careful consideration.  Why cut out support for legacy platforms and existing third-party and home-brew implementations?  Is there a way to offer either/or?  Maybe call it something different for the earlier platforms, make the two mutually exclusive?  Even with the RPI4 and 5, people might not use the "official" case....  One of my RPI4s is using the CanaKit heat sink case, and I can max it out for days at a time without hitting the thermal back-down limit, and that has no fan, so there's a lot to consider here.

steinhelten, 2024-01-29:
@spl237 is there anyting I can do to help facilitate a solution that continues to support those of us with fans on RPI3s?

lurch, 2024-01-30:
Does https://www.raspberrypi.com/documentation/computers/configuration.html#raspi-config-cli still work for you?

steinhelten, 2024-01-30:
No.  It does nothing.

***@***.***:/usr/bin# cp /boot/firmware/config.txt /tmp
***@***.***:/home/john# raspi-config nonint do_fan 1 17 65
***@***.***:/usr/bin# diff /boot/firmware/config.txt /tmp/config.txt
# Then I run my modified version interactively (see diff below)
***@***.***:/usr/bin# raspi-config-jl
***@***.***:/usr/bin# diff /boot/firmware/config.txt /tmp/config.txt
48d47
< dtoverlay=gpio-fan,gpiopin=17,temp=65000
***@***.***:/home/john#

I did the following change to enable it in the menu, but if you feel it's
clearer to call them different things, one for a third-party solution on 3s
and "official" case fans for 4s (and 5s), I'd be OK with that.  In fact, I
called it a "custom" fan instead of a "case" fan.

***@***.***:/usr/bin# diff raspi-config raspi-config-jl
3275c3275
<     [03]) FUN=$(whiptail --title "Raspberry Pi Software Configuration
Tool (raspi-config)" --menu "Performance Options" $WT_HEIGHT $WT_WIDTH
$WT_MENU_HEIGHT --cancel-button Back --ok-button Select \
---
>     0) FUN=$(whiptail --title "Raspberry Pi Software Configuration Tool
(raspi-config)" --menu "Performance Options" $WT_HEIGHT $WT_WIDTH
$WT_MENU_HEIGHT --cancel-button Back --ok-button Select \
3276a3277,3280
>           3>&1 1>&2 2>&3) ;;
>     3) FUN=$(whiptail --title "Raspberry Pi Software Configuration Tool
(raspi-config)" --menu "Performance Options" $WT_HEIGHT $WT_WIDTH
$WT_MENU_HEIGHT --cancel-button Back --ok-button Select \
>           "P2 Overlay File System" "Enable/disable read-only file system"
\
>           "P3 Fan" "Set behaviour of GPIO custom fan" \

I have not yet identified the code that is preventing it from doing
anything with the CLI.  Is that something you want me to research, or is
there a resource already familiar with the code?

steinhelten, 2024-01-30:
Yes, it works.  The first time I tried this, I failed to notice that "0" was to turn the fan ON and "1" was to turn the fan OFF, so it took me a bit to figure out what was happening.

```
root@bringebaer:/usr/bin# cp /boot/firmware/config.txt /tmp
root@bringebaer:/usr/bin# raspi-config nonint do_fan 0 17 65
root@bringebaer:/usr/bin# diff /boot/firmware/config.txt /tmp/config.txt
48d47
< dtoverlay=gpio-fan,gpiopin=17,temp=65000
root@bringebaer:/usr/bin# raspi-config nonint do_fan 1
root@bringebaer:/usr/bin# diff /boot/firmware/config.txt /tmp/config.txt
root@bringebaer:/usr/bin# 
```
This works silently - it doesn't even tell you that a reboot is needed to take effect.  But by the time I get to this level,  I could just copy the "sed" command out of the script and use that.

Even so, why do we have to take it out of the menu?  We don't have to call it the same thing.  In fact, in the change I made, I called it a "custom" fan instead of a "case" fan. 

```
root@bringebaer:/usr/bin# diff raspi-config raspi-config-jl
3275c3275
<     [03]) FUN=$(whiptail --title "Raspberry Pi Software Configuration Tool (raspi-config)" --menu "Performance Options" $WT_HEIGHT $WT_WIDTH $WT_MENU_HEIGHT --cancel-button Back --ok-button Select \
---
>     0) FUN=$(whiptail --title "Raspberry Pi Software Configuration Tool (raspi-config)" --menu "Performance Options" $WT_HEIGHT $WT_WIDTH $WT_MENU_HEIGHT --cancel-button Back --ok-button Select \
3276a3277,3280
>           3>&1 1>&2 2>&3) ;;
>     3) FUN=$(whiptail --title "Raspberry Pi Software Configuration Tool (raspi-config)" --menu "Performance Options" $WT_HEIGHT $WT_WIDTH $WT_MENU_HEIGHT --cancel-button Back --ok-button Select \
>           "P2 Overlay File System" "Enable/disable read-only file system" \
>           "P3 Fan" "Set behaviour of GPIO custom fan" \
root@bringebaer:/usr/bin# 
```

spl237, 2024-01-30:
As far as I can remember, we have never supported overclocking on the Pi 3, which is why it is not (and again, as far as I can remember) has never been configurable in raspi-config for that platform, so we haven't removed it - it was never there to start with!

We have also never sold a fan for any Pi cases other than those for the Pi 4 and Pi 5, because earlier platforms did not require active cooling. (Unless overclocked, in which case see my previous point above...)

You already know how to enable a custom fan by editing the configuration file, and given that in order to fit a fan to earlier models of Pi, you will need the DIY skills to be able to make one yourself, it's not that much of a stretch to suggest that you might also have to edit a config file by hand.

Having an option to control a piece of hardware that doesn't exist unless you make yourself is just going to cause confusion for many users. So the fan control reflects only the official Raspberry Pi hardware to which it applies.

closed 2024-01-30 by spl237

steinhelten, 2024-01-30:
I am forced to disagree because
1) It was definitely there with buster (see pictures below) before the RPI4 existed.  It is demonstrably false to say that it was never there before.  It was.
2) There are numerous 3rd party options that don't require you to make it yourself (links to follow - only a partial list)
3) You can DEFINITELY push the RPI3 into thermal stress with reasonable workloads, and I NEVER "overclock" though I DO use the FULLY SUPPORTED 1200MHz and 1400MHz speeds.  Recompiling Marlin with -j4 does it rather quickly.  It starts backing the clock down pretty fast.  To say that the earlier platforms did not require active cooling is only true if you don't push them very hard or are OK with the clock speed dropping down.  I have even had them SHUT DOWN (with heatsinks)
4) Heatsinks weren't standard on the earlier models either, but trust me, you want them.
5) You can call it what you will in the menu "optional 3rd party fan" for instance.

https://www.amazon.com/Miuzei-Raspberry-Heatsinks-Supply-Compatible/dp/B07BTHNW9W
https://www.amazon.com/Enokay-Raspberry-Model-Dual-Heatsink/dp/B07XC14MM1/
https://www.ebay.com/itm/383777675284
https://www.amazon.com/Geekworm-Raspberry-Cooling-Expansion-X728-A1/dp/B08B5TQH6W/

[image]
[image]

steinhelten, 2024-01-30:
Oh, and tangentially - if "Overlay File System" is the only option you're going to offer - you might want to make it "P1" instead of "P2" since it has already changed from "P3".