raspberrypi/bookworm-feedback #191: Unexpected graphics display freeze

qrp73, 2024-01-07:
I catch this issue several times, it happens about 1-2 times per week. First I catch it about 1-2 week ago. Before that I never catch it on bookworm, so it seems that something is broken with latest updates.

How it happens: at some point the display (connected to HDMI) stops to update, mouse pointer no more moves. But I still can listen music from VLC player (minimized). I can connect through SSH, but btop don't show anything suspicious. 

The free memory is available, CPU load close to zero.

I tried to kill the following processes: wf-panel, wayfire, pcmanfm, and some other, they are successfully restarted and I see welcome screen process, but it don't appears on the display. The display still shows frozen picture...

I tried to disconnect HDMI and connect it again, but it leads to display signal disappearing, so the system don't see display anymore.

I do ```sudo reboot``` from SSH console, but it take very long time for reboot. It seems that some process unload take too long time. 

After reboot all works ok.

logs from journalctl, freeze happens at about 21:40, when I realize that system is freezed I disconnected EXTREME flash disk and tried Alt+SysRq+F:
```
Jan 07 20:55:35 raspi udisksd[665]: Mounted /dev/sda1 at /media/pi/EXTREME on behalf of uid 1000
Jan 07 20:55:49 raspi systemd[1]: Started pcscd.service - PC/SC Smart Card Daemon.
Jan 07 21:01:25 raspi systemd[1]: pcscd.service: Deactivated successfully.
Jan 07 21:13:56 raspi systemd[1]: Started pcscd.service - PC/SC Smart Card Daemon.
Jan 07 21:17:01 raspi CRON[105921]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jan 07 21:17:01 raspi CRON[105922]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Jan 07 21:17:01 raspi CRON[105921]: pam_unix(cron:session): session closed for user root
Jan 07 21:27:06 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 393216 bytes), total 32768 (slots), used 960 (slots)
Jan 07 21:27:06 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 393216 bytes), total 32768 (slots), used 960 (slots)
Jan 07 21:41:41 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 60 (slots)
Jan 07 21:41:41 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 2359296 bytes), total 32768 (slots), used 0 (slots)
Jan 07 21:41:41 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 2224128 bytes), total 32768 (slots), used 60 (slots)
Jan 07 21:41:41 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 60 (slots)
Jan 07 21:41:41 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 2359296 bytes), total 32768 (slots), used 0 (slots)
Jan 07 21:41:42 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 2224128 bytes), total 32768 (slots), used 60 (slots)
Jan 07 21:47:41 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:41 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:41 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:42 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:42 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:43 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:43 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:43 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:44 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:47:45 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
[... 485 lines cut ...]
Jan 07 21:48:17 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:17 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:17 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:17 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:17 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:17 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:18 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:18 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
Jan 07 21:48:18 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=17. ret=-28
[... cut ...]
Jan 07 21:48:20 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=9. ret=-28
Jan 07 21:48:20 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=9. ret=-28
Jan 07 21:48:20 raspi kernel: vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry. Requested size=9. ret=-28
Jan 07 21:49:05 raspi sshd[109162]: Accepted password for *** from 192.168.***.*** port ***** ssh2
Jan 07 21:49:05 raspi sshd[109162]: pam_unix(sshd:session): session opened for user pi(uid=1000) by (uid=0)
...
```

cmitu, 2024-01-08:
Already reported in https://github.com/raspberrypi/linux/issues/5674 ?

popcornmix, 2024-01-08:
I'll close this as a duplicate. Continue discussion in the linked linux kernel issue.

closed 2024-01-08 by popcornmix

qrp73, 2024-01-12:
I think if you close this issue, you're needs to open https://github.com/raspberrypi/linux/issues/5674 because it also closed, so currently both are closed, but the bug is still there...