raspberrypi/bookworm-feedback #192: no thin border around terminal windows

UgaBugaOfdeSpeer, 2024-01-10:
If I remember well, in the past (bullseye) there was always a thin border line around each terminal window.
This seems not to be the case anymore with "bookworm".

I assume there is somewhere a configuration parm that can be changed to have the thin borders back again.

Does somebody know how to fix this?

lurch, 2024-01-11:
I think this is a X11 / Wayland thing, more than it is a Bullseye / Bookworm thing, as I still get the visible window borders on a Pi 3B+ running Bookworm (which uses X11). So I guess you must be using a Pi4 or a Pi5 :slightly_smiling_face: 

I've not looked into it myself, but you might find what you're looking for in https://github.com/WayfireWM/wayfire/wiki/Configuration

lurch, 2024-07-23:
Did you manage to resolve your problem @UgaBugaOfdeSpeer and/or is it resolved in the latest Raspberry Pi OS Bookworm image?