raspberrypi/bookworm-feedback #193: vcgencmd man page lists a "display_power" command which doesn't exist

raindog308, 2024-01-12:
The man page for vcgencmd has:

```
       display_power 0|1|-1
       display_power 0|1|-1 display
              Show current display power state, or set the  display  power  state.   vc‐
              gencmd  display_power  0  will turn off power to the current display.  vc‐
              gencmd display_power 1 will turn on power to the display. If no  parameter
              is  set, this will display the current power state. The final parameter is
              an optional display ID, as returned by tvservice -l or from the table  be‐
              low, which allows a specific display to be turned on or off.

```
However:

```
# cat /etc/debian_version 
12.4
# vcgencmd commands
commands="commands, set_logging, bootloader_config, bootloader_version, cache_flush, codec_enabled, get_mem, get_rsts, measure_clock, measure_temp, measure_volts, get_hvs_asserts, get_config, get_throttled, pmicrd, pmicwr, read_ring_osc, version, readmr, otp_dump, pmic_read_adc, power_monitor"
# vcgencmd display_power 0
vc_gencmd_read_response returned -1
error=1 error_msg="Command not registered"
```

Note that this man page also refers to tvservice, which doesn't exist any more.

lurch, 2024-01-29:
This issue should probably be transferred to https://github.com/raspberrypi/utils @pelwell ?

pelwell, 2024-01-29:
But it's already correct there: https://github.com/raspberrypi/utils/blob/master/vcgencmd/vcgencmd.1#L87-L89

pelwell, 2024-01-29:
Sorry - wrong issue.

timg236, 2024-01-29:
I think the man page is correct. As pelwell notes the firmware will only respond with gencmds available on that firmware/platform and since the command was run on Pi5 none of the display related commands will be available.

closed 2024-01-29 by lurch