raspberrypi/bookworm-feedback #194: dtparam claims to support hdmi parameter but it does not

raindog308, 2024-01-12:
```
# dtparam -h hdmi
hdmi                    Set to "off" to disable the HDMI interface
                        (default "on")
# dtparam hdmi=off
* Unknown parameter 'hdmi'

```
This is on OS 12.4

lurch, 2024-01-29:
@pelwell Another issue for https://github.com/raspberrypi/utils ?

pelwell, 2024-01-29:
Not all boards support all parameters. I'm going to guess this is on a Pi 5 - it's good form to say which platform is being discussed.

If the problem is actually that the device in question doesn't support the "hdmi" parameter, open an issue in our linux repo: https://github.com/raspberrypi/linux/issues

lurch, 2025-01-09:
Closing due to lack of activity. I'll re-open if @raindog308 is able to provide more details.

closed 2025-01-09 by lurch