raspberrypi/bookworm-feedback #195: Raspbian bookworm

Smiril, 2024-01-16:
I got a huge hung on a raspberry pi 4  64Bit
Os: raspbian bookworm 64Bit 
The only app that runs is cubicSDR and after starting the app(load hackrf and a freq) it seems to hung

not clickable on anything on the desktop 

it just freezes

lurch, 2024-01-16:
That sounds like a problem you need to report to the cubicSDR developers?

popcornmix, 2024-01-16:
raspbian is probably not what you mean (raspbian is a 32-bit build of debian, not handled by Raspberry Pi itself).
RPiOS is the name of the OS images produced by Raspberry Pi.

But cubicSDR is not RPiOS or raspbian specific - it is a debian package:
https://packages.debian.org/bookworm/cubicsdr

But support for the package is probably best handled here: https://github.com/cjcliffe/CubicSDR/issues

Possibly this is related? https://github.com/cjcliffe/CubicSDR/issues/979

But I've never used cubicSDR so cannot help more.

lurch, 2024-01-16:
Closing this in favour of the upstream issue.

closed 2024-01-16 by lurch