raspberrypi/bookworm-feedback #196: WARNING: CPU: 0 PID: 388514 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x200/0x270

qrp73, 2024-01-17:
Some strange error periodically happens, it leads to mouse cursor freeze for 5-10 seconds, the following records appears in the log:

```
Jan 17 00:42:31 raspi kernel: swiotlb_tbl_map_single: 2 callbacks suppressed
Jan 17 00:42:31 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 475136 bytes), total 32768 (slots), used 2 (slots)
Jan 17 00:42:31 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 106 (slots)
Jan 17 00:42:31 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 479232 bytes), total 32768 (slots), used 0 (slots)
Jan 17 00:42:31 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 475136 bytes), total 32768 (slots), used 2 (slots)
Jan 17 00:42:32 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 106 (slots)
Jan 17 00:42:32 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 479232 bytes), total 32768 (slots), used 0 (slots)
Jan 17 00:42:32 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 158 (slots)
Jan 17 00:42:32 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 90 (slots)
Jan 17 00:42:32 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 22 (slots)
Jan 17 00:42:32 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 158 (slots)
Jan 17 00:42:43 raspi kernel: swiotlb_tbl_map_single: 1 callbacks suppressed
Jan 17 00:42:43 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 22 (slots)
Jan 17 00:44:18 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 434176 bytes), total 32768 (slots), used 18 (slots)
Jan 17 00:44:18 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 331776 bytes), total 32768 (slots), used 0 (slots)
Jan 17 00:44:18 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 94 (slots)
Jan 17 00:44:18 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 434176 bytes), total 32768 (slots), used 18 (slots)
Jan 17 00:44:18 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 331776 bytes), total 32768 (slots), used 0 (slots)
Jan 17 00:44:18 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 94 (slots)
Jan 17 00:46:30 raspi kernel: ------------[ cut here ]------------
Jan 17 00:46:30 raspi kernel: WARNING: CPU: 0 PID: 388514 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x200/0x270 [videobuf2_c>
Jan 17 00:46:30 raspi kernel: Modules linked in: tun ntfs3 sg rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash aes_arm64 aes_generic a>
Jan 17 00:46:30 raspi kernel: CPU: 0 PID: 388514 Comm: V4L2DecoderThre Tainted: G         C         6.1.0-rpi7-rpi-v8 #1  Debian 1:6.1.63-1+rpt1
Jan 17 00:46:30 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Jan 17 00:46:30 raspi kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jan 17 00:46:30 raspi kernel: pc : __vb2_queue_cancel+0x200/0x270 [videobuf2_common]
Jan 17 00:46:30 raspi kernel: lr : __vb2_queue_cancel+0x38/0x270 [videobuf2_common]
Jan 17 00:46:30 raspi kernel: sp : ffffffc009f03af0
Jan 17 00:46:30 raspi kernel: x29: ffffffc009f03af0 x28: ffffffe6b77f3d18 x27: ffffff805ac54000
Jan 17 00:46:30 raspi kernel: x26: 0000000000000000 x25: 0000000000000000 x24: ffffff80607b7600
Jan 17 00:46:30 raspi kernel: x23: ffffff8045542d30 x22: ffffff80570b9a98 x21: ffffff80570b9a98
Jan 17 00:46:30 raspi kernel: x20: 000000000000000a x19: ffffff80570b9a98 x18: 0000000000000003
Jan 17 00:46:30 raspi kernel: x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc009f035e0
Jan 17 00:46:30 raspi kernel: x14: 0000000000000004 x13: ffffff8041540028 x12: 0000000000000000
Jan 17 00:46:30 raspi kernel: x11: ffffff80535af560 x10: ffffff80535af478 x9 : ffffffe6bd8fcf0c
Jan 17 00:46:30 raspi kernel: x8 : ffffffc009f03a00 x7 : 0000000000000000 x6 : 0000000000000228
Jan 17 00:46:30 raspi kernel: x5 : ffffffe6b77c02c4 x4 : ffffff805cf91400 x3 : 000000008010000a
Jan 17 00:46:30 raspi kernel: x2 : fffffffe0173e460 x1 : 0000000000000000 x0 : 0000000000000006
Jan 17 00:46:30 raspi kernel: Call trace:
Jan 17 00:46:30 raspi kernel:  __vb2_queue_cancel+0x200/0x270 [videobuf2_common]
Jan 17 00:46:30 raspi kernel:  vb2_core_streamoff+0x2c/0xd0 [videobuf2_common]
Jan 17 00:46:30 raspi kernel:  vb2_streamoff+0x20/0x70 [videobuf2_v4l2]
Jan 17 00:46:30 raspi kernel:  v4l2_m2m_streamoff+0x50/0x154 [v4l2_mem2mem]
Jan 17 00:46:30 raspi kernel:  v4l2_m2m_ioctl_streamoff+0x20/0x2c [v4l2_mem2mem]
Jan 17 00:46:30 raspi kernel:  v4l_streamoff+0x2c/0x40 [videodev]
Jan 17 00:46:30 raspi kernel:  __video_do_ioctl+0x174/0x3f0 [videodev]
Jan 17 00:46:30 raspi kernel:  video_usercopy+0x20c/0x77c [videodev]
Jan 17 00:46:30 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
Jan 17 00:46:30 raspi kernel:  v4l2_ioctl+0x48/0x6c [videodev]
Jan 17 00:46:30 raspi kernel:  __arm64_sys_ioctl+0xb4/0xfc
Jan 17 00:46:30 raspi kernel:  invoke_syscall+0x50/0x120
Jan 17 00:46:30 raspi kernel:  el0_svc_common.constprop.0+0x68/0x124
Jan 17 00:46:30 raspi kernel:  do_el0_svc+0x34/0xd0
Jan 17 00:46:30 raspi kernel:  el0_svc+0x30/0x94
Jan 17 00:46:30 raspi kernel:  el0t_64_sync_handler+0xf4/0x120
Jan 17 00:46:30 raspi kernel:  el0t_64_sync+0x18c/0x190
Jan 17 00:46:30 raspi kernel: ---[ end trace 0000000000000000 ]---
Jan 17 00:47:05 raspi kernel: ------------[ cut here ]------------
Jan 17 00:47:05 raspi kernel: WARNING: CPU: 0 PID: 388514 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x200/0x270 [videobuf2_c>
Jan 17 00:47:05 raspi kernel: Modules linked in: tun ntfs3 sg rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash aes_arm64 aes_generic a>
Jan 17 00:47:05 raspi kernel: CPU: 0 PID: 388514 Comm: V4L2DecoderThre Tainted: G        WC         6.1.0-rpi7-rpi-v8 #1  Debian 1:6.1.63-1+rpt1
Jan 17 00:47:05 raspi kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Jan 17 00:47:05 raspi kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jan 17 00:47:05 raspi kernel: pc : __vb2_queue_cancel+0x200/0x270 [videobuf2_common]
Jan 17 00:47:05 raspi kernel: lr : __vb2_queue_cancel+0x38/0x270 [videobuf2_common]
Jan 17 00:47:05 raspi kernel: sp : ffffffc009f03af0
Jan 17 00:47:05 raspi kernel: x29: ffffffc009f03af0 x28: ffffffe6b77f3d18 x27: ffffff805ac54000
[... 55 lines cut ...]
Jan 17 00:47:07 raspi kernel:  video_usercopy+0x20c/0x77c [videodev]
Jan 17 00:47:07 raspi kernel:  video_ioctl2+0x20/0x40 [videodev]
Jan 17 00:47:07 raspi kernel:  v4l2_ioctl+0x48/0x6c [videodev]
Jan 17 00:47:07 raspi kernel:  __arm64_sys_ioctl+0xb4/0xfc
Jan 17 00:47:07 raspi kernel:  invoke_syscall+0x50/0x120
Jan 17 00:47:07 raspi kernel:  el0_svc_common.constprop.0+0x68/0x124
Jan 17 00:47:07 raspi kernel:  do_el0_svc+0x34/0xd0
Jan 17 00:47:07 raspi kernel:  el0_svc+0x30/0x94
Jan 17 00:47:07 raspi kernel:  el0t_64_sync_handler+0xf4/0x120
Jan 17 00:47:07 raspi kernel:  el0t_64_sync+0x18c/0x190
Jan 17 00:47:07 raspi kernel: ---[ end trace 0000000000000000 ]---
Jan 17 00:48:12 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1526 (slots)
Jan 17 00:48:14 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 303104 bytes), total 32768 (slots), used 432 (slots)
Jan 17 00:48:14 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 462848 bytes), total 32768 (slots), used 70 (slots)
Jan 17 00:48:14 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 466944 bytes), total 32768 (slots), used 0 (slots)
Jan 17 00:48:14 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 303104 bytes), total 32768 (slots), used 432 (slots)
Jan 17 00:48:14 raspi kernel: vc4-drm gpu: swiotlb buffer is full (sz: 462848 bytes), total 32768 (slots), used 70 (slots)
```

After a long freeze the system resume normal operation, mouse also resume normal operation.

The only running process is chromium-browser with no opened tabs (start page only).
The issue happened when I move cursor over minimize button on the window title.

The same issue periodically happens with different scenarios.

lurch, 2024-01-17:
@popcornmix / @pelwell This should probably be transferred to our Linux repo?

popcornmix, 2024-01-17:
Maybe. My gut feel is the `__vb2_queue_cancel` WARN is not causing the mouse pointer freeze for 10-15 seconds, and is possibly harmless (@6by9 may be able to clarify).

The `vc4-drm gpu: swiotlb buffer is full` may be more likely related to the mouse hangs. @pelwell what causes that? Is it just a (some sort of) memory is full issue? In that case it may be more a bookworm issue than a kernel issue.

6by9, 2024-01-17:
> Maybe. My gut feel is the `__vb2_queue_cancel` WARN is not causing the mouse pointer freeze for 10-15 seconds, and is possibly harmless (@6by9 may be able to clarify).

Correct, it's videobuf2 complaining that the driver hasn't handed all the buffers back, but after warning it then cleans up on the driver's behalf.
Why just the start page kicks off a video decode is a little curious though. What do you use as your start page?