raspberrypi/bookworm-feedback #198: Can't initiate VNC connection to Login Screen, but can view Login Screen during VNC connection

rmitchellgit, 2024-01-23:
I know there are several issues open about using VNC with Bookworm, but the most recent updates seem to have made some progress on these issues and now I am experiencing a new issue that I haven't seen addresses on here.

This is on a Raspberry Pi 4 B with Bookworm (tested x32 and x64) and fully updated.
This happens when using Real VNC Viewer from a Mac (M1 processor, macOS 14.3) and Windows 10.

If the Pi is on the Login screen, Real VNC Viewer connects, but constantly disconnects and reconnects.  It disconnects so quickly that I can't even begin to type the password in to log in.

If the Pi is on the desktop, I can connect with Real VNC Viewer, then log out, and I stay connected to the VNC session on the log in screen without the constant disconnects.  If I end my VNC session and try to reconnect, then I have the same issue as above.

Tiger VNC does NOT have this issue.
So this may be more of a Real VNC Viewer issue than a Bookworm / Pi issue.  However, this is a change from the issues that Bookworm had originally.

Previous versions of Bookworm on the Pi would disconnect your VNC session completely if a user logged out, and would not connect at all if a user was not logged in.
See this linked thread, especially the linked comment.
https://github.com/raspberrypi/bookworm-feedback/issues/42#issuecomment-1755841385

So it is interesting that there has been progress on this issue since
- I can stay connected when logging out
- Real VNC Viewer is able to connect briefly to the log in screen

So I guess the real issue here is this:
Is there any reason that it would allow me to stay connected when I log out, but can't start a connection successfully if I am logged out?

any1, 2024-01-24:
What you're describing fits with the issue that sometimes RealVNC viewer fails to decode ZRLE encoded images generated by wayvnc. If you change the encoding to "raw" in the viewer's settings, the problem should go away but you'll be using a lot more bandwidth.

rmitchellgit, 2024-01-24:
That got me on the right track, but I'm not sure if that is completely accurate.

See this link:
https://help.realvnc.com/hc/en-us/articles/360002254618-VNC-Viewer-Parameter-Reference-

Under PreferredEncoding, it says "this parameter is ignored unless Quality is set to CUSTOM"

I set PreferredEncoding to Raw and left Quality set to Auto, and it did not work, as is expected.

I left Preferred Encoding set to Raw and set the Quality to High, and it did work.

I set Preferred Encoding back to ZRLE2 and set the Quality to High, and it did work.

So it seems that his has more to do with the Quality setting than the PreferredEncoding setting?
Unless Quality is somehow automatically changing the PreferredEncoding setting and it doesn't display that change in the properties?

any1, 2024-01-24:
Yes, I think you're right about it being the quality setting rather than the encoding. I've seen others report the same. So, it probably has something to do with the pixel format.