raspberrypi/bookworm-feedback #199: Java / Eclipse graphics broken on Raspberry Pi OS 5.1 (Deebian 12, bookworm) on Raspberry Pi 4B

nyholku, 2024-01-28:
I have a Java application where the basic Swing JButtons look strange, see below, this application works perfectly on bullseye.

[image]

To investigate this I tried to load the latest Eclipse to my bookworm installation but launching the installer the window just shows vertical line segments, see below:

[image]

To double check I made a clean install of latest Raspberry Pi OS (5.1) and re-downloaded latest Eclipse for AArch64 and it this is an easy way to replicate this problem.

My gut feeling is that this is a Raspberry Pi OS, specifically to do with the graphics drivers but I'm ready to report this to OpenJDK ro Eclipse communities if that seems advisable.

nyholku, 2024-01-28:
I just spend better part of morning installing Raspberry Pi OS versions and Eclipse.

As can be seen from below (those with links in violet/purple) I've tried eight of the thirteen 64bit version of Raspberry Pi OS.

Only the last one, 

`https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2023-12-06/
`

fails.

[image]

XECDesign, 2024-01-29:
Is there a quicker/simpler way to reproduce it than installing eclipse? A minimal testcase with instructions would be helpful.

nyholku, 2024-01-29:
At the moment I do not know of any more ways than those tow that I showed above.

If you install/run my app:

[EazyCNC-2.0.38 for Raspberry Pi (aarch64/Raspberry Pi OS)](http://www.eazycnc.com/downloads/EazyCNC-2.0.38-aarch64.tar.gz)

this should show the problem pictured in the first pic above.

The second image is from running Eclipse installer, so you do not actually have to install Eclipse, just try to install it.

Both methods require downloading some code (mine is smaller) and running the code.

My hunch is that a simple a  JFrame and JButton could reproduce the problem, what I have in my app istandard Swing buttons there.

Some one suggested that this related to Wayland so I'm going to try X11.

As this is very tedious downloading, installing and testing on single SD card I am going to get some more SD cars so I can speed experimenting up.

XECDesign, 2024-01-29:
To save you time, I'm seeing it under X11 as well. I suspect the mesa update to be the most likely culprit.

nyholku, 2024-01-29:
Thank you, that was very nice of you. It does save time, I spent better part of yesterday experimenting and it gets old very fast. :)

XECDesign, 2024-01-29:
Doesn't seem to be mesa either. Issue happens with the older version as well. I'll keep digging.

XECDesign, 2024-01-29:
> Only the last one,
> 
> `https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2023-12-06/ `
> 
> fails.

Hm? I just tried a clean 2023-10-10 image and it also fails.

nyholku, 2024-01-29:
I can retest this as I get home from office, maybe I got my notes/versions mixed up, I did so many tests Yesterday and the long times that it takes to burn each image to an SD card does make it hard to concentrate on the task at hand.

nyholku, 2024-01-29:
Hi,
I just retested this and my app fails on 2023-10-10 so I must have made a mistake when I reported that it works for me.

Interesting, I also discussed this first here:

https://forums.raspberrypi.com/viewtopic.php?p=2187047#p2187047

and someone reported that Eclipse works for him/he with X11 on Pi5

XECDesign, 2024-02-07:
Having previously spent time looking at the many issues openjdk has, it always ends up being a rabbit hole I regret going down.

Closing for now, but if someone says "it worked in this bookworm image, but not this one" then I can take a look again and figure out who to forward it to. For now, I'll chalk it up to java being java.

closed 2024-02-07 by XECDesign

arsi-apli, 2024-05-14:
The problem is with Nimbus LaF:
[image]
With Metal LaF everything works without problems:
[image]

nesslinger-it, 2024-05-16:
> The problem is with Nimbus LaF: ![obrázok](https://private-user-images.githubusercontent.com/22594510/330339244-84b5250e-a81d-41b8-bc2b-dc8b0dc435c9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU4NzE5ODUsIm5iZiI6MTcxNTg3MTY4NSwicGF0aCI6Ii8yMjU5NDUxMC8zMzAzMzkyNDQtODRiNTI1MGUtYTgxZC00MWI4LWJjMmItZGM4YjBkYzQzNWM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE2VDE1MDEyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUxNzI4YzdmYzI5NWFmOGFlMzk0N2VhNGZjZDg2M2MwODIxYjExNjk5MmY1MDdhN2ZlM2E1ZmI2YjRhMDI5ZmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.p96he763TIMfO_9US6J3GHw8yKH_ao4tzaVdl4PlejQ) With Metal LaF everything works without problems: ![obrázok](https://private-user-images.githubusercontent.com/22594510/330339611-524d9ffb-4f84-4b8d-ae85-80135d29441e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU4NzE5ODUsIm5iZiI6MTcxNTg3MTY4NSwicGF0aCI6Ii8yMjU5NDUxMC8zMzAzMzk2MTEtNTI0ZDlmZmItNGY4NC00YjhkLWFlODUtODAxMzVkMjk0NDFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE2VDE1MDEyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3OTVlMTdmOWI4NGEzZTliMDRhYTY0NjBjMTFmYWYyMzEzN2M2M2QxOGEzOTFmY2FkM2RmMWYyZGQzZTcwNGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4LG1etFxSQ1zQxUHjyHsA8-6Z6C13jP8rHMk6R1rz8c)

Thanks for the hint. But it seems that the application needs the correct librarys to do this change.  If i run for example 
`java -Dswing.defaultlaf=javax.swing.plaf.metal.metallookandfeel MyApp`
 than i get some errors that the metal look and feel could not be loaded. So maybe this solutions does not work with all Java Applications. 

If i add this simple line in my code
`UIManager.setLookAndFeel(new MetalLookAndFeel());`
before the initComponents() than all looks very well!
Tested it with Raspi 400 and the latest Ubuntu 22.04 LTS today.

Thank you very much!