raspberrypi/bookworm-feedback #201: E: Sub-process /usr/bin/dpkg returned an error code (1)

seamusdemora, 2024-01-28:
### Describe the bug

dpkg: error processing package initramfs-tools (--configure):

### Steps to reproduce the behaviour

```
$ sudo apt update
$ sudo apt -y full-upgrade 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  code firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek libegl-mesa0 libgbm1 libgl1-mesa-dri libglapi-mesa
  libglx-mesa0 mesa-vulkan-drivers raspi-config rpi-eeprom
14 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 174 MB of archives.
After this operation, 1,866 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 libgl1-mesa-dri arm64 23.2.1-1~bpo12+rpt3 [6,967 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 libglx-mesa0 arm64 23.2.1-1~bpo12+rpt3 [146 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/main arm64 libegl-mesa0 arm64 23.2.1-1~bpo12+rpt3 [109 kB]
Get:4 http://archive.raspberrypi.com/debian bookworm/main arm64 libglapi-mesa arm64 23.2.1-1~bpo12+rpt3 [44.3 kB]
Get:5 http://archive.raspberrypi.com/debian bookworm/main arm64 libgbm1 arm64 23.2.1-1~bpo12+rpt3 [38.6 kB]
Get:6 http://archive.raspberrypi.com/debian bookworm/main arm64 code arm64 1.85.2-1705560689 [89.0 MB]
Get:7 http://archive.raspberrypi.com/debian bookworm/main arm64 firmware-atheros all 1:20230210-5+rpt3 [16.8 MB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main arm64 firmware-brcm80211 all 1:20230210-5+rpt3 [5,852 kB]
Get:9 http://archive.raspberrypi.com/debian bookworm/main arm64 firmware-libertas all 1:20230210-5+rpt3 [31.2 MB]
Get:10 http://archive.raspberrypi.com/debian bookworm/main arm64 firmware-misc-nonfree all 1:20230210-5+rpt3 [13.0 MB]
Get:11 http://archive.raspberrypi.com/debian bookworm/main arm64 firmware-realtek all 1:20230210-5+rpt3 [1,492 kB]
Get:12 http://archive.raspberrypi.com/debian bookworm/main arm64 mesa-vulkan-drivers arm64 23.2.1-1~bpo12+rpt3 [4,849 kB]
Get:13 http://archive.raspberrypi.com/debian bookworm/main arm64 raspi-config all 20240108 [29.7 kB]
Get:14 http://archive.raspberrypi.com/debian bookworm/main arm64 rpi-eeprom all 21.1-1 [4,651 kB]
Fetched 174 MB in 10s (18.1 MB/s)
apt-listchanges: Reading changelogs...
(Reading database ... 98592 files and directories currently installed.)
Preparing to unpack .../00-libgl1-mesa-dri_23.2.1-1~bpo12+rpt3_arm64.deb ... 

...  ending with:

Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi7-rpi-v8
raspi-firmware: missing /boot/firmware, did you forget to mount it?
run-parts: /etc/initramfs/post-update.d//z50-raspi-firmware exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
$
```

### Device (s)

Raspberry Pi 5

### System

System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

[... 1119 lines cut ...]
[    3.268786] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[    3.272674] [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 1
[    3.276897] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    3.279024] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    3.280848] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    3.284726] vc4-drm axi:gpu: [drm] Cannot find any crtc or sizes
[    3.488927] Adding 102384k swap on /var/swap.  Priority:-2 extents:1 across:102384k SSFS
[    3.981882] Bluetooth: hci0: BCM: features 0x2f
[    3.983254] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    3.983258] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    3.983609] Bluetooth: hci0: BCM: Using default device address (m.m.m.m)
[    4.095935] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    4.095946] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    4.098390] pps pps0: new PPS source ptp0
[    4.099595] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    4.130927] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    4.137461] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.137470] Bluetooth: BNEP filters: protocol multicast
[    4.137478] Bluetooth: BNEP socket layer initialized
[    4.140402] Bluetooth: MGMT ver 1.22
[    4.151098] NET: Registered PF_ALG protocol family
[   10.034337] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   10.206455] Key type cifs.spnego registered
[   10.206466] Key type cifs.idmap registered
[   10.206879] CIFS: Attempting to mount \\SynologyNAS-1\rpi_share

popcornmix, 2024-01-28:
Transferred to bookworm feedback, as this doesn't look like a kernel issue.

XECDesign, 2024-01-29:
What's the output of `mountpoint /boot/firmware`?

Although the filesystems section of the report shows /dev/mmcblk0p1 being mounted on /boot/firmware, I'm not seeing it the dmesg output. And looking at the script that gives the error, the only way to hit the error seems to be if /boot/firmware is not mounted.

seamusdemora, 2024-01-30:
> What's the output of `mountpoint /boot/firmware`?
>
> Although the filesystems section of the report shows /dev/mmcblk0p1 being mounted on /boot/firmware, I'm not seeing it the dmesg output. And [...]

```bash
$ mountpoint /boot/firmware
/boot/firmware is a mountpoint
``` 

Back to you...

XECDesign, 2024-01-30:
The relevant part of the script:

```
firmware_dst="/boot/firmware"

if ischroot ; then
  true # chroot detected - skip mount point check
elif [ -e /usr/bin/systemd-detect-virt ] && systemd-detect-virt -q ; then
  true # virtualization detected - skip mount point check
elif [ "$(stat -f -c %T "${firmware_dst}")" = "nfs" ]; then
  true
elif ! mountpoint -q "${firmware_dst}" ; then
  echo "raspi-firmware: missing ${firmware_dst}, did you forget to mount it?" >&2
  exit 1
fi
```

The only way I can see it failing in your case is if it wasn't mounted at the time. There's not much to go on and I can't reproduce the issue.

seamusdemora, 2024-01-31:
@XECDesign 

Thanks for following up, but I'm convinced it wasn't a "mounting issue" that was responsible. This Pi runs nearly 24x7, but gets re-boots regularly. I would've seen "other symptoms" if `/boot/firmware` was not mounted. 

And on a positive note, the issue seems to have cleared itself (or rather `apt` cleared it). At least I've not seen any other issues.

closed 2024-01-31 by seamusdemora