raspberrypi/bookworm-feedback #203: update for wf-panel-pi held back

DirkUK, 2024-01-29:
result of update / full-upgrade today (64bit on both Pi5 and Pi400):
```
The following packages have been kept back:
  wf-panel-pi
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

```
trying to install it:
```
sudo apt install wf-panel-pi
.
.
The following packages have unmet dependencies:
 wf-panel-pi : Depends: libfm-gtk4 (>= 1.3.2-1+rpt12) but 1.3.2-1+rpt11 is to be installed
E: Unable to correct problems, you have held broken packages.

```
checking status for libfm-gtk4 seems to show that the correct version (1.3.2-1+rpt12) is not (yet) in the repo:
```
pi@pi5desktop:~$ apt policy libfm-gtk4
libfm-gtk4:
  Installed: 1.3.2-1+rpt11
  Candidate: 1.3.2-1+rpt11
  Version table:
 *** 1.3.2-1+rpt11 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     1.3.2-1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages

```

Joe-DN, 2024-01-30:
As expected, this is also breaking the pi-gen build tool.

```
git clone --branch arm64 https://github.com/RPI-Distro/pi-gen.git
cd pi-gen
./build.sh
```

results in the same error

```
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 wf-panel-pi : Depends: libfm-gtk4 (>= 1.3.2-1+rpt12) but 1.3.2-1+rpt11 is to be installed
E: Unable to correct problems, you have held broken packages.
```

ryan-james-smith, 2024-01-30:
Yes in the http://archive.raspberrypi.com/debian/dists/bookworm/main/binary-arm64/Packages
The section for wf-panel-pi has the rpt12 requirement:
Package: wf-panel-pi
Version: 0.59
Architecture: arm64
Maintainer: Simon Long <simon@raspberrypi.com>
Installed-Size: 2345
Depends: libatk1.0-0 (>= 1.12.4), libatkmm-1.6-1v5 (>= 2.28.3), libc6 (>= 2.34), libcairo2 (>= 1.2.4), libdbusmenu-gtk3-4 (>= 0.4.2), libfm-gtk4 (>= 1.3.2-1+rpt12), libfm4 (>= 0.1.99), libgcc-s1 (>= 3.0), libgdk-pixbuf-2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.74.0), libglibmm-2.4-1v5 (>= 2.66.5), libgtk-3-0 (>= 3.24), libgtk-layer-shell0 (>= 0.6.0), libgtkmm-3.0-1v5 (>= 3.24.7), libmenu-cache3 (>= 0.5.0), libnm0 (>= 1.24.0), libnma0 (>= 1.10.6-1+rpt1), libnotify4 (>= 0.7.0), libpackagekit-glib2-18 (>= 1.1.0), libpango-1.0-0 (>= 1.14.0), libpangocairo-1.0-0 (>= 1.14.0), libpangomm-1.4-1v5 (>= 2.46.3), libpulse0 (>= 0.99.1), libsecret-1-0 (>= 0.18), libsigc++-2.0-0v5 (>= 2.2.0), libstdc++6 (>= 11), libudev1 (>= 183), libwayland-client0 (>= 1.20.0), libwf-config1 (>= 0.7.1), wayfire, libfm-modules, eject, lxmenu-data, packagekit, raspi-config
Recommends: pulseaudio, pipewire, pipewire-pulse, pi-bluetooth, pulseaudio-module-bluetooth, libspa-0.2-bluetooth, lp-connection-editor
Homepage: <insert the upstream URL, if relevant>
Priority: optional
Section: unknown
Filename: pool/main/w/wf-panel-pi/wf-panel-pi_0.59_arm64.deb
Size: 411304
SHA256: c36265cca89d6db20c058c978a54801b0d5253daf66882510cb8294507cc0c8c
SHA1: 3b1dc7442278a33829a83d54f2d434356d8a50d6
MD5sum: 0fff0ac4381fff67d188d102968bbc6e
Description: Raspberry Pi desktop panel for Wayfire
 Raspberry Pi desktop panel based on wf-shell
 
 although the libfm-gtk4 package lists rpt11
 Package: libfm-gtk4
Source: libfm
Version: 1.3.2-1+rpt11
Architecture: arm64
Maintainer: Debian LXDE Maintainers <pkg-lxde-maintainers@lists.alioth.debian.org>
Installed-Size: 497
Depends: libatk1.0-0 (>= 1.12.4), libc6 (>= 2.29), libcairo2 (>= 1.10.0), libfm4 (= 1.3.2-1+rpt11), libgdk-pixbuf-2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.37.3), libgtk-3-0 (>= 3.21.5), libmenu-cache3 (>= 0.4.0), libpango-1.0-0 (>= 1.20.0), libfm-gtk-data (>= 1.2.0~)
Recommends: eject
Suggests: libfm-tools, nautilus-actions
Multi-Arch: same
Homepage: http://pcmanfm.sourceforge.net/
Priority: optional
Section: libs
Filename: pool/main/libf/libfm/libfm-gtk4_1.3.2-1+rpt11_arm64.deb
Size: 146228
SHA256: 35983feea744dad022c6e4262e29911b5c5a9e6fbb82e2bdb6f1297ab5f74d60
SHA1: f5811006650ec05a5bccd3c5314e57655922da15
MD5sum: a17a202334923bbf096748bdfe725c39
Description: file management support (GTK+ 2.0 GUI library)
 LibFM provides file management functions built on top of Glib/GIO
 giving a convenient higher-level API.
 .
 This package contains the GTK+ version 2.0 GUI.
 
 Not really of much use as the maintainer will already know this no doubt, but as I have to put the offending rpi aside until I can install rpt12 I was bored and felt like i needed to do something vaguely related.

by, 2024-01-30:
libfm-gtk4 arm64 1.3.2-1+rpt12 is now available on the feed.

closed 2024-01-30 by DirkUK