raspberrypi/bookworm-feedback #204: OS release information

mats-nk, 2024-01-29:
The info in `/etc/os-release` differs between Bullseye and Bookworm.

Many scripts uses `/etc/os-release` to detect what OS release is used and with Bookworm it is impossible to detect that it is a "Raspberry Pi OS" aka raspbian OS running.

It would be appreciated if Bookworm continue to be detected as "Raspberry Pi OS" or raspbian in the future.

Bullseye `cat /etc/os-release`

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

Bookworm `cat /etc/os-release`

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

mats-nk, 2024-01-30:
Suggested :
```
PRETTY_NAME="Raspberry Pi OS GNU/Linux 12 (bookworm)"
NAME="Raspberry Pi OS GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="https://www.raspberrypi.com/software/"
SUPPORT_URL="https://forums.raspberrypi.com/"
BUG_REPORT_URL="https://github.com/raspberrypi/bookworm-feedback/issues/"
```

XECDesign, 2024-01-30:
Duplicate of https://github.com/RPi-Distro/repo/issues/293

closed 2024-01-30 by XECDesign

lurch, 2024-02-06:
As the issue @XECDesign has linked to demonstrates, this is a 32-bit / 64-bit OS difference, not a Bullseye / Bookworm difference.

mats-nk, 2024-02-09:
It is not even a Bookworm/Bullseye issue it is a done wrong and done right issue and it is still a higher motivation to motivate to do wrong than to correct it. So sad.

lurch, 2024-02-13:
Well if it matters to you that much, I guess you'll have to just keep using the 32-bit versions of Raspberry Pi OS (based on upstream Raspbian) rather than using the 64-bit versions (based on upstream Debian). No need to be sad :smiley: