raspberrypi/bookworm-feedback #205: [regression] disconnect/connect HDMI cable leads to no sound and some kind of wayfire reload

qrp73, 2024-02-03:
Steps to reproduce:
1) Connect USB sound card
2) Open VLC player and start playing some music
3) Disconnect HDMI cable and connect it again

Expected result: music should continue play even with no display and after reconnecting HDMI all should works as before, the music play should continue

Actual result: the sound immediately stops when HDMI cable is disconnected and didn't restored even after HDMI cable reconnected again. The apps which playing sounds cannot play sound anymore until app restart

The same issue with apps which playing sound through OpenAL. 

Previously there was no such issue, all worked ok. It starts to happens with some apt update, but I cannot say how long ago it starts to happens, probably about week or one month ago...

spl237, 2024-02-06:
Try using the Advanced Options menu in raspi-config to switch the audio system to pulseaudio instead of pipewire - does that make a difference?

popcornmix, 2024-02-06:
I could reproduce this. Switching to pulseaudio does not have the issue (so may be a temporary workaround).
I could also observe this with a bluetooth device.

Looks like wireplumber (and then pipewire) is "stopped by signal" (not sure if this a crash, or deliberate shutdown based on the hdmi audio device disappearing).
```
Feb 06 15:04:12 pi5 wireplumber[1317]: stopped by signal: Terminated
Feb 06 15:04:12 pi5 systemd[823]: Stopping wireplumber.service -
Multimedia Service Session Manager...
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/ldac
Feb 06 15:04:12 pi5 wireplumber[1317]: disconnected from pipewire
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSink/aptx_hd
Feb 06 15:04:12 pi5 systemd[823]: Stopped wireplumber.service -
Multimedia Service Session Manager.
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/aptx_hd
Feb 06 15:04:12 pi5 systemd[823]: Stopping pipewire.service - PipeWire
Multimedia Service...
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSink/aptx
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/aptx
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSink/sbc
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/sbc
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSink/sbc_xq
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/sbc_xq
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/faststream
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/faststream_duplex
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSink/opus_05
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/opus_05
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Feb 06 15:04:12 pi5 bluetoothd[712]: Endpoint unregistered:
sender=:1.41 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Feb 06 15:04:12 pi5 xdg-desktop-por[1095]: Caught PipeWire error:
connection error
Feb 06 15:04:12 pi5 xdg-desktop-portal-wlr[1340]: 2024/02/06 15:04:12
[ERROR] - pipewire: fatal error event from core
Feb 06 15:04:12 pi5 systemd[823]: Stopped pipewire.service - PipeWire
Multimedia Service.
Feb 06 15:04:12 pi5 systemd[823]: xdg-desktop-portal-wlr.service: Main
process exited, code=exited, status=1/FAILURE
Feb 06 15:04:12 pi5 systemd[823]: xdg-desktop-portal-wlr.service:
Failed with result 'exit-code'.
```

qrp73, 2024-02-10:
> Try using the Advanced Options menu in raspi-config to switch the audio system to pulseaudio instead of pipewire - does that make a difference?

yes, when I switch to pulseaudio there is no such issue, but I need better control on sound card sample rate from user app to avoid unwanted resampling. In pipewire I just put all supported sample rate into /usr/share/pipewire/pipewire.conf as ```default.clock.allowed-rates = [ 44100, 48000, 96000, 192000, 384000 ]``` and it automatically switch sound card sample rate according to requested sample rate from app. For example If app requests 384 kHz, it switch sound card to 384 kHz, so there is no resampling with it's unwanted artifacts.

With pulseaudio I don't know how to achieve that. The only way that I find is to setup hardcoded sample rate in /etc/pulse/daemon.conf. But since I'm using different sample rate, it requires to reconfigure pulseaudio each time when I need to change sample rate in app...

Could you please fix pipewire service?

spl237, 2024-02-10:
> Could you please fix pipewire service?

We're trying to - the problem is not trivial.

apassu, 2024-02-19:
I have a very similar problem with quite the same logs.
I have a RPI 4 with an external HDMI monitor (with Alexa builts in the monitor) and when Alexa activates itself there is a crash of pipewire/wireplumber.

**SYSTEMS INFO:**

Raspberry PI 4
Raspbian 12 (bookworm)
Linux rpi 6.1.0-rpi8-rpi-v8 https://github.com/raspberrypi/linux/issues/1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
HDMI Monitor Samsung Smart Monitor M7 43" (with Alexa built in)
raspi-config -> 6 Advanced Options -> A7 Audio Config -> 2 Pipewire

**Steps to reproduce the behaviour**
The monitor has Alexa built in and when it activates itself there is a crash of pipewire/wireplumber.

**LOGS:**

2024-02-14T14:13:51.581663+01:00 rpi sh[4488]: Warning: ../src/main.c: 319: Selected output HDMI-A-1 went away
2024-02-14T14:13:51.585451+01:00 rpi sh[4488]: Warning: ../src/main.c: 330: No fallback outputs left. Detaching...
2024-02-14T14:13:52.479597+01:00 rpi wireplumber[5848]: stopped by signal: Terminated
2024-02-14T14:13:52.485894+01:00 rpi systemd[1750]: Stopping wireplumber.service - Multimedia Service Session Manager...
2024-02-14T14:13:52.537204+01:00 rpi wireplumber[5848]: disconnected from pipewire
2024-02-14T14:13:52.574138+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/ldac
2024-02-14T14:13:52.575006+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/aptx_hd
2024-02-14T14:13:52.576500+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_hd
2024-02-14T14:13:52.577591+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/aptx
2024-02-14T14:13:52.578691+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx
2024-02-14T14:13:52.580101+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/sbc
2024-02-14T14:13:52.581530+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/sbc
2024-02-14T14:13:52.582848+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/sbc_xq
2024-02-14T14:13:52.584948+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/sbc_xq
2024-02-14T14:13:52.586777+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_1
2024-02-14T14:13:52.588349+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_0
2024-02-14T14:13:52.589798+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
2024-02-14T14:13:52.591290+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
2024-02-14T14:13:52.594042+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/faststream
2024-02-14T14:13:52.596998+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/faststream_duplex
2024-02-14T14:13:52.600100+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/opus_05
2024-02-14T14:13:52.601406+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/opus_05
2024-02-14T14:13:52.603212+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSink/opus_05_duplex
2024-02-14T14:13:52.603778+01:00 rpi systemd[1750]: Stopped wireplumber.service - Multimedia Service Session Manager.
2024-02-14T14:13:52.604164+01:00 rpi systemd[1750]: wireplumber.service: Consumed 1.551s CPU time.
2024-02-14T14:13:52.605368+01:00 rpi bluetoothd[911]: Endpoint unregistered: sender=:1.92 path=/MediaEndpoint/A2DPSource/opus_05_duplex
2024-02-14T14:13:52.609849+01:00 rpi systemd[1750]: Stopping pipewire.service - PipeWire Multimedia Service...
2024-02-14T14:13:52.637246+01:00 rpi xdg-desktop-portal-wlr[5866]: 2024/02/14 14:13:52 [ERROR] - pipewire: fatal error event from core
2024-02-14T14:13:52.638465+01:00 rpi xdg-desktop-por[2270]: Caught PipeWire error: connection error
2024-02-14T14:13:52.651400+01:00 rpi systemd[1750]: Stopped pipewire.service - PipeWire Multimedia Service.
2024-02-14T14:13:52.651914+01:00 rpi systemd[1750]: pipewire.service: Consumed 2.980s CPU time.
2024-02-14T14:13:52.685947+01:00 rpi systemd[1750]: Started pipewire.service - PipeWire Multimedia Service.
2024-02-14T14:13:52.692411+01:00 rpi systemd[1750]: Started wireplumber.service - Multimedia Service Session Manager.
2024-02-14T14:13:52.693201+01:00 rpi systemd[1750]: xdg-desktop-portal-wlr.service: Main process exited, code=exited, status=1/FAILURE
2024-02-14T14:13:52.693686+01:00 rpi systemd[1750]: xdg-desktop-portal-wlr.service: Failed with result 'exit-code'.
2024-02-14T14:13:52.880896+01:00 rpi systemd[1750]: xdg-desktop-portal-wlr.service: Scheduled restart job, restart counter is at 22.
2024-02-14T14:13:52.881560+01:00 rpi systemd[1750]: Stopped xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).
2024-02-14T14:13:52.913039+01:00 rpi rtkit-daemon[1336]: Successfully made thread 5946 of process 5938 owned by '1000' high priority at nice level -11.
2024-02-14T14:13:52.914130+01:00 rpi rtkit-daemon[1336]: Supervising 1 threads of 1 processes of 1 users.
[... cut ...]
2024-02-19T21:48:47.723053+01:00 rpi systemd[1711]: Started xdg-desktop-portal-wlr.service - Portal service (wlroots implementation).

I noticed that, even if I switched to Pulse Audio, the pipewire service is enabled and active:

$  systemctl list-units --user | grep wire
  pipewire.service                                                                                     loaded active running   PipeWire Multimedia Service
  pipewire.socket                                                                                      loaded active running   PipeWire Multimedia System Socket