raspberrypi/bookworm-feedback #206: unexpected wayfire desktop logout with closing all apps and opening login screen

qrp73, 2024-02-07:
With latest updates it started to happens very often.
The issue is that system unexpectedly log out and closing all opened apps. The free memory is present, so the root of cause is unknown. Often OpenGL app are running at that moment.

Here is log file:
```
Feb 07 03:15:33 raspi xdg-desktop-por[1350]: Error reading events from display: Broken pipe
Feb 07 03:15:33 raspi polkitd[676]: Unregistered Authentication Agent for unix-session:1 (system bus name :1.31, object path /org/mate/PolicyKit1/Authenticat>
Feb 07 03:15:33 raspi systemd[1037]: xdg-desktop-portal-gtk.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 03:15:33 raspi systemd[1037]: xdg-desktop-portal-gtk.service: Failed with result 'exit-code'.
Feb 07 03:15:33 raspi lightdm[1030]: pam_unix(lightdm-autologin:session): session closed for user pi
Feb 07 03:15:33 raspi systemd-logind[720]: Session 1 logged out. Waiting for processes to exit.
```

AbhijithBn, 2024-02-15:
Hi

We encountered a similar issue where wayfire crashed and lightdm shows the login screen.

Here are the logs from /var/log/lightdm/lightdm.log.
PID 973 was of wayfire.

[+7049.89s] DEBUG: Session pid=973: Exited with return value 255
[+7049.89s] DEBUG: Seat seat0: Session stopped
[+7049.89s] DEBUG: Seat seat0: Stopping display server, no sessions require it
[+7049.89s] DEBUG: Seat seat0: Display server stopped
[+7049.89s] DEBUG: Seat seat0: Active display server stopped, starting greeter
[+7049.89s] DEBUG: Seat seat0: Creating greeter session
[+7050.25s] DEBUG: Seat seat0: Creating display server of type wayland
[+7050.25s] DEBUG: Using VT 8
[+7050.25s] DEBUG: Launching process 26208: /usr/share/dispsetup.sh
[+7050.25s] DEBUG: Process 26208 exited with return value 0
[+7050.25s] DEBUG: Seat seat0: Exit status of /usr/share/dispsetup.sh: 0
[+7050.25s] DEBUG: Seat seat0: Display server ready, starting session authentication
[+7050.26s] DEBUG: Session pid=26210: Started with service 'lightdm-greeter', username 'lightdm'
[+7050.26s] DEBUG: Releasing VT 7
[+7050.36s] DEBUG: Session pid=26210: Authentication complete with return value 0: Success
[+7050.36s] DEBUG: Seat seat0: Session authenticated, running command
[+7050.36s] DEBUG: Session pid=26210: Running command /usr/bin/wayfire-pi -c /usr/share/greeter.ini
[+7050.36s] DEBUG: Creating shared data directory /var/lib/lightdm/data/lightdm
[+7050.36s] DEBUG: Session pid=26210: Logging to /var/log/lightdm/seat0-greeter.log
[+7050.63s] DEBUG: Activating VT 8
[+7050.64s] DEBUG: Activating login1 session c1
[+7050.66s] DEBUG: Seat seat0 changes active session to c1
[+7050.66s] DEBUG: Session c1 is already active
[+7051.71s] DEBUG: Greeter connected version=1.26.0 api=1 resettable=false
[+7052.04s] DEBUG: Greeter start authentication for pi
[+7052.04s] DEBUG: Session pid=26413: Started with service 'lightdm', username 'pi'
[+7052.05s] DEBUG: Session pid=26413: Got 1 message(s) from PAM
[+7052.05s] DEBUG: Prompt greeter with 1 message(s)

sfatula, 2024-02-27:
Sounds like the same as my issue:

https://github.com/raspberrypi/bookworm-feedback/issues/184

AbhijithBn, 2024-04-02:
For our use case, it crashes after 30-40mins of running after boot.
Our application involves opening Chromium browser to view webpages so wayfire-pi returns this

```

II 02-04-24 12:09:53.242 - [src/main.cpp:280] Starting wayfire version 0.7.5
II 02-04-24 12:09:53.243 - [libseat] [libseat/backend/seatd.c:64] Could not connect to socket /run/seatd.sock: No such file or directory
II 02-04-24 12:09:53.243 - [libseat] [libseat/libseat.c:76] Backend 'seatd' failed to open seat, skipping
II 02-04-24 12:09:53.243 - [libseat] [libseat/libseat.c:76] Backend 'logind' failed to open seat, skipping
II 02-04-24 12:09:53.244 - [libseat] [seatd/seat.c:39] Created VT-bound seat seat0
II 02-04-24 12:09:53.245 - [libseat] [seatd/server.c:145] New client connected (pid: 10753, uid: 0, gid: 0)
II 02-04-24 12:09:53.245 - [libseat] [libseat/backend/seatd.c:633] Started embedded seatd
II 02-04-24 12:09:53.245 - [libseat] [seatd/seat.c:170] Added client 7 to seat0
II 02-04-24 12:09:53.245 - [libseat] [seatd/seat.c:480] Opened client 7 on seat0
II 02-04-24 12:09:53.245 - [libseat] [libseat/libseat.c:73] Seat opened with backend 'builtin'
II 02-04-24 12:09:53.246 - [libseat] [libseat/backend/seatd.c:212] Enabling seat
II 02-04-24 12:09:53.246 - [backend/session/session.c:109] Successfully loaded libseat session
EE 02-04-24 12:09:53.249 - [libseat] [seatd/seat.c:281] Could not make device fd drm master: Device or resource busy
II 02-04-24 12:09:53.281 - [backend/backend.c:220] Found 1 GPUs
II 02-04-24 12:09:53.281 - [backend/drm/backend.c:186] Initializing DRM backend for /dev/dri/card1 (vc4)
II 02-04-24 12:09:53.281 - [backend/drm/drm.c:242] Found 4 DRM CRTCs
II 02-04-24 12:09:53.281 - [backend/drm/drm.c:200] Found 24 DRM planes
II 02-04-24 12:09:53.284 - [render/egl.c:201] Supported EGL client extensions: EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_EXT_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless
EE 02-04-24 12:09:53.285 - [EGL] command: eglQueryDeviceStringEXT, error: EGL_BAD_PARAMETER (0x300c), message: "eglQueryDeviceStringEXT"
EE 02-04-24 12:09:53.285 - [EGL] command: eglQueryDeviceStringEXT, error: EGL_BAD_PARAMETER (0x300c), message: "eglQueryDeviceStringEXT"
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
II 02-04-24 12:09:53.573 - [render/egl.c:347] Using EGL 1.4
II 02-04-24 12:09:53.573 - [render/egl.c:348] Supported EGL display extensions: EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_context_flush_control EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display
II 02-04-24 12:09:53.573 - [render/egl.c:350] Supported EGL device extensions: EGL_EXT_device_drm
II 02-04-24 12:09:53.573 - [render/egl.c:352] EGL vendor: Mesa Project
II 02-04-24 12:09:53.575 - [render/gles2/renderer.c:727] Creating GLES2 renderer
II 02-04-24 12:09:53.575 - [render/gles2/renderer.c:728] Using OpenGL ES 3.1 Mesa 23.2.1-1~bpo12+rpt3
II 02-04-24 12:09:53.575 - [render/gles2/renderer.c:729] GL vendor: Broadcom
II 02-04-24 12:09:53.575 - [render/gles2/renderer.c:730] GL renderer: V3D 7.1
II 02-04-24 12:09:53.575 - [render/gles2/renderer.c:731] Supported GLES2 extensions: GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_generate_mipmap_sRGB GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_col [... cut ...]
EE 02-04-24 12:09:53.584 - [src/main.cpp:144] Fatal error: Fatal error(SIGABRT)
EE 02-04-24 12:09:53.591 - #1  ?? ??:0
EE 02-04-24 12:09:53.718 - #2  __default_sa_restorer ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:67
EE 02-04-24 12:09:53.857 - #3  __pthread_kill_implementation ./nptl/pthread_kill.c:44
EE 02-04-24 12:09:53.982 - #4  __GI_raise ../sysdeps/posix/raise.c:27
/usr/bin/wayfire failed with exit status 255.
II 02-04-24 12:09:53.987 - [libseat] [seatd/client.c:471] Client disconnected
```

lurch, 2024-04-03:
@AbhijithBn Does this happen reliably? Is there a public website that we can use to try and reproduce this problem?

AbhijithBn, 2024-04-04:
Our product [piSignage](https://pisignage.com/) uses Chromium browser to display Webpages that the end user would like to display on Signage Boards and while doing so we encountered this error.

It is not limited to certain websites, it can be replicated by spawning 2 or more Chromium browser (Puppeteer Windows) causes this issue.

mikehooper, 2024-04-10:
I'm getting the same at least since Oct 10. Usually when closing a Python pygame window.

```
uname -a
Linux raspberrypi 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian

Apr 03 09:16:44 raspberrypi xdg-desktop-por[1322]: Error reading events from display: Broken pipe
Apr 04 09:25:54 raspberrypi xdg-desktop-por[1313]: Error reading events from display: Broken pipe
Apr 04 11:26:42 raspberrypi xdg-desktop-por[1317]: Error reading events from display: Broken pipe
Apr 04 12:03:29 raspberrypi xdg-desktop-por[1313]: Error reading events from display: Broken pipe
Apr 05 10:00:19 raspberrypi xdg-desktop-por[1314]: Error reading events from display: Broken pipe
Apr 05 10:40:06 raspberrypi xdg-desktop-por[1309]: Error reading events from display: Broken pipe
Apr 05 11:12:49 raspberrypi xdg-desktop-por[1311]: Error reading events from display: Broken pipe
Apr 05 11:17:48 raspberrypi xdg-desktop-por[1315]: Error reading events from display: Broken pipe
Apr 06 14:10:54 raspberrypi xdg-desktop-por[1313]: Error reading events from display: Broken pipe
Apr 09 10:25:21 raspberrypi xdg-desktop-por[1317]: Error reading events from display: Broken pipe
Apr 10 11:57:37 raspberrypi xdg-desktop-por[1320]: Error reading events from display: Broken pipe

Apr 10 11:57:31 raspberrypi rtkit-daemon[1048]: Successfully made thread 147463 of process 147457 owned by '1000' high priority at nice level -15.
Apr 10 11:57:31 raspberrypi rtkit-daemon[1048]: Supervising 4 threads of 3 processes of 1 users.
Apr 10 11:57:37 raspberrypi xdg-desktop-por[1320]: Error reading events from display: Broken pipe
Apr 10 11:57:37 raspberrypi polkitd[717]: Unregistered Authentication Agent for unix-session:1 (system bus name :1.33, object path /org/mate/PolicyKit1/AuthenticationAgent, locale en_GB.UTF-8) (disconnected from bus)
Apr 10 11:57:37 raspberrypi systemd[1006]: xdg-desktop-portal-gtk.service: Main process exited, code=exited, status=1/FAILURE
Apr 10 11:57:37 raspberrypi systemd[1006]: xdg-desktop-portal-gtk.service: Failed with result 'exit-code'.
Apr 10 11:57:37 raspberrypi pulseaudio[1036]: X11 I/O error handler called
Apr 10 11:57:37 raspberrypi pulseaudio[1036]: X11 I/O error exit handler called, preparing to tear down X11 modules
Apr 10 11:57:37 raspberrypi lightdm[988]: pam_unix(lightdm-autologin:session): session closed for user pi
Apr 10 11:57:38 raspberrypi systemd-logind[743]: Session 1 logged out. Waiting for processes to exit.

```

mikehooper, 2024-04-12:
I appear to be able to replicate the issue with the following Python/pygame code:

`import pygame,time

l = 0
while l<100:
    print(l)
    pygame.init()
    screen = pygame.display.set_mode((400,300))
    time.sleep(2)
    pygame.quit()
    l+=1`

closed 2024-04-25 by spl237

AbhijithBn, 2024-04-26:
Thank you for the fix