raspberrypi/bookworm-feedback #207: Overlay Filessytem is not working as expected in CM4 eMMC compared to RPi 4 SD Card

jsn9979, 2024-02-07:
In CM4, after enabling "Overlay File System" in raspi-config and reboot, the /etc/fstab shows:

```
cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=d07f1fbb-01  /boot           vfat    defaults,ro          0       2
PARTUUID=d07f1fbb-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```

The rootfs (/home for example) is still rw.

As for comparison in RPi 4 with SD card, the /etc/fstab shows:

```
cat /etc/fstab
#
#  This fstab is for overlayroot. The real one can be found at
#  /media/root-ro/etc/fstab
#  The original entry for '/' and other mounts have been updated to be placed
#  under /media/root-ro.
#  To permanently modify this (or any other file), you should change-root into
#  a writable view of the underlying filesystem using:
#      sudo overlayroot-chroot
#
proc /proc proc defaults 0 0 # overlayroot:fs-virtual
PARTUUID=f7948fd0-01 /boot/firmware vfat defaults,ro 0 2 # overlayroot:fs-unsupported
#PARTUUID=f7948fd0-02 /media/root-ro/ ext4 ro,defaults,noatime,noauto 0 1
/media/root-ro/ / overlay lowerdir=/media/root-ro/,upperdir=/media/root-rw/overlay/,workdir=/media/root-rw/overlay-workdir/_ 0 1
# a swapfile is not a swap partition, no line here
# use dphys-swapfile swap[on|off] for that
```

XECDesign, 2024-02-07:
fstab doesn't need to be modified in this case. Check the output of `mount` instead.

jsn9979, 2024-02-15:
> fstab doesn't need to be modified in this case. Check the output of `mount` instead.

the overlay is mounted at root.

```
Filesystem      Size  Used Avail Use% Mounted on

udev            3.5G     0  3.5G   0% /dev
tmpfs           764M  1.3M  763M   1% /run
overlay         3.8G  294M  3.5G   8% /
tmpfs           3.8G     0  3.8G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  255M   46M  210M  18% /boot
tmpfs           764M   20K  764M   1% /run/user/1000
```

jsn9979, 2024-02-15:
The overlay is mounted at root.

```
Filesystem      Size  Used Avail Use% Mounted on

udev            3.5G     0  3.5G   0% /dev
tmpfs           764M  1.3M  763M   1% /run
overlay         3.8G  294M  3.5G   8% /
tmpfs           3.8G     0  3.8G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  255M   46M  210M  18% /boot
tmpfs           764M   20K  764M   1% /run/user/1000
```

> fstab doesn't need to be modified in this case. Check the output of `mount` instead.

closed 2024-02-15 by jsn9979