raspberrypi/bookworm-feedback #209: broken font glyphs on bless app window

qrp73, 2024-02-11:
Steps to reproduce:
1) install bless hex editor:
```sudo apt install bless```
2) open some text file with bless

Expected result: normal window content is displayed with hex content of the file

Actual result: broken glyphs are displayed:
[image]

lurch, 2024-02-12:
It looks like `bless` is a [Mono](https://www.mono-project.com/) application, so I wonder if this might be a "mono not properly supporting Wayland" issue, similar to how #199 ended up being a "java not properly supporting Wayland" issue?

qrp73, 2024-02-13:
Yes, it is Mono application, but as I see there is nothing in mono which cannot support Wayland. 
But it uses binding for GTK+, so it seems that the issue is due to [GTK+](https://www.gtk.org/) don't support Wayland.

spl237, 2024-02-14:
This should now be fixed if you update.

closed 2024-02-14 by spl237