raspberrypi/bookworm-feedback #213: WayVNC Repetition Issue When Connecting Over Internet

mghadam, 2024-02-19:
I am using RealVNC to connect to my Raspberry Pi "Bookworm WayVNC" in another country. There is a 200ms delay in the connection. When I try to press keys, I notice that WayVNC repeats some of them many times. For example, when I try to type "192", I see it appearing like "a922222222222". If I clear and try to enter it again, it then happens like "aaaaaaa92".

any1, 2024-02-21:
That looks like network jitter, perhaps due to packet loss. If the packet containing the key-release event gets dropped, it will be retransmitted and this will cause it to be delayed, perhaps enough to trigger the repeat.

This actually wouldn't be a problem if repeat events happened on the client side and not the server side. I'll look into automatically turning off repeat for virtual keyboards, but in the meantime, take a look at `kb_repeat_rate` here: https://github.com/WayfireWM/wayfire-wiki/blob/master/Core-options.md

I think that if you set it to 0, your problem will go away.

mghadam, 2024-02-21:
Thank you, this is resolved now!

closed 2024-02-21 by mghadam

mousseq, 2024-03-27:
I have the same issue with Pi OS Bookworm. I run VNC (remmina) on a Ubuntu desktop (mantic) and the Pi OS on a Pi 4. Even over my local LAN, I see spurious key repetition. This occurs when I use my mouse to switch focus between windows. The mouse click results in endless 9s appearing until I interrupt the process. This cause headaches by injecting the characters into editor and debugger windows. If I turn off key repeat as suggested above, the problem does indeed disappear, but the cost is have no key repeat at all. This is almost as much a problem. Is there no alternative to this? Or is there a fix to be implemented in Wayland to address this? Clearly X didn't suffer the problem so it must be possible to change this behavior.

any1, 2024-03-27:
See https://gitlab.freedesktop.org/wayland/wayland/-/merge_requests/368