raspberrypi/bookworm-feedback #214: Official 7-inch touchscreen + Pi

RayXxxxxx, 2024-02-20:
When the official seven-inch screen +x4 closes the browser, the desktop crashes and enters the login interface.

RayXxxxxx, 2024-02-20:
Kernel Version:  6.1.58-rpi7-rpi-2712

Logs:
[+0.00s] DEBUG: Logging to /var/log/lightdm/lightdm.log
[+0.00s] DEBUG: Starting Light Display Manager 1.26.0, UID=0 PID=3337
[+0.00s] DEBUG: Loading configuration dirs from /usr/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /usr/share/lightdm/lightdm.conf.d/01_debian.conf
[+0.00s] DEBUG: Loading configuration dirs from /usr/local/share/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration dirs from /etc/xdg/lightdm/lightdm.conf.d
[+0.00s] DEBUG: Loading configuration from /etc/lightdm/lightdm.conf
[+0.00s] DEBUG: Registered seat module local
[+0.00s] DEBUG: Registered seat module xremote
[+0.00s] DEBUG: Registered seat module unity
[+0.00s] DEBUG: Using D-Bus name org.freedesktop.DisplayManager
[+0.00s] DEBUG: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
[+0.00s] DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
[+0.01s] DEBUG: Monitoring logind for seats
[+0.01s] DEBUG: New seat added from logind: seat0
[+0.01s] DEBUG: Seat seat0: Loading properties from config section Seat:*
[+0.01s] DEBUG: Seat seat0 has property CanMultiSession=no
[+0.01s] DEBUG: Seat seat0: Starting
[+0.01s] DEBUG: Seat seat0: Creating user session
[+0.01s] DEBUG: Loading users from org.freedesktop.Accounts
[+0.01s] DEBUG: User /org/freedesktop/Accounts/User1000 added
[+0.13s] DEBUG: Seat seat0: Creating display server of type wayland
[+0.13s] DEBUG: Using VT 7
[+0.13s] DEBUG: Launching process 3402: /usr/share/dispsetup.sh
[+0.13s] DEBUG: Process 3402 exited with return value 0
[+0.13s] DEBUG: Seat seat0: Exit status of /usr/share/dispsetup.sh: 0
[+0.13s] DEBUG: Seat seat0: Display server ready, starting session authentication
[+0.13s] DEBUG: Session pid=3404: Started with service 'lightdm-autologin', username 'pi'
[+0.13s] DEBUG: Acquired bus name org.freedesktop.DisplayManager
[+0.13s] DEBUG: Registering seat with bus path /org/freedesktop/DisplayManager/Seat0
[+0.15s] DEBUG: Session pid=3404: Authentication complete with return value 0: Success
[+0.15s] DEBUG: Seat seat0: Session authenticated, running command
[+0.15s] DEBUG: Registering session with bus path /org/freedesktop/DisplayManager/Session0
[+0.15s] DEBUG: Session pid=3404: Running command /etc/X11/Xsession /usr/bin/wayfire-pi
[+0.15s] DEBUG: Creating shared data directory /var/lib/lightdm/data/pi
[+0.15s] DEBUG: Session pid=3404: Logging to .xsession-errors
[+0.18s] DEBUG: Activating VT 7
[+0.19s] DEBUG: Activating login1 session 7
[+0.20s] DEBUG: Seat seat0 changes active session to 7
[+0.20s] DEBUG: Session 7 is already active

6by9, 2024-02-21:
Thanks for the report. This had been noted internally and is already being investigated.

spl237, 2024-02-26:
This should now be fixed if you update from apt.

closed 2024-02-26 by spl237

RayXxxxxx, 2024-02-27:
Thanks, problem solved!