raspberrypi/bookworm-feedback #215: Kiosk mode quitting fullscreen after TV turned on.

ivandolman, 2024-02-20:
I use kiosk mode on my RPI and it works great, but with Bookworm, every time I turn my TV on, it's not maximized, so I have to restart my RPI. Quite annoying, I've looked for a fix, but couldn't find one. I'm not sure whether this is a feature or a bug, it's very annoying nonetheless.

popcornmix, 2024-02-20:
Are you booting with the TV turned off, and later turning it on?
Or are you booting with TV turned on, everything is fine, you then turn TV off, and later on, and then have an issue?

ivandolman, 2024-02-22:
Second one, I'm booting with TV on, everything is fine, but when I turn of my TV and then turn it on again the problem appears.

lurch, 2024-02-26:
ping @nathan-contino in case this is related to https://github.com/raspberrypi/documentation/issues/3184

btoconnor, 2024-04-01:
I believe I am seeing the same problem.  I'm using `ddcutil` to turn the monitor off / on during certain circumstances.  Anytime the monitor is powered off and then on again, the chrome browser is no longer in kiosk mode.  I use the `--start-maximized` flag on my kiosk command, so the window is fullscreen, but it is not in kiosk mode.

Adding `vc4.force_hotplug=1` to my `/boot/firmware/cmdline.txt` file does appear to fix my problem.

Thanks!

thebilberry, 2024-05-11:
Same problem for me!
Unfortunately, the above mentioned adding of `vc4.force_hotplug=1` to `/boot/firmware/cmdline.txt` did **not** help. As soon as the **display is switched off** (I have observed this via VNC), the browser **leaves the full screen mode**. 
Any other helpful hints?

sjobbe, 2024-09-02:
This is a chromium issue. I have the same problem running a screen for my home automation. I turn on the screen in my technical room and chrome exits kiosk/full screen mode. Firefox does not do this but the problem with Firefox is that it crashes after a day or two displaying Grafana.

I'm left with two shitty options.

jjuzwik, 2024-10-28:
@jnormington This is also happening to me, just running chromium in kiosk mode, on a pi5, with a Vizio smartcast enabled tv. If I disconnect the HDMI cord, or if I power off the tv and back on, chromium loses fullscreen. The force hot plug mentioned above, made no difference to this behavior. 

Could I use your script to fix this? Maybe some guidance on how to implement it? Thx!

TilmanGuenther, 2024-12-23:
Extremely frustrating, I am having the same issue. I have tried everything (hotplug etc) and nothing works. 

Weirdly, this only happens after a certain period of time? I have a script which sends commands to the monitor over HDMI to turn it off and on (smart mirror) and if the monitor has been off for less than 10 or so minutes, no issues. 

When waiting longer, I suspect that the desktop environment is somehow killed, because on start it takes a bit longer than it should, and when it comes back, full screen is off and the mouse cursor is in the middle of the screen.

Is there a way that I can just completely delete this "helpful feature"? I want the output to be always outputting video, no matter if there is anything plugged in, no matter if the screen is on, no matter if 5000 years have gone by.

ImNoOSRS, 2025-06-02:
Anyone solve this yes? This bug is crazy annoying.