raspberrypi/bookworm-feedback #216: VLC changes aspect ratio instead of cropping properly

parandandrd, 2024-02-20:
Using a fully updated RPi OS 64-bit system as of 2/20/24, it appears that VLC scales video when selecting a cropped aspect ratio rather than actually cropping it as expected. This worked normally a few weeks ago (sorry I can't be more specific).

Switching to a different renderer (other than "Wayland dmabuf video output") restores normal cropping behavior but of course has associated layout issues.

**Steps to reproduce**

1. Open a video stream in VLC.
2. Change "Video > Crop" to something other than the default aspect ratio for the video being played.
3. The video is scaled to the selected aspect ratio rather than being cropped as expected.

6by9, 2024-02-20:
@jc-kynesim

jc-kynesim, 2024-02-20:
You are right! I'll fix it. I think I know where I must have broken it.

parandandrd, 2025-01-26:
This seems to be fixed now.

closed 2025-01-26 by parandandrd