raspberrypi/bookworm-feedback #217: Recent Apt Update package wayfire causes desktop to logout when HDMI input source is changed on my monitor

bit-bash, 2024-02-24:
On my RPi 5 running 64-bit bookworm desktop OS, there was a recent apt update that upgraded several packages.
After the upgrade, when I changed the input on my HDMI monitor from INP1 (RPi) to INP2 (PC) then back to INP1 (RPi) my desktop session ended and I was at the login screen.  Everything running in my session was lost.
I traced the issue down to the update package wayfire.
If I revert to the previous apt release of wayfire, the problem does not occur. The problem re-occurs when I re-install the update wayfire package.
Recent changes to wayfire from the changelog in the update are:
wayfire (0.7.5-1~bpo11+1~rpt32) bookworm; urgency=medium
  * Fix for crash when closing pixdecor window on touchscreen
 -- Simon Long <simon@raspberrypi.com>  Tue, 20 Feb 2024 14:07:02 +0000
wayfire (0.7.5-1~bpo11+1~rpt31) bookworm; urgency=medium
  * Andri headless changes
 -- Simon Long <simon@raspberrypi.com>  Mon, 05 Feb 2024 08:35:12 +0000
I'm not sure what my HDMI monitor is doing when I change the input source, but I suspect it's probably something to do with power saving modes. But, what ever my HDMI display is "saying" to the RPi is causing the RPi to end my desktop session and log me out.

spl237, 2024-02-26:
If you update to the latest version of wayfire now in apt, this should be fixed, but I do not understand why HDMI input changes should have caused the problem to occur - leaving this open until you can confirm that the patch fixes it.

spl237, 2024-02-27:
Ok, thanks for confirming!

closed 2024-02-27 by spl237

mgriego, 2024-06-22:
I know this issue is close, but I've noticed this happen to me as well even on the latest wayfire package (`0.7.5-1~bpo11+1~rpt35`), but it's VERY sporadic. I sometimes I turn my monitor off when I leave my Pi running. At least one time, when I came back and turned the monitor back on, wayfire ended my session. The logs indicated the session survived after turning the monitor off. It wasn't until the monitor turned back on that wayfire logged me out, so some signal between the monitor and the Pi on monitor power-on caused wayfire to log the session out. But like I said, this doesn't happen all the time. I now tend to just leave my monitor on so that it doesn't log me out.

Oh, one other thing that might be pertinent is that I use `swaylock` to lock the session when I step away using a launcher menu entry I created that calls `/usr/bin/swaylock -i /usr/share/rpd-wallpaper/aurora.jpg`.

pelkuk, 2024-07-11:
> On my RPi 5 running 64-bit bookworm desktop OS, there was a recent apt update that upgraded several packages. After the upgrade, when I changed the input on my HDMI monitor from INP1 (RPi) to INP2 (PC) then back to INP1 (RPi) my desktop session ended and I was at the login screen. Everything running in my session was lost. I traced the issue down to the update package wayfire. If I revert to the previous apt release of wayfire, the problem does not occur. The problem re-occurs when I re-install the update wayfire package. Recent changes to wayfire from the changelog in the update are: wayfire (0.7.5-1~bpo11+1~rpt32) bookworm; urgency=medium
> 
> * Fix for crash when closing pixdecor window on touchscreen
>   -- Simon Long [simon@raspberrypi.com](mailto:simon@raspberrypi.com)  Tue, 20 Feb 2024 14:07:02 +0000
>   wayfire (0.7.5-1~bpo11+1~rpt31) bookworm; urgency=medium
> * Andri headless changes
>   -- Simon Long [simon@raspberrypi.com](mailto:simon@raspberrypi.com)  Mon, 05 Feb 2024 08:35:12 +0000
>   I'm not sure what my HDMI monitor is doing when I change the input source, but I suspect it's probably something to do with power saving modes. But, what ever my HDMI display is "saying" to the RPi is causing the RPi to end my desktop session and log me out.

Hi! 
Exactly the same problem! Rpi400 newest version for today 64bit. When I switch my monitor from HDMI (Rpi) input to Display Port (PC) and then  back to hdmi, my  session ended and I was at the login screen. Everything running in my session was lost!
Exactly the same! :-( I have to return to 32 bit version.

pelkuk, 2024-07-12:
So I dont understand why the case is closed, if I face it today with newest upgrades

spl237, 2024-07-12:
> So I dont understand why the case is closed, if I face it today with newest upgrades

Because the issue which was causing the problem as previously reported was found and fixed, and those who had originally reported it confirmed that the problem had gone away.

So if you are still seeing a similar problem, either a) you do not actually have the fix, or b) what you are seeing is a different problem which happens to have the same symptoms, or c) it is specific to your hardware, such as your particular model of monitor.

To eliminate a), please download the latest OS image (4 July 2024) from https://www.raspberrypi.com/software/operating-systems/, flash it to a new SD card, and see if the problem still occurs.

pelkuk, 2024-07-14:
Dear Simon,
Thank you for your answer!

Once again I did exactly all as you wrote: installed with Raspberry Pi
Installer the latest (recommended!) 64 bit OS of 4 July 2024 with all
upgrades.
So the OS completely is up to date.

Raspberry PI 400.
Monitor - Philips 278E9Q

Then I turned off my PC which connected to the second monitor port -
Display Port.

GUI Screen Blanking - OFF

Then I simply turned off the monitor's power and after a while (about 10 -
15 minutes) turned it on again.
And my session was LOG OUT, I only saw an invitation to log in!

The most unpleasant thing in this situation is that this does not always
happen, but sporadically.
It seems user "mgriego" already wrote about a similar situation!

I am forced to use an older version on another SD card, which does not have
this bug, which is unacceptable for Raspberry permanent online work.
I hope we will fix it!

Sincerely yours
Sasha

spl237, 2024-07-14:
In that case, please try switching to the new labwc compositor to see if that fixes the problem.

Open a terminal, run raspi-config, and under Advanced Settings -> Wayland, choose option 3, labwc, then reboot.

pelkuk, 2024-07-15:
With labws  I haven't encountered our problem yet.
Maybe it's OK.
I tried with Screen Blanking OFF and with Screen Blanking ON.
And then I tried with
xset s 300 300
And I will continue testing for some time.

On the other hand, the monitor does not sleep neither after 300 seconds nor
after any other time.
Why can't OS just turn off the HDMI signal after a set period of
inactivity???
After all, this is exactly what people want and stumble upon problems like
ours in search of a solution.

spl237, 2024-07-15:
> And then I tried with xset s 300 300 

That's not going to work. xset, as the name suggests, is for X settings. It has no effect under Wayland.

The only way to blank the screen under labwc is by calling wlopm to power down the monitor, and not all monitors respond correctly to this.

pelkuk, 2024-07-15:
I really appreciate your quick responses.

Generally there is no need to power down the monitor.
When the monitor is turned on but there is no signal source,
most monitors will automatically enter sleep mode.

It seems to me (I am not an expert of RPI) there are no simple ways on
Raspberry Pi 400 put a monitor into sleep mode by just putting hdmi signal
down.
So like OS Windows does.

Ok, we could use screensavers. But it is not a sleep mode.

pelkuk, 2024-07-16:
Dear Simon,
I am happy to inform you that my dream of putting the monitor to sleep mode
has come true on the version listed below
(by disabling xscreensaver and entering some xset commands),
and therefore I do not see any point in using later versions with their
bugs and difficulties.

Thank you for your cooperation!

***@***.***:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
***@***.***:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
aarch64 GNU/Linux

***@***.***:~ $ xset q
Screen Saver:
  prefer blanking:  yes    allow exposures:  yes
  timeout:  300    cycle:  300
Colors:
  default colormap:  0x20    BlackPixel:  0x0    WhitePixel:  0xffffff
Font Path:
  /usr/share/fonts/X11/Type1,built-ins
DPMS (Energy Star):
  Standby: 300    Suspend: 300    Off: 300
  DPMS is Enabled
  Monitor is On