raspberrypi/bookworm-feedback #218: Right-click close Firefox browser causes OS logout

Rueful-Pi, 2024-02-24:
When I right-click Firefox browser tab and select close, Raspberry Pi OS goes black and enters login screen. Only happens for me on Firefox browser. Experienced issue using 64-bit Bookworm OS on Raspberry Pi 5 with non-touch regular monitor. I would provide logs, but I am beginner with Raspberry Pi and Linux so I don't know how to create or where to find the log for this issue. If instructed on how to, I will get them and post those logs here.

Rueful-Pi, 2024-02-24:
**Update:** Same thing happened when I right-clicked the tab for File Manager and selected close. I was browsing the Documents folder at the time with one instance of the Chromium browser open behind.

bit-bash, 2024-02-25:
This seems to be related to the same issue I reported here:   #217 
If you want to test this, I have attached the previous package version of wayfire for 64-bit bookwork release.
You can overwrite the "broken" wayfire package with:  **sudo tar -xf wayfire-prev-pkg.tar.gz -C /**
To restore the "broken" wayfire do this: **sudo apt reinstall wayfire**
I can reproduce this with right-clicking to close the File Manager with the borked wayfire but it doesn't break with the previous package.

[wayfire-prev-pkg.tar.gz](https://github.com/raspberrypi/bookworm-feedback/files/14395357/wayfire-prev-pkg.tar.gz)

Rueful-Pi, 2024-02-25:
@bit-bash Thanks for the reply. I will try out your fix, much appreciated.

I read your linked post, I have been experiencing my Pi kicking to login screen and closing session when I turn off my TV monitor, but not when I switch inputs, even happens if I turn off TV while I'm viewing the windows desktop with the monitor. I have a windows PC connected with DisplayPort, and the Pi with hdmi. 

Thanks again

Rueful-Pi, 2024-02-25:
@bit-bash Erika!!! The problem is solved! Yes, I think you are right the updated wayfire package must be causing this issue. Thanks so much. Right-clicking tabs to close seems to be a tough habit for me to break. lol I had lost progress on a few things I was working on, when I kept doing that to close tabs I was finished with.

spl237, 2024-02-26:
Should now be fixed if you update from apt.

closed 2024-02-26 by spl237