raspberrypi/bookworm-feedback #219: Wayland; Screen not turning off RPI 5 Bookworm

1liminal1, 2024-02-26:
Hi guys, 

I am trying to turn off a display for a Kiosk with a command 

vc_gencmd_read_response does not work with Wayland 

SO I tried ```WAYLAND_DISPLAY="wayland-1" wlr-randr --output HDMI-A-1 --off```

And I get this same issue where it just logs the user out. 

https://forums.raspberrypi.com/viewtopic.php?p=2196846&hilit=wlr+randr#p2196846

When can we get a fix for this?

Thanks

lurch, 2024-02-26:
I've just asked @spl237 about this, and he says that `wlr-randr` is the "wrong command" to be using to power screens off. He suggests setting a short DPMS time in `.config/wayfire.ini` instead, e.g.
```
[idle]
dpms_timeout = 10
```
will cause display to power-down after 10 seconds of inactivity.

1liminal1, 2024-02-26:
Thanks so much for that, but I want a command to turn off the screen. I want to do it based on presence passed from Home Assistant. 

Thanks

1liminal1, 2024-02-27:
Thats interesting, how would I turn the screen back on though?

Thanks :)

lurch, 2024-02-27:
I've just been testing on a Pi4 (running Wayfire) that I'm connected to over SSH - if I comment out the `dpms_timeout` value entirely, then the screen comes back on again :slightly_smiling_face:

1liminal1, 2024-02-28:
Awesome, So I have a script and MQTT all set up, just need to get home to test. 

Thanks for the ideas, ill let you know how I go :)

1liminal1, 2024-02-28:
No good :( It just logs me out whenever I add the ```dpms_timeout``` :( 

Have you seen that?

JamesH65, 2024-02-28:
Have you done a recent apt update? There was a recent fix for unexpected log outs that might be relevant.

1liminal1, 2024-02-29:
Thanks guys, I ditched native Raspian and went with DakOS. Maybe one day then can get it working :)

closed 2024-02-29 by 1liminal1