raspberrypi/bookworm-feedback #220: RPi 5 wifi AP-STA mode loses wlan0 connection after hours or days of correct operation

phylax2020, 2024-02-27:
I have a Raspberry Pi 5b running the latest bookworm version from Raspberry Pi Imager v1.8.4: Pi OS (64-bit) with Raspberry Pi Desktop. I followed the different steps to install the OS with WiFi on a USB memory stick. I also did update the OS with apt update and apt full-upgrade. The current OS version is:
Linux RaspberryPi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

Next I generated a virtual WLAN1 interface on wlan0 and turned off power save in /etc/rc.local:

```
iw wlan0 set power_save off

# enable AP-STA mode
iw dev wlan0 interface add wlan1 type __ap
sleep 2
iw wlan1 set power_save off

```
Next i nstalled an access point on WLAN1 with NetworkManger:

```
#!/bin/bash
# File: setup_pi-ager-ap.sh
# START NetworkManger commands to setup AccessPoint on wirtual wlan1
# wlan1 must be configured by : iw dev wlan0 interface add wlan1 type __ap

set -x
trap read debug
 # configure access point with networkmanager
 nmcli con delete PI_AGER_AP
 nmcli con add type wifi ifname wlan1 mode ap con-name PI_AGER_AP ssid pi-ager-ap
 nmcli con modify PI_AGER_AP 802-11-wireless.band bg
 nmcli con modify PI_AGER_AP 802-11-wireless.channel 6
 nmcli con modify PI_AGER_AP 802-11-wireless-security.key-mgmt wpa-psk
 nmcli con modify PI_AGER_AP 802-11-wireless-security.psk 12345678
 nmcli con modify PI_AGER_AP ipv4.addr 192.168.4.1/24
 nmcli con modify PI_AGER_AP ipv4.method shared
 nmcli con modify PI_AGER_AP ipv6.method disabled
 nmcli con up PI_AGER_AP

```
All works fine for few hours or up to several day. Then suddenly the network indicator on the desktop is grayed out and states,
that wlan0 is no langer available, the hotspot at IP: 192.168.4.1 is still active. The WLAN router (FRITZBOX 7490) has a strong and stable WLAN signal and is located nearby the RPi 5 and is still connected to other devices in the WLAN network.

Also the access point is still reachable by using smart phones or tablet computers. The log generated by journalctl shows the following errors, when the wlan0 connection disappears:

```
Feb 25 11:16:12 RaspberryPi NetworkManager[865]: <info>  [1708856172.6016] dhcp6 (wlan0): state changed no lease
Feb 25 11:16:12 RaspberryPi NetworkManager[865]: <info>  [1708856172.6016] dhcp6 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Feb 25 11:16:12 RaspberryPi NetworkManager[865]: <info>  [1708856172.6181] dhcp6 (wlan0): state changed new lease
Feb 25 11:16:12 RaspberryPi dbus-daemon[730]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=865 comm="/usr/sbin/NetworkManager --no-daemon")
Feb 25 11:16:12 RaspberryPi NetworkManager[865]: <info>  [1708856172.6224] dhcp4 (wlan0): state changed new lease, address=192.168.0.237
Feb 25 11:16:12 RaspberryPi systemd[1]: Starting NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service...
Feb 25 11:16:12 RaspberryPi dbus-daemon[730]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Feb 25 11:16:12 RaspberryPi systemd[1]: Started NetworkManager-dispatcher.service - Network Manager Script Dispatcher Service.
Feb 25 11:16:22 RaspberryPi systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Feb 25 11:17:01 RaspberryPi CRON[420065]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Feb 25 11:17:01 RaspberryPi CRON[420066]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Feb 25 11:17:01 RaspberryPi CRON[420065]: pam_unix(cron:session): session closed for user root
Feb 25 12:17:01 RaspberryPi CRON[424903]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Feb 25 12:17:01 RaspberryPi CRON[424904]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Feb 25 12:17:01 RaspberryPi CRON[424903]: pam_unix(cron:session): session closed for user root
Feb 25 12:49:28 RaspberryPi systemd[1]: Starting apt-daily.service - Daily apt download activities...
Feb 25 12:49:29 RaspberryPi systemd[1]: apt-daily.service: Deactivated successfully.
Feb 25 12:49:29 RaspberryPi systemd[1]: Finished apt-daily.service - Daily apt download activities.
Feb 25 13:17:01 RaspberryPi CRON[429790]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Feb 25 13:17:01 RaspberryPi CRON[429791]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Feb 25 13:17:01 RaspberryPi CRON[429790]: pam_unix(cron:session): session closed for user root
Feb 25 14:17:01 RaspberryPi CRON[434627]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Feb 25 14:17:01 RaspberryPi CRON[434628]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Feb 25 14:17:01 RaspberryPi CRON[434627]: pam_unix(cron:session): session closed for user root
Feb 25 14:32:56 RaspberryPi wpa_supplicant[866]: wlan0: CTRL-EVENT-DISCONNECTED bssid=c8:0e:14:27:6d:b5 reason=3
Feb 25 14:32:56 RaspberryPi wpa_supplicant[866]: Removed BSSID c8:0e:14:27:6d:b4 from ignore list (expired)
Feb 25 14:32:56 RaspberryPi NetworkManager[865]: <info>  [1708867976.5641] device (wlan0): supplicant interface state: completed -> disconnected
[... 109 lines cut ...]
Feb 27 00:06:04 pi-ager NetworkManager[800]: <warn>  [1708988764.1150] device (wlan0): Activation: (wifi) association took too long, failing activation
Feb 27 00:06:04 pi-ager NetworkManager[800]: <info>  [1708988764.1150] device (wlan0): state change: config -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Feb 27 00:06:04 pi-ager NetworkManager[800]: <info>  [1708988764.1153] manager: NetworkManager state is now CONNECTED_LOCAL
Feb 27 00:06:04 pi-ager NetworkManager[800]: <warn>  [1708988764.1155] device (wlan0): Activation: failed for connection 'FRITZ!Box 7490'
Feb 27 00:06:04 pi-ager NetworkManager[800]: <info>  [1708988764.1157] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Feb 27 00:06:04 pi-ager NetworkManager[800]: <info>  [1708988764.7474] device (wlan0): supplicant interface state: disconnected -> inactive
Feb 27 00:06:04 pi-ager NetworkManager[800]: <info>  [1708988764.7474] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> inactive

```
The device states with 'nmcli dev' is as follows:

```
DEVICE         TYPE      STATE                   CONNECTION 
wlan1          wifi      connected               PI_AGER_AP 
lo             loopback  connected (externally)  lo         
wlan0          wifi      disconnected         --         
p2p-dev-wlan0  wifi-p2p  disconnected         --         
p2p-dev-wlan1  wifi-p2p  disconnected         --         
eth0           ethernet  unavailable         --         

```
The problem persists until you restart NetworkManager.
If I set up the system with the 32Bit OS-Lite version of Bookworm in the same way and run it on a Pi zero 2W, there are no problems.

Does anyone have an idea of what might be causing this error and how to fix it?

phylax2020, 2024-03-12:
Further research into this issue yielded the following results:

If there is no traffic over WLAN for a long period of time, the system decides to interrupt the WLAN0 connection. The connection remains interrupted and is not reactivated automatically. The minimum WLAN idle time is exactly 12 hours. The only way to get out of this state is to restart the NetworkManager.

[image]

In order to make this effect visible, I used a bash script that periodically checks the status of the WLAN connection every 2 minutes via crontab. If the WLAN connection is interrupted, a message is written to the journal and the NetworkManager is restarted.
```

#!/bin/bash
# restart NetworkManager when wlan0 disappeared

if [ "$(nmcli -g GENERAL.STATE dev show wlan0)" = "30 (disconnected)" ]; then
	echo "wlan0 disconnected" | systemd-cat
	systemctl restart NetworkManager
fi
```

Does anyone know where in the OS there is a timeout of exactly 12 hours that leads to this problem? As already reported, the WLAN power management is switched off and is therefore not a possible cause.

fwoomer, 2024-04-03:
I'm experiencing the same problem.  

However, it seems that I don't have to restart NetworkManager.  I can just 

`sudo ifconfig wlan0 down
sudo ifconfig wlan0 up`

And everything comes back and works properly again...until next time.

lurch, 2024-04-04:
I know it's not quite what you're asking about (and I'm afraid I've never used AP-STA mode myself), but FYI https://www.raspberrypi.com/tutorials/host-a-hotel-wifi-hotspot/ suggests using a separate USB WiFi dongle in addition to the built-in WiFi interface.

phylax2020, 2024-04-05:
The commands:
sudo ifconfig wlan0 down 
followed by
sudo ifconfig wlan0 up
don't work, wlan0 keeps disconnected:

[image]

However, my workaround with restarting NetworkManager works perfectly.
The test was now also carried out with the following kernel version:

`Linux pi-ager 6.1.73-v8+ #1726 SMP PREEMPT Wed Jan 24 15:02:30 GMT 2024 aarch64 GNU/Linux`

I would be very interested to know whether the problem is in the Linux kernel or in modules in the kernel environment.

phylax2020, 2024-04-06:
There is an intersting thread that sheds some light on the problem with concurrent STA and AP mode on the RPi 5. The RPi 5 uses the Infinon CYW43455 wireless chip and in that thread is shown, when STA mode fails after a while outputting the same error messages seen on my Raspberry Pi 5 system. In summary, the firmware for this wireless chip is faulty:

https://community.infineon.com/t5/AIROC-Wi-Fi-and-Wi-Fi-Bluetooth/CYW43455-STA-mode-fails-for-DFS-enabled-channels-in-concurrent-STA-and-AP-mode/td-p/340650

Clownfish33, 2024-06-03:
Not just WIFI.  I have my pi5 connected to a router via Ethernet and get the same disconnect after a period and of inactivity.  I use the pi to run a remote observatory so this is very unacceptable as I now have to go to the observatory to restart the pi each night.  Looks like I need to revert back to a pi4.

Clownfish33, 2024-06-06:
> @Clownfish33 “Is it possible that your problem is with your router, or some kind of incompatibility between your router and your Pi?”

I solved the problem, and it wasn’t the network router or cables.  I removed my NVME hat and SSD card and put the normal SD card back in.  Voila the disconnect issue after a 12 hour idle went away.

My SSD card was an exact copy of the SD card.  I swapped them back again, and the problem returned.  It only occurred when I boot from the SSD card.

JamesH65, 2024-06-06:
A common culprit here would be power supply.  Although make sure your OS is up to date, there have been improvements in there too.

MarlinMr, 2024-09-03:
I seem to be having the same problem. Wlan0 disconnects.

absynth42, 2024-09-23:
+1, having the same issue with a Pi5 + Pineboard HatAI + Coral PCIe, latest bookworm. The Pi5 is in the same room as the Wifi AP and other Pis in the same room do not have these connection issues. I am using the original 27W power supply.

FrancYescO, 2024-10-12:
Same here on (2x) RPi3B and Fritz7590

absynth42, 2024-10-15:
This seems to point into an OS issue rather than a hardware issue, since it occurs on Pi3 up to 5. I have disabled avahi and that seemed to remove the issue for a couple of weeks, however it resurfaced last weekend. wlan0 was still up, but it had no IPv4 address anymore.

glossyio, 2024-10-22:
I wanted to +1 this issue.  

I have three (3) RPi5 / 4GB running headless across 3 different physical locations (different WiFi networks/routers with each one is using a different, official RPi5 PSU) and they each will inevitably lose WiFi connection after some unspecified amount of time.  I **do not think it's simply network inactivity**.  The router sees it as disconnected; however, the actual devices are still properly running and should be sending across very small data HTTP payloads every so often (sometimes may be many hours between payloads).

To try to further diagnose the issue, I had 1 unit connected to HDMI.  After it had been disconnected I looked at the RPi5 desktop and saw in the notification area the WiFi network symbol was still UP and connected, but it was reading a very low WiFi signal. I opened up a browser and it immediately opened a page and the WiFi signal read stronger. At this point it showed up again on the router! (see scree shot below of router)

Router showing lost connection then back online without a restart or any network commands:
[image]

If I leave the headless units alone long enough, **they will usually come back online at some point** (and may go down again) over the course of 1-2 weeks. However, my use case is running these as headless traffic monitors, so I would prefer they are not Inactive or Offline unless there is an actual issue.

glossyio, 2024-10-23:
> In the screenshot, it looks like you are using Unifi wifi APs (U6-LR). Are all of your networks Unifi-powered? I'm using Unifi too, maybe there's a common issue there?

I think it's unrelated to network infrastructure hardware. I have a U6-LR and an older UAP-nanoHD (one outside, one inside), and a RPi on each of them that both act the same way. Additionally, I have 1 RPi at another physical location connected to a Centurylink Fiber router (one of these [C4000s](https://www.centurylink.com/home/help/internet/modems-and-routers/greenwave-c4000.html)), and it acts the same way.

> I also found out that disabling avahi makes the issue FAR less common. The Pi5 will disconnect once every two weeks or somesuch, not after less than one day.

I am unfamiliar with avahi, but I will check into this!

I have a RPi5 sitting on the desk that I re-imaged earlier this week to test this exact issue (this is the one I have connected via HDMI). If I leave it idle a few hours, the network will no longer see it. If I physically connect the keyboard/mouse/monitor to it, it shows as still connected to the WiFi AP; however, opening a browser will _sometimes_ reconnect it to the router, but might also give DNS errors and/or timeouts.

For the test today with the logs below:

Oct 22 at 11:57 I physically connected to it and ran `raspi-config` to re-enter the WiFi credentials into the 'preconfigured' profile.  WiFi reconnected, I did a little browsing and then left the RPi5 idle. My router shows it subsequently, a couple hours later, **lost connection at Oct 22 at 13:45**.

Oct 23 at 9:54 I physically connected to it again and opened up a browser window to load raspberrypi.com that was previously cached, but other websites see a DNS error and then just Time Out.

I grabbed the log below, rebooted the RPi, and it connected just fine again. I am not really sure what I am looking at from the logs below but nothing stands out to me except that there are _no WLAN0 messages_ when the router marked it inactive on Oct 22 at 13:45, so did the RPi do anything with the wifi component at that time?

Here's the `journalctl | grep "wlan0"` output:
```
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7167] device (wlan0): Activation: starting connection 'preconfigured' (093f7c84-93b4-4f22-954d-e5f88af71c16)
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7168] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7175] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7179] device (wlan0): Activation: (wifi) access point 'preconfigured' has security, but secrets are required.
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7179] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7197] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7203] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7206] device (wlan0): Activation: (wifi) connection 'preconfigured' has security, and secrets exist.  No new secrets needed.
Oct 22 11:57:32 tm-p71 wpa_supplicant[1053]: wlan0: Trying to associate with SSID 'unifi-pn-3670'
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7369] device (wlan0): supplicant interface state: disconnected -> associating
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.7370] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> associating
Oct 22 11:57:32 tm-p71 wpa_supplicant[1053]: wlan0: Associated with 68:d7:9a:4c:13:3e
Oct 22 11:57:32 tm-p71 wpa_supplicant[1053]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Oct 22 11:57:32 tm-p71 wpa_supplicant[1053]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.8819] device (wlan0): supplicant interface state: associating -> 4way_handshake
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.8819] device (p2p-dev-wlan0): supplicant management interface state: associating -> 4way_handshake
Oct 22 11:57:32 tm-p71 wpa_supplicant[1053]: wlan0: WPA: Key negotiation completed with 68:d7:9a:4c:13:3e [PTK=CCMP GTK=CCMP]
Oct 22 11:57:32 tm-p71 wpa_supplicant[1053]: wlan0: CTRL-EVENT-CONNECTED - Connection to 68:d7:9a:4c:13:3e completed [id=0 id_str=]
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.8863] device (wlan0): supplicant interface state: 4way_handshake -> completed
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.8864] device (wlan0): Activation: (wifi) Stage 2 of 5 (Device Configure) successful. Connected to wireless network "unifi-pn-3670"
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.8864] device (p2p-dev-wlan0): supplicant management interface state: 4way_handshake -> completed
Oct 22 11:57:32 tm-p71 NetworkManager[1028]: <info>  [1729623452.8865] device (wlan0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
[... cut ...]
Oct 23 09:55:03 tm-p71 NetworkManager[1028]: <info>  [1729702503.2447] dhcp4 (wlan0): state changed no lease
Oct 23 09:55:03 tm-p71 NetworkManager[1028]: <info>  [1729702503.2448] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Oct 23 09:55:03 tm-p71 NetworkManager[1028]: <info>  [1729702503.2449] device (p2p-dev-wlan0): supplicant management interface state: 4way_handshake -> completed
Oct 23 09:55:03 tm-p71 NetworkManager[1028]: <info>  [1729702503.2512] dhcp4 (wlan0): state changed new lease, address=10.10.10.251
```

c0ri, 2024-11-23:
I am seeing this issue as well. I put in a script to restart the networking if it finds it's down.  Very disappointed in Pi5 after I waited so long to get one.

phylax2020, 2024-11-23:
After long waiting for a fix I upgraded my current test systems on RPi5b and RPi 4b to the latest Version 6.6.51 of PI-OS:

```
pi@RaspberryPi:~ $ uname -a
Linux RaspberryPi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
pi@RaspberryPi:~ $
```
Part of this OS version is a new firmware for the WLAN chip CYW43455 from Cypress, that is integrated on RPi5b and RPi4b :

```
pi@RaspberryPi:~ $ journalctl | grep BCM4345/6 | tail -1
Nov 14 23:04:58 RaspberryPi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
pi@RaspberryPi:~ $
```
With this firmware from Aug 29 2023 version 7.45.265  I can confirm, that both RPi5b and RPi4b ran 10 days continuously without problems. 
Many thanks to the developers who solved this problem 👍

ghollingworth, 2024-12-02:
@glossyio 

Your log seems to suggest it is regularly changing the domain from WORLD to US to GB.  What region have you selected in your WiFi settings?

glossyio, 2024-12-02:
> @glossyio
> 
> Your log seems to suggest it is regularly changing the domain from WORLD to US to GB. What region have you selected in your WiFi settings?

@ghollingworth - you know... I followed https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager to specify the WiFi credentials and country.  I am not sure why, and I feel a bit foolish now, but I did keep it on "GB" instead of using "US" for all my RPi images (3 devices now).  Are you thinking this may be the cause of the intermittent connectivity issue?  I live in Oregon, US.

> With this firmware from Aug 29 2023 version 7.45.265 I can confirm, that both RPi5b and RPi4b ran 10 days continuously without problems. Many thanks to the developers who solved this problem 👍

Also, this was not a fix for me. I have been running this wifi chip firmware version for at least several months without a change in behavior... I'm actually not sure when I last ran the `rpi-update` command.

ghollingworth, 2024-12-03:
The problem here is that we do not know what is the actual cause of it to stop working.  In your log you have a warning "solicit: failure sending router solicitation: No buffer space available" which feels like it might be a problem, although it doesn't align with the time your router lost connection to the device.

So, it might be worth seeing if setting the region to US to begin with fixes the problem (and checking the "sudo journalctl -u wpa_supplicant"  to make sure it is different...)

That will tell us it's related to that setting...

It looks like you AP specifically tells the device to be "US", which is why it is switching...  Mine doesn't do that I can set it to US and GB and the AP doesn't care (BT Home Hub crap!)

glossyio, 2024-12-04:
I really appreciate you taking a look at this, @ghollingworth .  I won't get to re-imaging this for a week or so, but I will use "US" for WiFi country and take note.  However, Based on the analysis of my logs below, I do not think that is the issue.

With my [wifi-drop script](https://github.com/glossyio/traffic-monitor/blob/bugfix/wifi-drops/utils/wlan_check.sh) that runs every 5-minutes (via cron) to detect network communication error: 
1. I see a communication error without any `wpa_supplicant` activity (for hours before it).  
2. My script then runs `systemctl restart NetworkManager`  and I do see `wpa_supplicant` log switching to `GB` and subsequently bouncing between `US` and `GB` within seconds a couple seconds.

Narrative for the logs below to illustrate network issues before `wpa_supplicant` log message:

1.  at **Dec 01 12:11** switched to `US` and stayed there. 
2. script detected communications error at **Dec 01 15:50** without any preceding `wpa_supplicant` logs
3. script resets the network and `wpa_supplicant` logs bounce between GB and US, lands on US 
4. 10-minutes  later, after 1 passed network communications check, at **12/01/24_16:00** script detects network error without any preceding `wpa_supplicant` log
5. script initiates a network reset
6. ... iterates a few more times over the next 15 minutes with errors and resets
7. as of now, there has been no network communications errors or `wpa_supplicant` logs for past 48-hours.

`sudo journalctl -u wpa_supplicant` log (contiguous to current date and time):

```
Nov 29 09:21:44 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-CONNECTED - Connection to 74:83:c2:dc:03:9c completed [id=0 id_str=]
Dec 01 12:11:34 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DISCONNECTED bssid=74:83:c2:dc:03:9c reason=3 locally_generated=1
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 12:11:35 buster-p02 wpa_supplicant[917]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Dec 01 12:11:36 buster-p02 wpa_supplicant[917]: nl80211: kernel reports: Registration to specific type not supported
Dec 01 12:11:42 buster-p02 wpa_supplicant[917]: wlan0: Trying to associate with SSID 'unifi-pn-3670'
Dec 01 12:11:42 buster-p02 wpa_supplicant[917]: wlan0: Associated with 68:d7:9a:4c:13:3e
Dec 01 12:11:42 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Dec 01 12:11:42 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
Dec 01 12:11:42 buster-p02 wpa_supplicant[917]: wlan0: WPA: Key negotiation completed with 68:d7:9a:4c:13:3e [PTK=CCMP GTK=CCMP]
Dec 01 12:11:42 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-CONNECTED - Connection to 68:d7:9a:4c:13:3e completed [id=0 id_str=]
Dec 01 15:51:33 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DISCONNECTED bssid=68:d7:9a:4c:13:3e reason=3 locally_generated=1
Dec 01 15:51:33 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 15:51:33 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Dec 01 15:51:33 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=GB
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: p2p-dev-wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Dec 01 15:51:34 buster-p02 wpa_supplicant[917]: nl80211: kernel reports: Registration to specific type not supported
Dec 01 15:51:40 buster-p02 wpa_supplicant[917]: wlan0: Trying to associate with SSID 'unifi-pn-3670'
Dec 01 15:51:40 buster-p02 wpa_supplicant[917]: wlan0: Associated with 74:83:c2:dc:03:9b
Dec 01 15:51:40 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Dec 01 15:51:40 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
Dec 01 15:51:40 buster-p02 wpa_supplicant[917]: wlan0: WPA: Key negotiation completed with 74:83:c2:dc:03:9b [PTK=CCMP GTK=CCMP]
Dec 01 15:51:40 buster-p02 wpa_supplicant[917]: wlan0: CTRL-EVENT-CONNECTED - Connection to 74:83:c2:dc:03:9b completed [id=0 id_str=]

[... End of log through Dec 03 17:00:00 ..]
```

[wifi-drop script](https://github.com/glossyio/traffic-monitor/blob/bugfix/wifi-drops/utils/wlan_check.sh) log (snipped redundant ping output, runs every 5 min, on error does 60-seconds sleep ping tests): 

```
12/01/24_15:45:01: ===============================================================================
12/01/24_15:45:01: Starting wlan_check
[... cut ...]
12/01/24_16:20:01: Starting wlan_check
12/01/24_16:20:01: ThingsBoard: duckduckgo.com
12/01/24_16:20:01: Gateway: 10.10.10.1
12/01/24_16:20:01: Nameserver: 10.10.10.1
12/01/24_16:20:01: Nameserver: 8.8.8.8
12/01/24_16:20:01: Nameserver: 1.1.1.1
12/01/24_16:20:01: Nameserver: 9.9.9.9
12/01/24_16:20:01: No Network Issues Detected
12/01/24_16:25:01: ===============================================================================

[... No network errors since then, through 12/03/24_17:00:00 ...] 
```

K-J-VV, 2024-12-05:
> > I wanted to +1 this issue.
> > I have three (3) RPi5 / 4GB running headless across 3 different physical locations (different WiFi networks/routers with each one is using a different, official RPi5 PSU) and they each will inevitably lose WiFi connection after some unspecified amount of time. I **do not think it's simply network inactivity**. The router sees it as disconnected; however, the actual devices are still properly running and should be sending across very small data HTTP payloads every so often (sometimes may be many hours between payloads).
> 
> In the screenshot, it looks like you are using Unifi wifi APs (U6-LR). Are all of your networks Unifi-powered? I'm using Unifi too, maybe there's a common issue there?
> 
> I also found out that disabling avahi makes the issue FAR less common. The Pi5 will disconnect once every two weeks or somesuch, not after less than one day.

I have the same issue @glossyio describes but I use a TP-Link/Omada network,

In my situation I have a single RPi5, it becomes "offline" after a day or several days. I have not ignored it long enough to test if it will come back online, a quick unplug/replug usually works but have not been able to find a better solution

My Pi boots and runs of NVME

brnhffmnn, 2024-12-12:
Adding my situation to the story line:
- brand new Rpi5, setup with Raspberry OS lite, pretty much vanilla installation without anything special
- booting from NVME
- original Raspberry 25W power supply
- about 24h after the first successful boot the wifi connect dropped, the last activity was about 12h before the drop

Restarting `NetworkManager` brings back the connection immediately.

craigerl, 2025-02-21:
I'll add a me-too, wifi access point is unreliable on 3A+, 4B, 5.   Clients disconnect after two minutes.  SSH'ing into access-point pi works, but if you run "top" or a command that generates a burst of traffic, it hangs.  netstat shows send-Q bytes piling up on the access-point pi.   Repeatedly requesting a small web page from the access point works for a dozen or so times, then hangs.  again, netstat on access-point pi shows send-q packets piling up on port 80.

None of this is a problem on the Pi Zero 2W.

craigerl, 2025-02-22:
In my case I had to add three lines below in bold to get the access point to work reliably on pi3/pi4/p5  (Zero2W doesn't need this tweak):

/etc/NetworkManager/system-connections/hotspot.nmconnection

[connection]
id=hotspot
uuid=dffb95a6-dedd-40d8-891e-098837c74d1b
type=wifi
interface-name=wlan0
timestamp=1740173461

[wifi]
band=bg
mode=ap
ssid=DigiPi

[wifi-security]
**group=ccmp;**
key-mgmt=wpa-psk
**pairwise=ccmp;**
**proto=rsn;**
psk=abcdefghij

[ipv4]
address1=10.0.0.5/24
method=shared

[ipv6]
addr-gen-mode=default
method=auto

[proxy]

ghollingworth, 2025-02-26:
Thanks...  We're still trying to identify anything common between your cases...

@craigerl   This issue is not about hotspot per se

@brnhffmnn 
What WiFi router do you have?
You say systemctl restart NetworkManager fixes the problem, what do you get if you do ```nmctl``` when it is down?

brnhffmnn, 2025-02-26:
I am running a FRITZ!Box 6660 Cable, Wi-Fi 6 on two bands (2,4GHz + 5GHz), single SSID, channel auto select, WPA2+WPA3.

After some short trial-and-error, I switched to wired connection and stopped bothering. But wifi is still active:

<details>
<summary>nmcli output</summary>

```
eth0: connected to Wired connection 1
        "eth0"
        ethernet (macb), 2C:CF:67:**:**:**, hw, mtu 1500
        ip4 default, ip6 default
        inet4 192.168.178.5/24
eth0: connected to Wired connection 1
        "eth0"
        ethernet (macb), 2C:CF:67:**:**:**, hw, mtu 1500
        ip4 default, ip6 default
        inet4 192.168.178.5/24
        route4 192.168.178.0/24 metric 100
        route4 default via 192.168.178.1 metric 100
        inet6 fdb7:****:****:****:468b/64
        inet6 2002:****:****:****:523f/64
        inet6 fe80::****:****:****:c37/64
        route6 fda2:****:****:****:1::/64 via fe80::****:****:****:d9ab metric 100
        route6 fdb7:****:****:****::/64 metric 100
        route6 fdb7:****::/64 via fe80::****:****:****:a9f0 metric 105
        route6 fe80::/64 metric 1024
        route6 2002:****:****::/64 metric 100
        route6 2002:****:****::/56 via fe80::****:****:****:a9f0 metric 100
        route6 default via fe80::****:****:****:a9f0 metric 100

br-7cbb1605a75c: connected (externally) to br-7cbb1605a75c
        "br-7cbb1605a75c"
        bridge, 02:42:50:25:5C:96, sw, mtu 1500
        inet4 172.18.0.1/16
        route4 172.18.0.0/16 metric 0
        inet6 fe80::****:****:****:5c96/64
        route6 fe80::/64 metric 256

lo: connected (externally) to lo
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536
        inet4 127.0.0.1/8
        inet6 ::1/128

wlan0: connected to preconfigured
        "Broadcom Wi-Fi"
        wifi (brcmfmac), 2C:CF:67:**:**:**, hw, mtu 1500
        inet4 192.168.178.64/24
        route4 192.168.178.0/24 metric 600
        route4 default via 192.168.178.1 metric 600
        inet6 fe80::****:****:****:8ae2/64
        route6 fe80::/64 metric 1024

docker0: connected (externally) to docker0
        "docker0"
        bridge, 02:42:27:61:8E:89, sw, mtu 1500
        inet4 172.17.0.1/16
        route4 172.17.0.0/16 metric 0

p2p-dev-wlan0: disconnected
        "p2p-dev-wlan0"
        wifi-p2p, hw

veth91686c8: unmanaged
        "veth91686c8"
        ethernet (veth), 22:F0:A9:57:43:DB, sw, mtu 1500

DNS configuration:
        servers: 192.168.178.1
        domains: fritz.box
        interface: eth0

        servers: fdb7:****:****:****:a9f0 2002:****:****:****:a9f0
        interface: eth0

        servers: 192.168.178.1
        domains: fritz.box
        interface: wlan0

        servers: fdb7:****:****:****:a9f0 2002:****:****:****:a9f0
        interface: wlan0
```
</details>

`nmcli device wifi` says:
```
IN-USE  BSSID              SSID  MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
*       1C:ED:6F:**:**:**  BWE   Infra  60    270 Mbit/s  89      ▂▄▆█  WPA2 WPA3
```

So it thinks it is connected, but the router does not show an active connection and the stats reveal that the last connection was on Feb 24th 12:24 am (two days ago).

ghollingworth, 2025-02-27:
What about if you ping the device from another device on the network?

ping 192.168.178.64

brnhffmnn, 2025-02-27:
The ping to the wifi IP is successful. *But* only if the wired connection is active, when I unplug the ethernet cable, pings timeout on *.64. Plugging the cable back in and pings work again.

I am aware how stupid this sounds 😆 

The rpi still shows an active wifi connection and the router only knows about the wired connection.

I rebooted. Now the wifi connection also shows up in the router. Ping works on the wifi IP -- even without wire.
And `nmcli device wifi` shows also all neighbouring wifi networks (see last comment: only my network showed up).

armaginus, 2025-02-28:
Just hopping into the conversation here. I've read through everything above and I'm having the EXACT same issues. Let me know if there's any useful information I can provide.

Running a Pi5
NVME SSD hat
OEM 27w Power Supply

Connected wirelessly

I do not have access to my router.

ghollingworth, 2025-03-03:
@brnhffmnn It looks like your router has got confused about the route to your Pi.  From your nmcli it says that the WiFi is *.64 and the ethernet is *.5

But when you have the ethernet plugged in it is routing the ping *.64 to the Ethernet instead.

In the situation where the WiFi is broken and you unplug the Ethernet, can you ping the router from the device through the WiFi?

ping 192.168.178.1

6by9, 2025-03-03:
IIRC There is a quirk with Linux's network stack where it will respond on the highest priority interface if you have two interfaces connected to the same network, regardless of which IP address you are talking to. (I recall wasting a fair amount of time because Wireshark on one interface was only seeing half the packets!)

Can you examine the ARP cache of your router to see what the MAC address it has assigned to the respective IP addresses?

tinue, 2025-03-10:
Just a "mee too". Devices are two Raspberry Pi, one Model 4, and one Model 5. Both use an SSD for booting, RPI4 uses an USB connected SSD, and the RPI5 uses an NVME connected SSD. Firmware, OS and and boot loader are up to date on both devices. Both run Raspberry Pi OS Lite, unmodified.

The main WLAN router is a Synology RT2600ac, but the Pis probably connect to the satellite AP, which is a Synology MR2200ac.

The Pi devices lose the IP address, but think the WLAN connection as such is still there. The main router does not see the Pi devices, though. I also see the ominous "no buffer space available" error. Shortly after the IP address is lost, apparently because the lease can't be renewed due to the previous error.

On the Pi 5 I have a keyboard and display connected, and usually recover with `wlan0 down/up`. The RPI4 is headless and needs to be recovered with a power cycle.

Some information:
nmcli:
```
IN-USE  BSSID              SSID       MODE   CHAN  RATE        SIGNAL  BARS  SECURITY
*       00:11:xx:xx:xx:xx  home       Infra  136   270 Mbit/s  67      ▂▄▆_  WPA2 WPA3
```

Log:
```
Mar 03 21:15:13 desktoppi wpa_supplicant[823]: wlan0: WPA: Group rekeying completed with 00:11:xx:xx:xx:xx [GTK=CCMP]
Mar 03 22:13:14 desktoppi avahi-daemon[751]: Withdrawing address record for fd9a:cbf5:3fdb:d74e:333f:175b:1a09:11e0 on wlan0.
Mar 03 22:13:14 desktoppi avahi-daemon[751]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fd9a:cbf5:3fdb:d74e:333f:175b:1a09:11e0.
Mar 03 22:13:14 desktoppi avahi-daemon[751]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::72c6:5b7b:45d2:481e.
Mar 03 22:13:14 desktoppi avahi-daemon[751]: Registering new address record for fe80::72c6:5b7b:45d2:481e on wlan0.*.
Mar 04 09:51:12 desktoppi NetworkManager[816]: <warn>  [1741078272.4738] ndisc[0x5555f2b062a0,"wlan0"]: solicit: failure sending router solicitation: No buffer space available (105)
Mar 04 11:28:40 desktoppi NetworkManager[816]: <info>  [1741084120.5506] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Mar 04 11:28:40 desktoppi NetworkManager[816]: <info>  [1741084120.5506] dhcp4 (wlan0): state changed no lease
Mar 04 11:28:41 desktoppi avahi-daemon[751]: Withdrawing address record for 192.168.243.130 on wlan0.
Mar 04 11:28:41 desktoppi avahi-daemon[751]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.243.130.
Mar 04 11:28:41 desktoppi avahi-daemon[751]: Interface wlan0.IPv4 no longer relevant for mDNS.
```

MrEAlderson, 2025-03-29:
That's what I run every minute using crontab, and I haven't had down reports from my monitoring service ever since. Replace the router's ip accordingly:
```bash
#!/bin/bash

if ! ping -c 3 -W 192.168.178.1 > /dev/null 2>&1; then
        echo "wlan0 disconnected"
        ip link set wlan0 down
        ip link set wlan0 up
fi`

ghollingworth, 2025-04-11:
Is the "no buffer space available" error common to everyone?

journalctl | grep "no buffer space available"

This is a generic error message in the kernel, but could help identify where the issue is being created...

glossyio, 2025-04-12:
> journalctl | grep "no buffer space available"

@ghollingworth -  I do not have a single instance of this in my 6-month history in journalctl.  In fact, when my disconnect happens, there is **nothing related preceding the event** in the journalctl.

Everything was fine for days, then network connectivity issue detected at April 10 18:15:01.  Here is the journalctl log leading up to and after that.  In fact, disconnects kept being detected until 19:50:00, but I only included up to 18:26:00:  https://0x0.st/8ptl.txt

Disconnected detected by my script (from above posts), in journalctl they appear as `glossyio : PWD=/home/glossyio ; USER=root ; COMMAND=/bin/bash /home/glossyio/code/utils/wlan_check.sh`:
```
04/10/25_18:10:01: ===============================================================================
04/10/25_18:10:01: Starting wlan_check
04/10/25_18:10:01: ThingsBoard: duckduckgo.com
04/10/25_18:10:01: Gateway: 10.10.10.1
04/10/25_18:10:01: Nameserver: 10.10.10.8
04/10/25_18:10:01: Nameserver: 8.8.8.8
04/10/25_18:10:01: Nameserver: 1.1.1.1
04/10/25_18:10:01: Nameserver: 9.9.9.9
04/10/25_18:10:01: No Network Issues Detected
04/10/25_18:15:01: ===============================================================================
04/10/25_18:15:01: Starting wlan_check
04/10/25_18:15:01: ThingsBoard: duckduckgo.com
04/10/25_18:15:11: ------------- host duckduckgo.com
04/10/25_18:15:11: ERROR: ;; communications error to 10.10.10.8#53: timed out
;; communications error to 10.10.10.8#53: timed out
;; no servers could be reached
04/10/25_18:15:17: Gateway: 10.10.10.1
04/10/25_18:15:17: Nameserver: 10.10.10.8
04/10/25_18:15:17: Nameserver: 8.8.8.8
04/10/25_18:15:17: Nameserver: 1.1.1.1
04/10/25_18:15:17: Nameserver: 9.9.9.9
04/10/25_18:15:17: ***** restarting network
04/10/25_18:15:17: IP addresses: 10.10.10.223 172.18.0.1 172.17.0.1
04/10/25_18:15:17: 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether d8:3a:dd:c0:6c:ef brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether d8:3a:dd:c0:6c:f0 brd ff:ff:ff:ff:ff:ff
4: br-540c850fe575: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default
    link/ether 02:42:af:19:01:85 brd ff:ff:ff:ff:ff:ff
5: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default
    link/ether 02:42:61:6f:f6:76 brd ff:ff:ff:ff:ff:ff
9: vethba86cb8@if8: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-540c850fe575 state UP mode DEFAULT group default
    link/ether 96:7f:6e:12:12:27 brd ff:ff:ff:ff:ff:ff link-netnsid 0
04/10/25_18:15:17: DEVICE           TYPE      STATE                   CONNECTION
wlan0            wifi      connected               preconfigured
br-540c850fe575  bridge    connected (externally)  br-540c850fe575
lo               loopback  connected (externally)  lo
docker0          bridge    connected (externally)  docker0
p2p-dev-wlan0    wifi-p2p  disconnected            --
eth0             ethernet  unavailable             --
vethba86cb8      ethernet  unmanaged               --
04/10/25_18:15:17: STATE      CONNECTIVITY  WIFI-HW  WIFI     WWAN-HW  WWAN
connected  full          enabled  enabled  missing  enabled
04/10/25_18:15:17: wlan0     IEEE 802.11  ESSID:"unifi-pn-3670"
          Mode:Managed  Frequency:5.18 GHz  Access Point: 74:83:C2:DC:03:9C
          Bit Rate=162 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=48/70  Signal level=-62 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:13  Invalid misc:0   Missed beacon:0
04/10/25_18:15:18:
04/10/25_18:20:01: ===============================================================================
04/10/25_18:20:01: Starting wlan_check

```

ericcj, 2025-04-15:
not sure it's the same problem but after seeing NetworkManager log supplicant management interface state: disconnected -> inactive I came upon this thread https://www.reddit.com/r/linux4noobs/comments/11am5rd/can_i_force_networkmanager_to_keep_trying_to/ indicating ssid's have limited retries and how to update it and connection.auth-retries to zero for infinite retrying