raspberrypi/bookworm-feedback #221: Systematic journal corruption on clean install of Bookworm Lite 64bit

demukti, 2024-02-29:
I've stumbled across this error in the journal after spending weeks building a project on top of this OS:

`system.journal corrupted or uncleanly shut down, renaming and replacing`

After some investigation I found that this error has been shown every boot from the very first reboot after installing with the Pi Imager.
So I tried a fresh install on a brand new SanDisk Extreme Pro and used another Pi 4B 1.5 (4GB). The error starts showing up immediately in every case. There have been no unclean shutdowns.

Switching debug on in cmdline.txt shows

`system.journal is from the future, refusing to append new data to it`

-     I've also tried rm /var/log/journal/* but the issue is always there (as expected since it's also there on a clean install).
-     The only thing that stops it is to turn persistent storage off but I don't want to do that.
-     I've also installed the Bullseye lite 64 version with the imager and the error is not there
-     I've tried with and without personal settings in the Imager and that makes no difference
-     ntp, fake-hw-clock, etc are all running as normal and I can't find other errors related to this
-     I've used google, chatgpt 4 and Bing co-pilot but there are no reports or fixes to be found anywhere.
-     Googling the error itself returns thousands of hits but most are about other issues or offer no solution. None are specifically about this exact OS that I can find and chatgpt has no clue.

The action that systemd reports that it will take is never done. The journal is not renamed and only one single journal file exists at all times. It is never renamed or restarted. Renaming and restarting it manually does nothing to stop the errors.

This should be very easy to replicate so I hope that some of the devs are willing to do that and help me fix this.

This should probably be a different post but just for reference; The reason I looked into this is that I get a sporadic reboot loop that seems to be journal or time related. There's no errors in any logs when this happens but the logging stops at the moment the time should start running (just before plymouth starts). So everything is the same time in the log (kernel, .. ) and right at the point where it should start adding seconds the log stops for every boot. Plymouth still shows and systemd is doing stuff but it's no longer logged and I suspect that errors in time functions are the cause. The loop is easily broken just by switching VT or connecting via SSH, otherwise it would continue indefinitely it seems.
I'm just adding this now in case anyone says the error is harmless and just to let it be.
The reboot issue is surely something in my setup because it wasn't there the first weeks I worked on this but I do believe it might be related to a time issue that is baked in at this point.

This is the debug log:
```

Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Time spent on flushing to /var/log/journal/48d069bf1c2340a18395b322c7d9419f is 52.788ms for 860 entries.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: System Journal (/var/log/journal/48d069bf1c2340a18395b322c7d9419f) is 56.0M, max 2.8G, 2.8G free.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: varlink: New incoming connection.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: varlink-18: Setting state idle-server
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: varlink-18: New incoming message: {"method":"io.systemd.Journal.FlushToVar","parameters":{}}
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: varlink-18: Changing state idle-server → processing-method
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Received client request to flush runtime journal.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Journal file /var/log/journal/48d069bf1c2340a18395b322c7d9419f/system.journal is from the future, refusing to append new data to it th>
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: File /var/log/journal/48d069bf1c2340a18395b322c7d9419f/system.journal corrupted or uncleanly shut down, renaming and replacing.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Fixed min_use=16.0M max_use=2.8G max_size=128.0M min_size=512.0K keep_free=1.4G n_max_files=100
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Reserving 333 entries in field hash table.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Reserving 233016 entries in data hash table.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Flushing to /var/log/journal/48d069bf1c2340a18395b322c7d9419f...
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Considering root directory '/run/log/journal'.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Root directory /run/log/journal added.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Considering directory '/run/log/journal/48d069bf1c2340a18395b322c7d9419f'.
Feb 28 16:26:11 tzvpi1 systemd-journald[304]: Directory /run/log/journal/48d069bf1c2340a18395b322c7d9419f added.
Feb 28 16:26:11 tzvpi1 systemd-tmpfiles[324]: Entry "/dev/kvm" matches include prefix "/dev".
```

Any help is greatly appreciated!
Thank you.

popcornmix, 2024-02-29:
To avoid duplicated answers, this question is also [url=https://forums.raspberrypi.com/viewtopic.php?t=366591]asked on forum[/url].

demukti, 2024-02-29:
Below is the output  journalctl | grep hwclock. 
It starts with normal logging and then a few times with debug logging on. This is all since I removed the log file yesterday and started fresh.
Specifically note that: 
`Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load`
has a date from over a month ago but is just listed in the chronologic order ignoring the datetime index.

```
Feb 28 01:13:48 tzvpi1 systemd[1]: Starting fake-hwclock.service - Restore / save the current clock...
Feb 28 01:13:48 tzvpi1 systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
Feb 28 01:13:48 tzvpi1 fake-hwclock[301]: Wed 28 Feb 00:13:48 UTC 2024
Feb 28 01:17:00 tzvpi1 systemd[1]: Stopping fake-hwclock.service - Restore / save the current clock...
Feb 28 01:17:00 tzvpi1 systemd[1]: fake-hwclock.service: Deactivated successfully.
Feb 28 01:17:00 tzvpi1 systemd[1]: Stopped fake-hwclock.service - Restore / save the current clock.
Feb 28 01:17:00 tzvpi1 systemd[1]: Starting fake-hwclock.service - Restore / save the current clock...
Feb 28 01:17:00 tzvpi1 systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
Feb 28 01:17:00 tzvpi1 fake-hwclock[300]: Wed 28 Feb 00:17:00 UTC 2024
Feb 28 01:19:05 tzvpi1 systemd[1]: Stopping fake-hwclock.service - Restore / save the current clock...
Feb 28 01:19:05 tzvpi1 systemd[1]: fake-hwclock.service: Deactivated successfully.
Feb 28 01:19:05 tzvpi1 systemd[1]: Stopped fake-hwclock.service - Restore / save the current clock.
Feb 28 01:19:05 tzvpi1 systemd[1]: Starting fake-hwclock.service - Restore / save the current clock...
Feb 28 01:19:05 tzvpi1 systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
Feb 28 01:19:05 tzvpi1 fake-hwclock[301]: Wed 28 Feb 00:19:05 UTC 2024
Feb 28 01:31:13 tzvpi1 systemd[1]: Stopping fake-hwclock.service - Restore / save the current clock...
Feb 28 01:31:13 tzvpi1 systemd[1]: fake-hwclock.service: Deactivated successfully.
Feb 28 01:31:13 tzvpi1 systemd[1]: Stopped fake-hwclock.service - Restore / save the current clock.
Feb 28 01:31:13 tzvpi1 systemd[1]: Starting fake-hwclock.service - Restore / save the current clock...
Feb 28 01:31:13 tzvpi1 systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
Feb 28 01:31:13 tzvpi1 fake-hwclock[299]: Wed 28 Feb 00:31:13 UTC 2024
Feb 28 16:17:22 tzvpi1 systemd[1]: Stopping fake-hwclock.service - Restore / save the current clock...
Feb 28 16:17:22 tzvpi1 systemd[1]: fake-hwclock.service: Deactivated successfully.
Feb 28 16:17:22 tzvpi1 systemd[1]: Stopped fake-hwclock.service - Restore / save the current clock.
Feb 28 16:17:22 tzvpi1 systemd[1]: Starting fake-hwclock.service - Restore / save the current clock...
Feb 28 16:17:22 tzvpi1 systemd[1]: Finished fake-hwclock.service - Restore / save the current clock.
Feb 28 16:17:22 tzvpi1 fake-hwclock[298]: Wed 28 Feb 15:17:22 UTC 2024
Feb 28 16:26:11 tzvpi1 systemd[1]: Stopping fake-hwclock.service - Restore / save the current clock...
Feb 28 16:26:11 tzvpi1 systemd[1]: fake-hwclock.service: Deactivated successfully.
Feb 28 16:26:11 tzvpi1 systemd[1]: Stopped fake-hwclock.service - Restore / save the current clock.
Feb 28 16:26:11 tzvpi1 systemd-sysv-generator[281]: Native unit for fake-hwclock.service already exists, skipping.
Feb 28 16:26:11 tzvpi1 systemd-sysv-generator[281]: Linked unit file: /lib/systemd/system/hwclock.service → /dev/null
Feb 28 16:26:11 tzvpi1 systemd-sysv-generator[281]: Native unit for hwclock.service already exists, skipping.
Feb 28 16:26:11 tzvpi1 systemd[1]: unit_file_build_name_map: normal unit file: /lib/systemd/system/fake-hwclock.service
Jan 26 22:48:37 tzvpi1 (-hwclock)[287]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Feb 28 16:26:11 tzvpi1 fake-hwclock[303]: Wed 28 Feb 15:26:11 UTC 2024
Feb 28 16:26:11 tzvpi1 systemd-udevd[328]: Reading rules file: /usr/lib/udev/rules.d/85-hwclock.rules
Feb 28 16:26:15 tzvpi1 systemd-udevd[328]: Reading rules file: /usr/lib/udev/rules.d/85-hwclock.rules
Feb 28 16:26:18 tzvpi1 systemd-udevd[328]: Reading rules file: /usr/lib/udev/rules.d/85-hwclock.rules
Feb 28 16:54:07 tzvpi1 systemd[1]: fake-hwclock.service: Installed new job fake-hwclock.service/stop as 770
Feb 28 16:54:07 tzvpi1 systemd[1]: Sent message type=signal sender=n/a destination=n/a path=/org/freedesktop/systemd1/unit/fake_2dhwclock_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=774 reply_cookie=0 signature=sa{sv}as error-name=n/a error-message=n/a
Feb 28 16:54:07 tzvpi1 systemd[1]: Sent message type=signal sender=n/a destination=n/a path=/org/freedesktop/systemd1/unit/fake_2dhwclock_2eservice interface=org.freedesktop.DBus.Properties member=PropertiesChanged cookie=775 reply_cookie=0 signature=sa{sv}as error-name=n/a error-message=n/a
Feb 28 16:54:07 tzvpi1 systemd[1]: fake-hwclock.service: stopping held back, waiting for: sysinit.target
Feb 28 16:54:08 tzvpi1 systemd-udevd[328]: Reading rules file: /usr/lib/udev/rules.d/85-hwclock.rules
Feb 28 16:54:09 tzvpi1 systemd[1]: fake-hwclock.service: Will spawn child (service_enter_stop): /sbin/fake-hwclock
Feb 28 16:54:09 tzvpi1 systemd[1]: fake-hwclock.service: About to execute /sbin/fake-hwclock save
Feb 28 16:54:09 tzvpi1 systemd[1]: fake-hwclock.service: Forked /sbin/fake-hwclock as 1132
Feb 28 16:54:09 tzvpi1 systemd[1]: fake-hwclock.service: Changed exited -> stop
[... cut ...]
Feb 28 17:20:21 tzvpi1 (-hwclock)[1089]: fake-hwclock.service: Executing: /sbin/fake-hwclock save
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Feb 28 21:00:30 tzvpi1 (-hwclock)[1411]: fake-hwclock.service: Executing: /sbin/fake-hwclock save
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Feb 29 14:04:29 tzvpi1 (-hwclock)[1270]: fake-hwclock.service: Executing: /sbin/fake-hwclock save
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
```

demukti, 2024-02-29:
This rogue date does not exist in the journal with using time arguments; the journal starts at Feb 28 but when using grep there's this:
```
$ journalctl | grep 'Jan 26'
Jan 26 22:48:37 tzvpi1 (mount)[284]: dev-mqueue.mount: Executing: /bin/mount mqueue /dev/mqueue -t mqueue -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (mount)[285]: sys-kernel-debug.mount: Executing: /bin/mount debugfs /sys/kernel/debug -t debugfs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (-hwclock)[287]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Jan 26 22:48:37 tzvpi1 (mount)[286]: sys-kernel-tracing.mount: Executing: /bin/mount tracefs /sys/kernel/tracing -t tracefs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (kmod)[290]: kmod-static-nodes.service: Executing: /bin/kmod static-nodes --format=tmpfiles --output=/run/tmpfiles.d/static-nodes.conf
Jan 26 22:48:37 tzvpi1 (modprobe)[292]: modprobe@configfs.service: Executing: /sbin/modprobe -abq configfs
Jan 26 22:48:37 tzvpi1 (setup.sh)[289]: keyboard-setup.service: Executing: /lib/console-setup/keyboard-setup.sh
Jan 26 22:48:37 tzvpi1 (modprobe)[294]: modprobe@dm_mod.service: Executing: /sbin/modprobe -abq dm_mod
Jan 26 22:48:37 tzvpi1 (modprobe)[295]: modprobe@drm.service: Executing: /sbin/modprobe -abq drm
Jan 26 22:48:37 tzvpi1 (modprobe)[297]: modprobe@efi_pstore.service: Executing: /sbin/modprobe -abq efi_pstore
Jan 26 22:48:37 tzvpi1 (modprobe)[299]: modprobe@fuse.service: Executing: /sbin/modprobe -abq fuse
Jan 26 22:48:37 tzvpi1 (modprobe)[300]: modprobe@loop.service: Executing: /sbin/modprobe -abq loop
Jan 26 22:48:37 tzvpi1 (mount)[284]: dev-mqueue.mount: Executing: /bin/mount mqueue /dev/mqueue -t mqueue -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (mount)[285]: sys-kernel-debug.mount: Executing: /bin/mount debugfs /sys/kernel/debug -t debugfs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (-hwclock)[287]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Jan 26 22:48:37 tzvpi1 (mount)[286]: sys-kernel-tracing.mount: Executing: /bin/mount tracefs /sys/kernel/tracing -t tracefs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (setup.sh)[289]: keyboard-setup.service: Executing: /lib/console-setup/keyboard-setup.sh
Jan 26 22:48:37 tzvpi1 (kmod)[290]: kmod-static-nodes.service: Executing: /bin/kmod static-nodes --format=tmpfiles --output=/run/tmpfiles.d/static-nodes.conf
Jan 26 22:48:37 tzvpi1 (modprobe)[292]: modprobe@configfs.service: Executing: /sbin/modprobe -abq configfs
Jan 26 22:48:37 tzvpi1 (modprobe)[293]: modprobe@dm_mod.service: Executing: /sbin/modprobe -abq dm_mod
Jan 26 22:48:37 tzvpi1 (modprobe)[295]: modprobe@drm.service: Executing: /sbin/modprobe -abq drm
Jan 26 22:48:37 tzvpi1 (modprobe)[297]: modprobe@efi_pstore.service: Executing: /sbin/modprobe -abq efi_pstore
Jan 26 22:48:37 tzvpi1 (modprobe)[298]: modprobe@fuse.service: Executing: /sbin/modprobe -abq fuse
Jan 26 22:48:37 tzvpi1 (modprobe)[299]: modprobe@loop.service: Executing: /sbin/modprobe -abq loop
Jan 26 22:48:37 tzvpi1 (mount)[282]: dev-mqueue.mount: Executing: /bin/mount mqueue /dev/mqueue -t mqueue -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (mount)[283]: sys-kernel-debug.mount: Executing: /bin/mount debugfs /sys/kernel/debug -t debugfs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (mount)[284]: sys-kernel-tracing.mount: Executing: /bin/mount tracefs /sys/kernel/tracing -t tracefs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Jan 26 22:48:37 tzvpi1 (kmod)[288]: kmod-static-nodes.service: Executing: /bin/kmod static-nodes --format=tmpfiles --output=/run/tmpfiles.d/static-nodes.conf
Jan 26 22:48:37 tzvpi1 (setup.sh)[286]: keyboard-setup.service: Executing: /lib/console-setup/keyboard-setup.sh
Jan 26 22:48:37 tzvpi1 (modprobe)[290]: modprobe@configfs.service: Executing: /sbin/modprobe -abq configfs
Jan 26 22:48:37 tzvpi1 (modprobe)[291]: modprobe@dm_mod.service: Executing: /sbin/modprobe -abq dm_mod
Jan 26 22:48:37 tzvpi1 (modprobe)[293]: modprobe@drm.service: Executing: /sbin/modprobe -abq drm
Jan 26 22:48:37 tzvpi1 (modprobe)[295]: modprobe@efi_pstore.service: Executing: /sbin/modprobe -abq efi_pstore
Jan 26 22:48:37 tzvpi1 (modprobe)[298]: modprobe@loop.service: Executing: /sbin/modprobe -abq loop
Jan 26 22:48:37 tzvpi1 (mount)[282]: dev-mqueue.mount: Executing: /bin/mount mqueue /dev/mqueue -t mqueue -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (mount)[283]: sys-kernel-debug.mount: Executing: /bin/mount debugfs /sys/kernel/debug -t debugfs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Jan 26 22:48:37 tzvpi1 (mount)[284]: sys-kernel-tracing.mount: Executing: /bin/mount tracefs /sys/kernel/tracing -t tracefs -o nosuid,nodev,noexec
Jan 26 22:48:37 tzvpi1 (setup.sh)[287]: keyboard-setup.service: Executing: /lib/console-setup/keyboard-setup.sh
Jan 26 22:48:37 tzvpi1 (modprobe)[290]: modprobe@configfs.service: Executing: /sbin/modprobe -abq configfs
Jan 26 22:48:37 tzvpi1 (kmod)[288]: kmod-static-nodes.service: Executing: /bin/kmod static-nodes --format=tmpfiles --output=/run/tmpfiles.d/static-nodes.conf
Jan 26 22:48:37 tzvpi1 (modprobe)[293]: modprobe@dm_mod.service: Executing: /sbin/modprobe -abq dm_mod
Jan 26 22:48:37 tzvpi1 (modprobe)[294]: modprobe@drm.service: Executing: /sbin/modprobe -abq drm
Jan 26 22:48:37 tzvpi1 (modprobe)[295]: modprobe@efi_pstore.service: Executing: /sbin/modprobe -abq efi_pstore
[... cut ...]
Jan 26 22:48:37 tzvpi1 (modprobe)[291]: modprobe@dm_mod.service: Executing: /sbin/modprobe -abq dm_mod
Jan 26 22:48:37 tzvpi1 (modprobe)[289]: modprobe@configfs.service: Executing: /sbin/modprobe -abq configfs
Jan 26 22:48:37 tzvpi1 (modprobe)[294]: modprobe@drm.service: Executing: /sbin/modprobe -abq drm
Jan 26 22:48:37 tzvpi1 (modprobe)[295]: modprobe@efi_pstore.service: Executing: /sbin/modprobe -abq efi_pstore
Jan 26 22:48:37 tzvpi1 (modprobe)[296]: modprobe@fuse.service: Executing: /sbin/modprobe -abq fuse
```

gitbls, 2024-02-29:
I see this on every one of my Pi systems: Pi4s, Pi5s, and even one Pi02W running 32-bit...on every boot.

The funny thing is that the logs from all old boots of the system are still available, so it's not like log information from prior reboots is lost forever.

Nonetheless, it is a bit perturbing.

demukti, 2024-02-29:
> I see this on every one of my Pi systems: Pi4s, Pi5s, and even one Pi02W running 32-bit...on every boot.
> 
> The funny thing is that the logs from all old boots of the system are still available, so it's not like log information from prior reboots is lost forever.
> 
> Nonetheless, it is a bit perturbing.

Do you also get that same rogue date or is it a different one? I'm still pretty new at Linux and it's kernel but just wondering if that day could come out of the kernel or if it is set by the system. I just can't figure out where that Jan 26 comes from in my case. My kernel was built on Jan 25 so that may be too close to be a coincidence?

```
$ uname -v
#1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)
```

gitbls, 2024-02-29:
> > I see this on every one of my Pi systems: Pi4s, Pi5s, and even one Pi02W running 32-bit...on every boot.
> > The funny thing is that the logs from all old boots of the system are still available, so it's not like log information from prior reboots is lost forever.
> > Nonetheless, it is a bit perturbing.
> 
> Do you also get that same rogue date or is it a different one? I'm still pretty new at Linux and it's kernel but just wondering if that day could come out of the kernel or if it is set by the system. I just can't figure out where that Jan 26 comes from in my case. My kernel was built on Jan 25 so that may be too close to be a coincidence?
> 
> ```
> $ uname -v
> #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)
> ```
> 
> Also, are all your pi's running bookworm? Because I don't get this on Bullseye Lite.

No, I don't see the rogue dates. 

And, apologies, I neglected to mention that all the Pi systems I mentioned are running Bookworm. No bullseye on my network except for a WSL instance or two.

demukti, 2024-02-29:
> No, I don't see the rogue dates.
Thank you for sharing your results! Just to make sure if you run:
`journalctl | grep 'Executing: /sbin/fake-hwclock' `
you just get a chronological output instead of this? :
```
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Feb 28 17:20:21 tzvpi1 (-hwclock)[1089]: fake-hwclock.service: Executing: /sbin/fake-hwclock save
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Feb 28 21:00:30 tzvpi1 (-hwclock)[1411]: fake-hwclock.service: Executing: /sbin/fake-hwclock save
Jan 26 22:48:37 tzvpi1 (-hwclock)[285]: fake-hwclock.service: Executing: /sbin/fake-hwclock load
Feb 29 14:04:29 tzvpi1 (-hwclock)[1270]: fake-hwclock.service: Executing: /sbin/fake-hwclock save
```

gitbls, 2024-02-29:
Unfortunately my customized Bookworm doesn't use fake-hwclock load and save, so nothing to show you from that. 

But, back on Bullseye (before I stopped using fake-hwclock) I have vague recollection of seeing date confusion like you're seeing and wondering WTF was going on.

demukti, 2024-02-29:
> Unfortunately my customized Bookworm doesn't use fake-hwclock load and save, so nothing to show you from that.
> 
> But, back on Bullseye (before I stopped using fake-hwclock) I have vague recollection of seeing date confusion like you're seeing and wondering WTF was going on.

I didn't know that was an option. I'll have to look into that.
If you're not using fake-hwclock that unfortunately suggests that these are two unrelated issues.

May I ask if you you're seeing the 'journal corrupted' only or also the 'journal is from the future, refusing to append new data to it' message (with debug mode active)?

gitbls, 2024-02-29:
> May I ask if you you're seeing the 'journal corrupted' only or also the 'journal is from the future, refusing to append new data to it' message (with debug mode active)?

Only seeing the 'journal corrupted' message.

by, 2024-03-22:
See also https://github.com/raspberrypi/linux/issues/5746

closed 2024-04-15 by XECDesign