raspberrypi/bookworm-feedback #223: Wayland dmabuf video output not available from `cvlc`

hughmcmaster, 2024-03-02:
I'm trying to play a 1080p25 video (bitrate ~8,000 kb/s; Main@L4.0) in VLC. Playback is acceptable but not completely smooth.

Playback is best when using Wayland dmabuf video output compared to OpenGL for Embedded Systems 2 output.

I'd like to use Wayland dmabuf video output from the command line, but the output renderer not supported for some reason.

`vlc -v video.mp4` uses **wl_dmabuf** with DRM Video Accel for hardware decoding.
`cvlc -v video.mp4` uses **gles2 gl** with DRM Video Accel for hardware decoding.

This is replicated when using the python VLC bindings, which also fall back to **gles2 gl**. If I try to use wl_dmabuf with the python VLC Instance, the console is spammed with:
```
[0000007f9c046d80] main video output error: video output creation failed
[0000007fa400b3c0] main decoder error: failed to create video output
```

This occurs on Raspberry Pi 4 8GB running Bookworm with wayland. Is there a way to use `wl_dmabuf` via `cvlc` and the python VLC bindings?

ghollingworth, 2024-03-04:
cvlc is just a script which ends up being

exec /usr/bin/vlc -I dummy "$@"

hughmcmaster, 2024-03-04:
Fair enough. A better question then: why is wl_dmabuf not supported via the dummy backend?

Does wayland require a window?

closed 2025-02-09 by hughmcmaster