raspberrypi/bookworm-feedback #224: Doubleclick on a folder always adds a tab in an already opened folder

nielsbaloe, 2024-03-02:
Doubleclick on a folder always adds a tab in an already opened folder.

Since 1995 I am used that when I doubleclick a folder on the desktop, it opens a new window. Please add an option to open new windows in a window and not a tab. There is an option in pcmanfm that looks like it, but it does not have any effect.

lurch, 2024-03-05:
Can you confirm what exact version of Raspberry Pi OS you're using? I've just tried what you describe on a fully-up-to-date version of Raspberry Pi OS Bookworm 64-bit on a Pi 5, and I'm getting the behaviour you requested where every folder I double-click on the desktop opens a new (separate) FileManager window.
I don't see an option to open a folder in a new tab even if I _wanted_ to? :confused:

nielsbaloe, 2024-03-06:
That sounds great! When I posted this I was using the latest version, fully updated. Maybe there is a glitch on my system, I will do a full re-install anyway in a few months to see whether my long list of things not working on wayland are fixed.

lurch, 2024-03-06:
MicroSD cards are very cheap these days, so you could always buy a spare card on which to flash a fresh copy of Raspberry Pi OS; instead of "re-installing" your existing card :slightly_smiling_face:

spl237, 2024-03-19:
I've never seen this happen - I suspect your system has become corrupt somehow.

closed 2024-03-19 by spl237