raspberrypi/bookworm-feedback #228: When wanting to copy the selected files&folders, you now have to press CTRL before start dragging

nielsbaloe, 2024-03-02:
If I drag files (on the same disk to a different folder) and thén press CTRL, it ignores the CTRL so then files are not copied. 

Very frustrating to suddenly learn to press CTRL fírst and then start dragging, this is the first OS that I work with that has this annoyance. Please detect CTRL while dragging too.

lurch, 2024-03-05:
I've _always_ pressed CTRL **before** I start dragging the file, I had no idea that it worked the other way around too! :laughing: 

I've just tested this, and there are actually four slightly different scenarios:
 1. Press and hold CTRL, and then drag a file over a folder - file displays a 'plus' icon. If you now release the mouse button the file gets copied into the new folder.
 1. Drag the file over a folder - file displays an 'arrow' icon. If you now release the mouse button the file gets moved into the new folder.
 1. Drag the file over a folder - file displays an 'arrow' icon. Press and hold CTRL (without moving the mouse). If you now release the mouse button the file gets moved into the new folder.
 1. Drag the file over a folder - file displays an 'arrow' icon. Press and hold CTRL and move the mouse slightly - file displays a 'plus' icon. If you now release the mouse button the file gets copied into the new folder.

So I guess your request is for number 3 to work the same as number 4. But as a temporary workaround, I guess number 4 isn't too difficult to use :slightly_smiling_face:

nielsbaloe, 2024-03-06:
I never drag files OVER a folder, I usually have folders opened so I drag then IN them. I'll see whether this (still feeling very weird new requirement) is something I can get used to. Thanks!!

closed 2024-03-19 by spl237