raspberrypi/bookworm-feedback #229: Chromium Browser mouse cursor bug

sudo-splinter-cell, 2024-03-04:
I have opened this bug report in December last year in the appropriate "rpi-distro/chromium browser" github page. However, since then, there have been numerour updates to Chromium and this issue wasn't addressed. I also did not get any reply to my bug report strenghtening my suspicion that maybe nobody is checking that particular page. So i am referring my bug report also in here hoping to get a little bit more attention:

**[https://github.com/RPi-Distro/chromium-browser/issues/40]**

sudo-splinter-cell, 2024-05-08:
Is this issue being looked into? Because this is the only reason i default to X11 still. It can't be just me.

lurch, 2024-05-13:
I've just tested this myself, and I'm unable to reproduce this.
Also, the default mouse cursor in Raspberry Pi OS Bookworm looks different to the one in your video, so have you changed which desktop you're using? Or are you using some 3rd-party OS?

sudo-splinter-cell, 2024-05-13:
> I've just tested this myself, and I'm unable to reproduce this. Also, the default mouse cursor in Raspberry Pi OS Bookworm looks different to the one in your video, so have you changed which desktop you're using? Or are you using some 3rd-party OS?

Actually, it is the latest RPIOS. However, i am running a different desktop on top of it (KDE Plasma). And this mouse cursor issue only happens on "wayland". Not on X11. If i remember it correctly, it was also happening on the default desktop environment of RPIOS (at least a couple months ago, and again, only on wayland). If they changed and corrected it, i don't know, because on KDE Plasma, it is still an issue.

If that means it is a KDE Plasma issue, i'll open the bug report on their forum. So, are you on wayland or x11?

lurch, 2024-05-13:
I'm using the default Wayfire desktop on the 64-bit Raspberry Pi OS running on a Pi 5. So yes, it sounds like you'll have to take this up with KDE...

closed 2024-05-13 by lurch

sudo-splinter-cell, 2024-05-18:
This only happens on the RPI maintained Chromium package. The debian Chromium package or "any other" software package i have tried does not have this issue. So i do not understand how this is a KDE Plasma issue??? I mean it could be of course, but i could not replicate this on any other software package i have tried, including the Chromium of debian.

lurch, 2024-05-18:
The RPi-maintained Chromium package is only officially supported on the RPi-maintained Wayfire desktop (on the RPi-maintained Raspberry Pi OS :wink: ). Where, as I mentioned above, the mouse cursor works totally fine. If you choose to try running things on a desktop that we don't officially support, that's entirely at your own risk.