raspberrypi/bookworm-feedback #230: USB-OTG on Raspi 5 doesn't work

zerogtiger, 2024-03-08:
Setup: 
- Raspberry Pi 5
- Debian 12 Bookworm 64 bit Lite (off of Raspi imager)

Steps to reproduce:
- Burn SD card with above mentioned OS
- Edit `config.txt` to include `dtoverlay=dwc2` after the final `[all]` line.
- Edit `cmdline.txt` to include `modules-load=dwc2,g_ether` before `rootwait`
- After boot, ran `sudo rpi-update`

Expected bahaviour
- After connecting the USB cable to the USB-C port on the RPi 5 to the computer, I expect to be able to SSH into the pi via `ssh <username>@<piname>.local`

Actual behaviour
- ssh hangs, unable to ping

Comments
- I have done the exact same steps on my Raspberry Pi 0 W (except for the `sudo rpi-update`) and I received the expected bahaviour. 
- I have read https://www.hardill.me.uk/wordpress/2023/12/23/pi5-usb-c-gadget/. The suggested method did not work. 
- I have read https://github.com/raspberrypi/bookworm-feedback/issues/77. The suggested methods did not work. People have said it's been fixed but I was unable to reproduce. 
- I have read https://forums.raspberrypi.com/viewtopic.php?t=360751. This was not exactly helpful.

he-is-harry, 2024-03-19:
Same here, I could not find a fix

zerogtiger, 2024-03-26:
Resolved: solutions for the Pi5 can be found here: https://forums.raspberrypi.com/viewtopic.php?t=358573 as well as https://forums.raspberrypi.com/viewtopic.php?t=364247

Remark: unbelievable that no engineers were able to provide any context or assistance.

closed 2024-03-26 by zerogtiger