raspberrypi/bookworm-feedback #231: PulseAudio applications failing with connection refused in Bookworm 64-bit

Memotech-Bill, 2024-03-08:
To reproduce, do a fresh install of Raspberry Pi OS Bookworm 64-bit with desktop, configured with ssh enabled. Then log in using SSH, and try and run any of the standard PulseAudio applications.

The following transcript shows a typical example:

````
pi@piradio:~ $ ps -A | grep wire
    830 ?        00:00:00 pipewire
    831 ?        00:00:01 wireplumber
    832 ?        00:00:00 pipewire-pulse
pi@piradio:~ $ PULSE_LOG=4 pactl list short
Parsing configuration file '/etc/pulse/client.conf'
Failed to open configuration file '/etc/pulse/client.conf.d/01-enable-autospawn.conf': No such file or directory
Using shared memfd memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
Trying to connect to {29873391b2e14803ba74b1c7a6046353}unix:/run/user/1000/pulse/native...
parsing address failed: {29873391b2e14803ba74b1c7a6046353}unix:/run/user/1000/pulse/native
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
pi@piradio:~ $ ls -l /run/user/1000/pulse/native
srw-rw-rw- 1 pi pi 0 Dec  5 05:06 /run/user/1000/pulse/native
pi@piradio:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             71M     0   71M   0% /dev
tmpfs            42M  1.1M   41M   3% /run
/dev/mmcblk0p2   15G  4.3G  9.1G  33% /
tmpfs           210M  8.0K  210M   1% /dev/shm
tmpfs           5.0M  8.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  510M   73M  438M  15% /boot/firmware
tmpfs            42M   40K   42M   1% /run/user/1000
pi@piradio:~ $ uname -a
Linux piradio 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
pi@piradio:~ $ 
````

From this it can be seen that:

* The pipewire, wireplumber and pipewire-pulse servers are all running.
* The `pactl` utility is failing to connect.
* Diagnostics have been enabled, from which it can be seen that it is failing to connect to `/run/user/1000/pulse/native`.
* `/run/user/1000/pulse/native` exists, with owner and group pi.
* `/run/user/1000' is a tmpfs
* OS details are given.
* This was run on a Raspberry Pi Zero 2W

Memotech-Bill, 2024-03-09:
I have now come across an oblique reference which suggests that by design it is not possible to connect to PulseAudio from an SSH session. I have yet to find any official / definitive reference for this.

In general it seems very difficult to find reliable documentation for Linux sound. Perhaps the Foundation could consider funding a documentation project?

cmitu, 2024-03-14:
> I have now come across an oblique reference which suggests that by design it is not possible to connect to PulseAudio from an SSH session. I have yet to find any official / definitive reference for this.

I don't think that's true. FWIW, the command you've used works fine for me on a Pi4 with Pipewire/Wireplumber/Pipewire-pulse started. 

````
PULSE_LOG=4 pactl list short
Parsing configuration file '/etc/pulse/client.conf'
Failed to open configuration file '/etc/pulse/client.conf.d/01-enable-autospawn.conf': No such file or directory
Using shared memfd memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
Trying to connect to /run/user/1000/pulse/native...
SHM possible: yes
Protocol version: remote 35, local 35
Negotiated SHM: no
Memfd possible: yes
Negotiated SHM type: private
1	libpipewire-module-rt	{
            nice.level    = -11
            #rt.prio      = 88
            #rt.time.soft = -1
            #rt.time.hard = -1
        }
2	libpipewire-module-protocol-native
3	libpipewire-module-profiler
5	libpipewire-module-metadata
7	libpipewire-module-spa-device-factory
9	libpipewire-module-spa-node-factory
11	libpipewire-module-client-node
13	libpipewire-module-client-device
15	libpipewire-module-portal
16	libpipewire-module-access	{
            # access.allowed to list an array of paths of allowed
            # apps.
            #access.allowed = [
            #    /usr/bin/pipewire-media-session
            #]

            # An array of rejected paths.
            #access.rejected = [ ]

            # An array of paths with restricted access.
            #access.restricted = [ ]

            # Anything not in the above lists gets assigned the
            # access.force permission.
            #access.force = flatpak
        }
17	libpipewire-module-adapter
19	libpipewire-module-link-factory
21	libpipewire-module-session-manager
536870912	module-always-sink
70	alsa_output.platform-bcm2835_audio.stereo-fallback	PipeWire	s16le 2ch 48000Hz	SUSPENDED
70	alsa_output.platform-bcm2835_audio.stereo-fallback.monitor	PipeWire	s16le 2ch 48000Hz	SUSPENDED
31	PipeWire	pipewire
33	PipeWire	wireplumber
34	PipeWire	wireplumber
124	PipeWire	pactl
55	alsa_card.platform-bcm2835_audio	alsa
56	alsa_card.platform-fef00700.hdmi	alsa
57	alsa_card.platform-fef05700.hdmi	alsa
````

Memotech-Bill, 2024-03-14:
Was this with a fresh install of Raspberry Pi OS Bookworm?

How was the pipewire-pulse daemon started? Running as root or as the pi user? By default it is started as the current user.

````
````pi@piradio:~ $ systemctl --user status pipewire-pulse
● pipewire-pulse.service - PipeWire PulseAudio
     Loaded: loaded (/usr/lib/systemd/user/pipewire-pulse.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-03-14 08:52:46 GMT; 19s ago
TriggeredBy: ● pipewire-pulse.socket
   Main PID: 878 (pipewire-pulse)
      Tasks: 3 (limit: 3912)
        CPU: 85ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire-pulse.service
             └─878 /usr/bin/pipewire-pulse

Mar 14 08:52:46 piradio systemd[860]: Started pipewire-pulse.service - PipeWire PulseAudio.
Mar 14 08:52:46 piradio pipewire-pulse[878]: mod.rt: Can't find org.freedesktop.portal.Desktop. Is xdg-desktop-portal running?
pi@piradio:~ $ PULSE_LOG=4 pactl list short
Parsing configuration file '/etc/pulse/client.conf'
Failed to open configuration file '/etc/pulse/client.conf.d/01-enable-autospawn.conf': No such file or directory
Using shared memfd memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
Trying to connect to {29873391b2e14803ba74b1c7a6046353}unix:/run/user/1000/pulse/native...
parsing address failed: {29873391b2e14803ba74b1c7a6046353}unix:/run/user/1000/pulse/native
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
pi@piradio:~ $ systemctl --user status pipewire-pulse
● pipewire-pulse.service - PipeWire PulseAudio
     Loaded: loaded (/usr/lib/systemd/user/pipewire-pulse.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-03-14 08:52:46 GMT; 1min 12s ago
TriggeredBy: ● pipewire-pulse.socket
   Main PID: 878 (pipewire-pulse)
      Tasks: 3 (limit: 3912)
        CPU: 85ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire-pulse.service
             └─878 /usr/bin/pipewire-pulse

Mar 14 08:52:46 piradio systemd[860]: Started pipewire-pulse.service - PipeWire PulseAudio.
Mar 14 08:52:46 piradio pipewire-pulse[878]: mod.rt: Can't find org.freedesktop.portal.Desktop. Is xdg-desktop-portal running?

cmitu, 2024-03-14:
> Was this with a fresh install of Raspberry Pi OS Bookworm?

No, it's not a fresh install, but otherwise a pretty stock installation.

> How was the pipewire-pulse daemon started? Running as root or as the pi user? By default it is started as the current user.

It's started as the intall user (`pi`)

```
 systemctl --user status pipewire-pulse
● pipewire-pulse.service - PipeWire PulseAudio
     Loaded: loaded (/usr/lib/systemd/user/pipewire-pulse.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-03-12 18:03:18 GMT; 1 day 22h ago
TriggeredBy: ● pipewire-pulse.socket
   Main PID: 981 (pipewire-pulse)
      Tasks: 3 (limit: 3910)
        CPU: 5min 14.927s
     CGroup: /user.slice/user-1000.slice/user@1000.service/session.slice/pipewire-pulse.service
             └─981 /usr/bin/pipewire-pulse

Mar 12 18:03:18 bookworm-pi systemd[961]: Started pipewire-pulse.service - PipeWire PulseAudio.
```