raspberrypi/bookworm-feedback #232: Blank black screen on boot

reukiodo, 2024-03-13:
Fresh install of http://downloads.raspberrypi.org/raspios_full_arm64/images/raspios_full_arm64-2024-03-13/2024-03-12-raspios-bookworm-arm64-full.img.xz on known-working 512GB Samsung Evo uSD card in RaspberryPi 4 causes blank black screen on DSI display on boot.

Workaround of `dtoverlay=vc4-fkms-v3d` in `/boot/firmware/config.txt` allows GUI to be shown on DSI display as found in several forum posts, but a workaround/edit shouldn't be necessary for out-of-the-box.

WIth default `dtoverlay=vc4-kms-v3d`
```
2024-03-13 16:11:15 root@raspberrypi4:~# kmsprint 
terminate called after throwing an instance of 'std::runtime_error'
  what():  No modesetting DRM card found
Aborted
2024-03-13 16:11:52 root@raspberrypi4:~# 
```

With `dtoverlay=vc4-fkms-v3d` workaround:
```
2024-03-13 16:22:27 root@raspberrypi4:~# kmsprint 
Connector 0 (89) DSI-1 (connected)
  Encoder 0 (88) DSI
    Crtc 0 (87) 800x480@60.05 27.777 800/61/2/44/+ 480/7/2/21/+ 60 (60.05) P|D 
      Plane 0 (31) fb-id: 97 (crtcs: 0) 0,0 800x480 -> 0,0 800x480 (XR24 AR24 XB24 AB24 RX24 BX24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 P030)
        FB 97 800x480 XR24
      Plane 7 (80) fb-id: 102 (crtcs: 0) 0,0 64x64 -> 455,151 64x64 (XR24 AR24 XB24 AB24 RX24 BX24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 P030)
        FB 102 64x64 AR24
2024-03-13 16:24:04 root@raspberrypi4:~# 
```

reukiodo, 2024-03-14:
Switching to X11 from Wayland solved this issue (can use default `dtoverlay=vc4-kms-v3d` again) as well as several other issues (vnc, CTRL+ALT+F# switching, etc) and everything is working!

lurch, 2024-04-03:
Can you switch back to Wayfire with `vc4-kms-v3d` and then check to see if this is the same problem as https://github.com/raspberrypi/linux/issues/5801 ?

reukiodo, 2024-04-03:
I'm not exactly sure what you want me to do?

I reverted back to wayland (though raspi-config calls this wayfire ?) and it is back to black screen and no VNC. I can still ssh in. I tried to run `kmsprint` and it doesn't even have DSI in the list:
```
2024-04-03 14:26:42 root@raspberrypi4:~# kmsprint 
Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
2024-04-03 14:26:45 root@raspberrypi4:~# 
```
and reloading the driver? doesn't seem to change it:
```
2024-04-03 14:27:41 root@raspberrypi4:~# rmmod rpi_panel_attiny_regulator
rmmod: ERROR: Module rpi_panel_attiny_regulator is not currently loaded
2024-04-03 14:27:54 root@raspberrypi4:~# modprobe rpi_panel_attiny_regulator
2024-04-03 14:28:02 root@raspberrypi4:~# kmsprint 
Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (41) HDMI-A-2 (disconnected)
  Encoder 1 (40) TMDS
2024-04-03 14:28:14 root@raspberrypi4:~# 
```
So I put added the config line and reboot:
```
2024-04-03 14:28:14 root@raspberrypi4:~# echo "force_turbo=1" >> /boot/firmware/config.txt
2024-04-03 14:29:46 root@raspberrypi4:~# reboot
2024-04-03 14:29:49 root@raspberrypi4:~# 
```
but it's still a black screen... and no VNC...

Thankfully switching back to X11 resolves everything and all is working again.

Well, not everything... it seems that `kmsprint` fails when on X11?
```
2024-04-03 14:43:52 root@raspberrypi4:~# kmsprint 
terminate called after throwing an instance of 'std::runtime_error'
  what():  No modesetting DRM card found
Aborted
2024-04-03 14:43:55 root@raspberrypi4:~# 
```

lurch, 2024-04-03:
> and reloading the driver? doesn't seem to change it:

Okay, so it looks like you're seeing a slightly different problem then. Thanks for testing!

In the non-working setup, do you see any errors in the `dmesg` output?

reukiodo, 2024-04-05:
<details><summary>dmesg output when using wayland</summary>

```
2024-04-05 07:51:07 root@raspberrypi4:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.20+rpt-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef642c0..0x000000003ef646bf (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef64700..0x000000003ef64735 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] pcpu-alloc: s85672 r8192 d29016 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:15:95:26 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=79b853cf-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7451600K/8245248K available (13376K kernel code, 2210K rwdata, 4260K rodata, 4864K init, 1083K bss, 269360K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43116 entries in 169 pages
[    0.000000] ftrace: allocated 169 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000331] Console: colour dummy device 80x25
[    0.000346] printk: console [tty1] enabled
[... 425 lines cut ...]
[   10.332988] brcmfmac_wcc: brcmf_wcc_attach: executing
[   10.350932] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   10.352150] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
[... cut ...]
[   12.969931] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   12.972888] bcmgenet fd580000.ethernet eth0: Link is Down
[   13.006220] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   16.756719] Bluetooth: RFCOMM TTY layer initialized
[   16.756746] Bluetooth: RFCOMM socket layer initialized
[   16.756762] Bluetooth: RFCOMM ver 1.11
2024-04-05 07:51:13 root@raspberrypi4:~# 
```
</details>

I'm not sure what to look for - it seems to only be looking for HDMI output I guess?

reukiodo, 2024-04-05:
<details><summary>dmesg output when using wayland</summary>

```
2024-04-05 07:58:00 root@raspberrypi4:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.20+rpt-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef642c0..0x000000003ef646bf (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef64700..0x000000003ef64735 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] pcpu-alloc: s85672 r8192 d29016 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:15:95:26 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=79b853cf-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7451580K/8245248K available (13376K kernel code, 2210K rwdata, 4260K rodata, 4864K init, 1083K bss, 269380K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43116 entries in 169 pages
[    0.000000] ftrace: allocated 169 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000337] Console: colour dummy device 80x25
[    0.000353] printk: console [tty1] enabled
[... 412 lines cut ...]
[   10.504897] alsactl[550]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   10.512616] Bluetooth: hci0: BCM: features 0x2f
[   10.514059] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[   10.514090] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[... cut ...]
[   12.955586] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   18.730051] Bluetooth: RFCOMM TTY layer initialized
[   18.730081] Bluetooth: RFCOMM socket layer initialized
[   18.730101] Bluetooth: RFCOMM ver 1.11
[   19.680207] platform 100000001.panel_disp: deferred probe pending
[   19.680220] mipi-dsi fe700000.dsi.0: deferred probe pending
[   19.680223] platform 100000002.reg_bridge: deferred probe pending
[   19.680226] i2c 10-0038: deferred probe pending
2024-04-05 07:58:03 root@raspberrypi4:~# 
```
</details>