raspberrypi/bookworm-feedback #233: No audio after latest update to PiOS

enemy38, 2024-03-14:
I just updated my raspberry pi 400's operating system OS to it latest version using the sudo apt-get update and upgrade yesterday and this morning I noticed that there is no audio at all in either video playback video web browser or vlc/mpv player.  I am using the current version of raspberry pi OS 64 bit.

spl237, 2024-03-19:
Which audio output device is selected if you right-click the volume icon on the taskbar?

Flnch, 2024-03-24:
Hi, since the latest updates I have (maybe the same) problem. After booting, there is no sound and the volume control button in the control bar disappeared. However, I cannot reproduce this issue. After rebooting the sound works most of the time.

ghollingworth, 2024-03-26:
So you just saw this once and now can't reproduce it anymore?

Flnch, 2024-03-26:
It happens more than once. But only every fourth / fifth time or so that I boot the Pi. Therefore, I cannot reliably reproduce the issue.

popcornmix, 2024-03-26:
Run `raspinfo | pastebinit` (`sudo apt install pastebinit` if you don't have it installed) and post the url, for a working audio boot and a non-working audio boot.

enemy38, 2024-04-02:
Hello again,  I'm so sorry that I took so long to respond to the question asked of me.  I work retail and it's been a busy few weeks for me.  To start, pi 400 boots into raspberry pi os there is no icon in the taskbar for audio.  This happens regardless if either pulse audio or pipewire audio is being used.  This happens every time that I power on the system and  I noticed that when if I log off and log back on without resetting it would boot in and the audio would work just fine.  I have to do this every time I boot into the system.  I also noticed that when I power on the system it would treat the hd tv that I use as a DVI monitor.  This seems to fix itself when the tv is turned off and on again.  Again I'm sorry I took so long to respond and I hope the info I've provided here will help.  I'll be off for the new couple of days to if you do need anything else just let me know.

lurch, 2024-04-03:
> I also noticed that when I power on the system it would treat the hd tv that I use as a DVI monitor.

That would explain the problem - DVI doesn't support audio, so if the Pi thinks that your monitor is DVI it won't send audio over the HDMI-connector. It sounds like you might be suffering from the same problem as described here (which also describes a workaround) ? https://forums.raspberrypi.com/viewtopic.php?t=327875

Alternatively, you might want to also try a different HDMI cable in case the EDID pins are playing up?

popcornmix, 2024-04-03:
Responding to [this request](https://github.com/raspberrypi/bookworm-feedback/issues/233#issuecomment-2020815880) would help us provide useful advice.

enemy38, 2024-04-03:
I ran the command after I installed the pastebin. 

This is the first one with the initial boot.
https://paste.debian.net/hidden/d7267b6d/

This is the second one after I log off and log back on without any system restart or power down.
https://paste.debian.net/hidden/c42a74aa/

I kept trying to copy and paste them from the terminal but they wouldn't let me so I just typed them out.  I hope this helps.

popcornmix, 2024-04-03:
The differences:
Before:
```
Default Sink: auto_null
Default Source: auto_null.monitor
```
After:
```
Default Sink: alsa_output.platform-fef05700.hdmi.hdmi-stereo
Default Source: alsa_output.platform-fef05700.hdmi.hdmi-stereo.monitor
```
The main issue (stopping audio working) is likely:
```
[    2.676455] EDID block 0 (tag 0x00) checksum is invalid, remainder is 251
[    2.676482]  [00] BAD  00 ff ff ff ff ff ff 00 5c 94 04 1a 01 00 00 00
[    2.676490]  [00] BAD  24 15 01 03 80 73 41 78 0a cf 74 a3 57 4c b0 23
[    2.676497]  [00] BAD  00 48 4c a1 08 00 81 80 61 40 45 40 31 40 01 01
[    2.676504]  [00] BAD  01 01 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c
[    2.676510]  [00] BAD  45 00 7e 8a 42 00 00 1e 01 1d 00 72 51 d0 1e 20
[    2.676516]  [00] BAD  6e 28 55 00 7e 8a 42 00 00 1e 00 00 00 fd 00 3a
[    2.676523]  [00] BAD  3e 1e 53 10 00 0a 20 20 20 20 20 20 00 00 00 fc
[    2.676529]  [00] BAD  00 45 57 32 34 54 33 4c 57 0a 20 20 20 20 01 f2
```
Normally edid errors are a faulty hdmi cable or display. Can you test with a different cable or display?
Also
```
HDMI0:             HDMI_HOTPLUG = 0x00000000
HDMI1:             HDMI_HOTPLUG = 0x00000001
```
You are connected to the second hdmi socket. You should really be using the first one (which is more capable). 
> When looking at the rear of the keyboard, the connector on the left is the primary display, so you'll need to ensure this one is always plugged in if using just one screen

But I don't think that will stop audio working.

enemy38, 2024-04-03:
I thought the hdmi connection closest to the power cable was the primary display output so I can change that.  The only thing I don't have is an extra micro hdmi to standard hdmi cable.  I'll have to order one or see if they have the micro to standard hdmi locally so I can test it.  Once I get that new cable I'll let you know and I'll run the pastebin thing for you again.

Flnch, 2024-04-07:
Sorry posting my logs took so long:

- Booted with sound: https://pastebin.com/dGZM8gH5
- Booted without sound: https://pastebin.com/Z2pk76WG

As I see it the same sink problem occurs:

```
Standard-Ziel: alsa_output.platform-107c701400.hdmi.hdmi-stereo
Standard-Quelle: alsa_output.platform-107c701400.hdmi.hdmi-stereo.monitor
```

vs.

```
Standard-Ziel: auto_null
Standard-Quelle: auto_null.monitor
```

The issue is only occurring once a few days so I am not sure whether this would be a cable issue as @popcornmix suggested in a comment above.

Flnch, 2024-04-13:
> I wonder if it is reporting `0`/`off` when it stops working.

This is not the case. The command always looks as follows, both for working and not working sound:

```
user@pi:~ $ amixer -d  -D sysdefault:CARD=vc4hdmi0 cget numid=1
VERIFY ID: numid=1,iface=MIXER,name=''
numid=1,iface=CARD,name='HDMI Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
```

***

But I am now able to reproduce the issue most of the time. My Pi is connected to my TV via HDMI. Sometimes, I first start the Pi and after 20-30 seconds turn on the TV. In those cases, the screen is detected and the visual output of the Pi is shown, but the audio is not available. If I first power on the TV, the sound works all of the times.

ErinTyres, 2024-06-20:
> If I first power on the TV, the sound works all of the times.

Thank you, that helped me fix my problem!  
Configuration:  Raspberry Pi 4 with Bookworm, old RCA TV that supports 1080p.
Problem:  Right click on the GUI's sound icon does not let me select HDMI.  (Problem does not occur with a newer display or a newer TV.  Problem does not occur with Raspberry Pi Buster.)
Solution:  Cycle power on the TV after the Pi boots up, then HDMI sound is selectable in the GUI's sound icon.