raspberrypi/bookworm-feedback #236: Wayfire 0.7.5 rpt33 crashes on touch event

thaihugo, 2024-03-19:
Hi,

There seems to be a bug with the latest Wayfire version available from the stable bookworm RPI version (not the one from the standard Debian version 7.4)

**Setup:** 
RPI 5 8GB, standard december image, latest apt upgrade. No specific modification, except for a custom wayfire.ini and default.ini that willl be posted next.
Touchscreen waveshare 10.1HP-CAPLCD Monitor (https://www.waveshare.com/wiki/10.1HP-CAPLCD_Monitor), normal USB event input, no specific drivers needed, 10 finger touch.

**Description:**
Once the system is updated to the 0.7.5-1~bpo11+1~rpt33 RPI version, touching event received by Wayfire will crash the session and close it. You get back to the login screen, and must relogin
Touching an app works normally. Scrolling works. It is touching anything else (the background) that crashes the session.

**Small log of the crash:**
`II 18-03-24 17:18:15.628 - [backend/headless/backend.c:68] Creating headless backend
II 18-03-24 17:18:15.628 - [backend/headless/backend.c:16] Starting headless backend
II 18-03-24 17:18:15.638 - [src/main.cpp:355] Using socket name wayland-1
II 18-03-24 17:18:15.790 - [src/core/seat/input-manager.cpp:46] handle new input: WaveShare  WaveShare 
II 18-03-24 17:18:15.790 - [src/core/seat/input-manager.cpp:46] handle new input: Logitech M350
II 18-03-24 17:18:15.798 - [src/core/seat/input-manager.cpp:46] handle new input: Logitech M350
II 18-03-24 17:18:15.798 - [src/core/seat/input-manager.cpp:46] handle new input: pwr_button
II 18-03-24 17:18:15.805 - [src/core/seat/input-manager.cpp:46] handle new input: vc4-hdmi-0
II 18-03-24 17:18:15.813 - [src/core/seat/input-manager.cpp:46] handle new input: vc4-hdmi-0
II 18-03-24 17:18:15.813 - [src/core/seat/input-manager.cpp:46] handle new input: vc4-hdmi-1
II 18-03-24 17:18:15.820 - [src/core/seat/input-manager.cpp:46] handle new input: vc4-hdmi-1
II 18-03-24 17:18:15.820 - [src/core/seat/input-manager.cpp:46] handle new input: Designer Compact Keyboard
II 18-03-24 17:18:15.828 - [src/core/seat/input-manager.cpp:46] handle new input: Designer Compact Keyboard
II 18-03-24 17:18:15.828 - [src/core/seat/input-manager.cpp:46] handle new input: Designer Compact Keyboard Mouse
II 18-03-24 17:18:15.828 - [backend/drm/drm.c:1502] Scanning DRM connectors on /dev/dri/card1
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1553] Found connector 'HDMI-A-1'
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1578] 'HDMI-A-1' connected
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1377] Physical size: 1210x680
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1439] Detected modes:
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1280x800 @ 59.493 Hz (preferred)
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1920x1080 @ 60.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1920x1080 @ 59.940 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1920x1080 @ 59.724 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1920x1080 @ 50.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1920x1080 @ 24.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1920x1080 @ 23.976 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1280x720 @ 60.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1280x720 @ 60.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1280x720 @ 59.940 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   1280x720 @ 50.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x576 @ 50.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x576 @ 50.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x480 @ 60.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x480 @ 60.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x480 @ 59.940 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x480 @ 59.940 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   720x480 @ 59.940 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   640x480 @ 60.000 Hz 
II 18-03-24 17:18:15.879 - [backend/drm/drm.c:1478]   640x480 @ 59.940 Hz 
II 18-03-24 17:18:15.880 - [backend/drm/drm.c:1553] Found connector 'HDMI-A-2'
II 18-03-24 17:18:15.880 - [backend/drm/drm.c:1614] connector HDMI-A-1: Requesting modeset
II 18-03-24 17:18:15.880 - [src/core/output-layout.cpp:1089] new output: HDMI-A-1
II 18-03-24 17:18:15.880 - [src/core/output-layout.cpp:535] loaded mode 1366x768@59790
II 18-03-24 17:18:15.888 - [backend/drm/drm.c:734] connector HDMI-A-1: Modesetting with 1280x800 @ 59.493 Hz
(type equals variant: [type: string, value: toplevel] | (type equals variant: [type: string, value: x-or] & focusable equals variant: [type: bool, value: 1]))
type equals variant: [type: string, value: overlay]
false
false
app_id equals variant: [type: string, value: foot]
(type equals variant: [type: string, value: toplevel] & floating equals variant: [type: bool, value: 1])
EE 18-03-24 17:18:15.967 - [types/wlr_cursor.c:875] Cannot map device "Logitech M350" to output (not found in this cursor)
EE 18-03-24 17:18:15.967 - [types/wlr_cursor.c:875] Cannot map device "pwr_button" to output (not found in this cursor)
EE 18-03-24 17:18:15.967 - [types/wlr_cursor.c:875] Cannot map device "vc4-hdmi-0" to output (not found in this cursor)
EE 18-03-24 17:18:15.967 - [types/wlr_cursor.c:875] Cannot map device "vc4-hdmi-1" to output (not found in this cursor)
EE 18-03-24 17:18:15.967 - [types/wlr_cursor.c:875] Cannot map device "Designer Compact Keyboard" to output (not found in this cursor)
EE 18-03-24 17:18:20.596 - [src/main.cpp:144] Fatal error: Segmentation fault
EE 18-03-24 17:18:20.604 - #1  wf::print_trace(bool) ??:?
[... 195 lines cut ...]
always_decorate = app_id is "Kodi"

[grid]
slot_l = <ctrl><alt>KEY_LEFT
slot_r = <ctrl><alt>KEY_RIGHT
slot_c = <ctrl><alt>KEY_UP
restore = <ctrl><alt>KEY_DOWN
#mouse_snap = false

[autostart]
autostart_wf_shell = false

#[autostart-static]
#autostart0 = waybar -s /home/hugo/.config/waybar/mocha-style.css
#autostart1 = swaybg -i /home/hugo/Downloads/catppuccinmontains.png
#autostart4 = wfrespawn wf-panel-pi
#autostart2 = wfrespawn pcmanfm --desktop --profile LXDE-pi
#autostart3 = lxsession-xdg-autostart

[input]
cursor_theme = PiXflat

[idle]
screensaver_timeout = -1
dpms_timeout = -1`

ghollingworth, 2024-03-19:
Can you reproduce with a clean image?

spl237, 2024-03-19:
Does the problem go away if you stop using swaybg and go back to using pcmanfm to draw the background?

thaihugo, 2024-03-20:
> Can you reproduce with a clean image?

Yes, same problem on a clean updated image. Also, could pinpoint to the latest wayfire update, as the starting version before updating is ok.

thaihugo, 2024-03-20:
> Does the problem go away if you stop using swaybg and go back to using pcmanfm to draw the background?

Removing swaybg does not solve the problem. However, pcmanfm "protects" the background, and stop the crashes, so it is a workaround, and probably the reason this bug has not been detected by more users.

spl237, 2024-03-20:
OK, found the problem - it is being caused by a fix for a different touchscreen issue! I've fixed this new issue - just need to make sure that hasn't broken the fix for the previous issue, and I will then push an update.

spl237, 2024-03-20:
Fix pushed to apt - the person who maintains our repo is away at the moment, so it may be a couple of days before this appears. If you still see the problem when wayfire 0.7.5-1~bpo11+1~rpt34 is installed, please re-open.

closed 2024-03-20 by spl237