raspberrypi/bookworm-feedback #237: RPI5 with X11 GUI stopped working after update

Skybird129, 2024-03-19:
Hi,

I'm running Raspberry Pi OS (64-bit) - Raspberry Pi OS with desktop Release date: March 12th 2024 on my Raspberry PI 5 8 GB. I used raspi-config to switch to X11. After installing the latest updates, the GUI stopped working after a reboot.

When I connect the Pi to my TV, I see the Raspberry PI OS splash screen for a brief time, before the screen turns black. VNC throws the error message "Cannot currently show the desktop". I'm still able to login via SSH.

Switching back to Wayfire didn't help.

Unfortunately, I've no idea how to troubleshoot this problem. Any help using X11 while able to use updates is greatly appreciated.

sworteu, 2024-04-03:
i have the same issue, since bookworm the x11 is changed. It doesn't seem to work as expected (that is as in the previous versions like bullseye etc). Siwtching back to x11 with raspi-config doesn't help eighter. There may have been some additional configurations missing or changes made that are unclear to me.

spookyfirehorse, 2024-06-09:
> Hi,
>
> I'm running Raspberry Pi OS (64-bit) - Raspberry Pi OS with desktop Release date: March 12th 2024 on my Raspberry PI 5 8 GB. I used raspi-config to switch to X11. After installing the latest [...]

may        rm -r  ~.*        but this delete all data in the home  hidden directory  i dont now wihich file it is but it helps

ralf-9000, 2024-06-10:
Since the last updates I have the same problem with bookworm on the first 2 CM4 systems, which I have migrated to bookworm.

I have played with setting the console to text mode and use startx. In the log is always hints that X can't open /dev/tty0, then for testing I do chmod 666 /dev/tty0 and after that xinit runs into "Cannot ope virtual console (Permission denied) ....

Ralf

quitmeyer, 2024-06-18:
I have a raspberry pi image of bookworm, updated 2 days ago that works fine in a rpi4. but if i put it in a rpi5, i get the "cannot currently show desktop" notification when vnc

popcornmix, 2024-06-18:
> Since the last updates I have the same problem with bookworm on the first 2 CM4 systems, which I have migrated to bookworm.

What do you mean by migrated? A clean install of bookworm, or did you in place upgrade bullseye to bookworm?

quitmeyer, 2024-06-18:
on a completely fresh install of bookworm made with rpi imager for a pi5, I enabled VNC and then enabled x11 and now I just get this blank gray screen instead of a cannot show desktop error
[image]

quitmeyer, 2024-06-18:
good question, if i connect to a monitor, i get a blinking cursor a little bit and then just a blank black screen

this also happens if i switch to x11 or wayfire

if i do startx like before on x11, it gives that error 

(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)

if i have it switched to wayland and run wayfire-pi i get these errors
```
pi@mothbox:~ $ sudo wayfire-pi
II 18-06-24 13:59:24.721 - [src/main.cpp:280] Starting wayfire version 0.7.5
II 18-06-24 13:59:24.721 - [libseat] [libseat/backend/seatd.c:64] Could not connect to socket /run/seatd.sock: No such file or directory
II 18-06-24 13:59:24.721 - [libseat] [libseat/libseat.c:76] Backend 'seatd' failed to open seat, skipping
II 18-06-24 13:59:24.721 - [libseat] [libseat/libseat.c:76] Backend 'logind' failed to open seat, skipping
II 18-06-24 13:59:24.722 - [libseat] [seatd/seat.c:39] Created VT-bound seat seat0
II 18-06-24 13:59:24.722 - [libseat] [seatd/server.c:145] New client connected (pid: 14551, uid: 0, gid: 0)
II 18-06-24 13:59:24.722 - [libseat] [libseat/backend/seatd.c:633] Started embedded seatd
II 18-06-24 13:59:24.722 - [libseat] [seatd/seat.c:170] Added client 7 to seat0
II 18-06-24 13:59:24.722 - [libseat] [seatd/seat.c:480] Opened client 7 on seat0
II 18-06-24 13:59:24.722 - [libseat] [libseat/libseat.c:73] Seat opened with backend 'builtin'
II 18-06-24 13:59:24.723 - [libseat] [libseat/backend/seatd.c:212] Enabling seat
II 18-06-24 13:59:24.723 - [backend/session/session.c:109] Successfully loaded libseat session
II 18-06-24 13:59:24.723 - [backend/session/session.c:473] Waiting for a DRM card device
EE 18-06-24 13:59:34.733 - [backend/backend.c:217] Found 0 GPUs, cannot create backend
EE 18-06-24 13:59:34.733 - [backend/backend.c:385] Failed to open any DRM device
II 18-06-24 13:59:34.733 - [libseat] [seatd/seat.c:418] No clients on seat0 to activate
II 18-06-24 13:59:34.741 - [libseat] [seatd/seat.c:524] Closed client 7 on seat0
II 18-06-24 13:59:34.741 - [libseat] [seatd/seat.c:192] Removed client 7 from seat0
EE 18-06-24 13:59:34.741 - [src/main.cpp:144] Fatal error: Segmentation fault
II 18-06-24 13:59:34.741 - [libseat] [seatd/client.c:471] Client disconnected
II 18-06-24 13:59:34.741 - [libseat] [libseat/backend/seatd.c:645] Stopped embedded seatd
EE 18-06-24 13:59:34.750 - #1  wf::print_trace(bool) ??:?
addr2line: '�,���': No such file
addr2line: '�,���': No such file
EE 18-06-24 13:59:34.755 - #2  X���� X����
EE 18-06-24 13:59:34.762 - #3  wlr_backend_get_drm_fd ??:?
EE 18-06-24 13:59:34.769 - #4  main ??:?
EE 18-06-24 13:59:34.884 - #5  __libc_start_call_main ../sysdeps/unix/sysv/linux/aarch64/libc-start.c:74
EE 18-06-24 13:59:34.978 - #6  call_init ../csu/libc-start.c:128
EE 18-06-24 13:59:34.985 - #7  _start ??:?
```

popcornmix, 2024-06-21:
wayfire-pi should be run without sudo (and from a directly attached tty, not ssh).

But assuming you get the same output without sudo, then it looks like wayfire is crashing with a segfault.

If can find minimal instructions to reproduce this (e.g. start from a fresh install, do x, y, and z, then see the crash) we may be able to investigate more.

tempelmann, 2024-07-18:
I ran into the same symptons on my Pi5 after trying to install software to enable a WaveShare LCD module. Eventually, by using `startx` and then looking at the log file at `.local/share/xorg/Xorg.0.log`, I found the error msg `Failed to load module "fbturbo" (module does not exist, 0)`. I eventually found an entry inside `/usr/share/X11/xorg.conf.d/` referring to that name - after deleting that file, the wayland desktop appeared immediately.

tempelmann, 2024-07-18:
Yes, that file and another one called `/usr/share/X11/xorg.conf.d/99-fbturbo.~`. I followed some confusing instructions at [https://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Getting_Started](https://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Getting_Started) (along with the sub page for manual installation) and I suspect those installed one or both these files.

lurch, 2024-07-18:
I'm afraid that if you get errors from following (old and outdated?) third-party instructions, then there's not much that we can do about that in Raspberry Pi OS!

Are any of the earlier posters in this issue still having problems on an installed-from-fresh and fully-updated Bookworm image?

quitmeyer, 2024-07-18:
> Are any of the earlier posters in this issue still having problems on an installed-from-fresh and fully-updated Bookworm image?

my stuff is working fine now in regards to this!