raspberrypi/bookworm-feedback #243: Today apt update for raspi-utils breaks normal video mode with no way to change it

qrp73, 2024-04-02:
Today I get latest apt upgrade for raspi-utils, after reboot the screen resolution is switched to non native 1200x720 instead of native 1280x1024 (which is always was used before). 

I opened Screen Configuration  and tried to change screen resolution to 1280x1024 manually, but it don't have effect and resolution still remains 1200x720. Also there is disappears refresh rate selection. Reboot, power off, reconnect HDMI also don't helps. The resolution still remains 1200x720

Please fix it asap, because it prevents to use raspi os due to non native display resolution.

popcornmix, 2024-04-03:
Report output of `uname -a` and `vcgencmd version` to begin with. What is the model of Pi?

An unreliable hdmi cable seems the simplest explanation based on the information provided.
(it sometimes works, it sometimes doesn't).

If you can absolutely confirm the issue was related to an update, then it should be possible to confirm that.
To start with, if you use a spare sdcard, install a fresh bookworm sdcard image, but don't update it.

Does this work 100% reliably without changes to config.txt/cmdline.txt?
Check multiple boots, and try adjusting position of hdmi cable in case it is more likely to fail when flexed in a certain direction.

If you have another hdmi cable to try, that would be useful. Also try using the other hdmi socket (if you are using pi4 or pi5). Try also any other hdmi socket on display, and another display if possible. The more information the better.

qrp73, 2024-04-03:
> Does this work 100% reliably without changes to config.txt/cmdline.txt? Check multiple boots, and try adjusting position of hdmi cable in case it is more likely to fail when flexed in a certain direction.

just tried to remove video mode from cmdline.txt and tried to reboot. Screen Confguration works as expected. Most of all this issue happens due to bad EDID reading from HDMI, but I have no idea how it's possible, because I'm sure 100% that HDMI cable was ok. I tried to reconnect it several times with reboot, tried to power off display power cable, but it didn't helped. 

And now EDID reading is ok and Screen Configuration works as expected, with no change to HDMI cable... So, I'm confused. What can be root of cause for bad EDID reading?

popcornmix, 2024-04-03:
> What can be root of cause for bad EDID reading?

A faulty cable can certainly result in intermittent reading of the EDID data.
Depending on exactly how it's flexed it may work or not work.

Without further information, that seems more likely than a software issue (which broke things, but fixed itself without additional updates).

If we start getting other similar reports, or you can identify a version of the software that behaves significantly better we can investigate more.

qrp73, 2024-04-11:
I cannot reproduce it anymore. I suspect it happens due to power off/on switch for the display. Maybe something going wrong with display firmware during power-on and it started to sent wrong EDID bytes. Probably display firmware keeps running from capacitors or/and HDMI power from RPI and it can explain why it still sent wrong EDID even after display power off/on. Full display power off with HDMI reconnect fixed this issue.  I tried to reproduce it with power off/on for the display, but all works ok.

lurch, 2024-04-11:
> I cannot reproduce it anymore.

I'll close this then; but please do comment if you discover a way to reproduce it in future.

closed 2024-04-11 by lurch