raspberrypi/bookworm-feedback #245: Logout from UI closes remote ssh sessions (and kills detached screen)

miceno, 2024-04-20:
### How to reproduce

1. Open a remote connection using VNC to RPi from your laptop in the same LAN as the RPi (using Wifi). You will see Wayland desktop.
2. Open a remote `ssh` session from your laptop in the same LAN as the RPi (using Wifi).
3. Open a `screen` window on the SSH session and detach it.
4. Logout from the Wayland desktop.

### Expected behaviour

After logout of Wayland desktop, the SSH is active and I can attach again to the detached `screen` session.

### Actual behaviour

The SSH session is closed when closing the Wayland session.

Upon reopening a new `ssh` to the RPi, and running `screen`, the previous detached session is dead:

```
Connection to raspberrypi.local closed by remote host.
Connection to raspberrypi.local closed.
$ ssh user@raspberrypi.local
Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Apr 20 11:52:06 2024
user@raspberrypi:~ $ screen -wipe
There is a screen on:
	3217.pts-0.raspberrypi	(04/20/24 11:46:17)	(Removed)
1 socket wiped out.
No Sockets found in /run/screen/S-astro.
```

There is no way to recover the detached `screen` session, and the only way is by wiping the dead session.

spl237, 2024-04-21:
At present, when a VNC session is closed, all processes belonging to that user are also closed, so if you have logged in via SSH as the same user, then that SSH session will also be terminated.

This was a recent change, and there are arguments on both sides as to whether this is the "right" thing to do, but it is intentional and not a bug.

andrum993, 2024-04-21:
> ...there are arguments on both sides as to whether this is the "right" thing to do...

Please name an argument for this new behaviour, as I'm struggling to think of any. Why would you want a footgun like this? Does any other remote control software do this???

spl237, 2024-04-22:
As with everything else in this day and age, "security". There is an argument that when a user logs out, that should not leave any of the processes started by that user running, as otherwise the user is not really "logged out" and their account is still active and could potentially be accessed.

I am not going to get into a debate about whether this is right or wrong (so please don't start one...) but the behaviour is, as far as I am aware, deliberate.

miceno, 2024-04-22:
@spl237 Thank you for the details!

This is not the place then to argue about it, so are you aware where that discussion is happenning about what to do when logging out from the GUI?

miceno, 2024-04-22:
> At present, when a VNC session is closed, all processes belonging to that user are also closed, so if you have logged in via SSH as the same user, then that SSH session will also be terminated.

So it is the code in `wayvnc` that is closing the session? Or is it a wrapper around it? Would you please point me to the code that is making this happen? I can then try to patch it in my setup.

cmitu, 2024-04-23:
> Does any other remote control software do this???

Reminds me of _GNU screen gets killed on logout_ - https://bugzilla.redhat.com/show_bug.cgi?id=1480460, which leads to https://lists.freedesktop.org/archives/systemd-devel/2016-May/036583.html. Perhaps it's something similar and can be controlled by configuring `systemd-logind` ?

spl237, 2024-05-07:
See above - it is because it is a process owned by the same user, so it gets terminated when the desktop closes, since this change - https://github.com/raspberrypi-ui/pishutdown/commit/113553f878ccac367dab48af657da8d20f0a605f

If you close the desktop by just killing the wayfire process manually, the SSH process will continue running.

andrum993, 2024-05-08:
Killing processes outside of the same session amounts to a mini-denial of service, breaks the principle of least surprise, and is, in my view, a daft feature. Please remove it.

closed 2025-06-30 by lurch