raspberrypi/bookworm-feedback #247: chromium-browser attempt to play youtube video leads to error and freeze

qrp73, 2024-05-06:
Steps to reproduce:
1) Open chromium-browser
2) Open link https://www.youtube.com/watch?v=Qjjf4dTO1Qo

Expected result: video is played

Actual result: video playback freezes and error appears in the log:
```
[ 9682.038036] bcm2835-codec bcm2835-codec: vidioc_s_fmt: Current buffer size of 314880 < min buf size 322560 - driver mismatch to MMAL
```

Firefox with disabled hw acceleration playing this video with no issue.

PS: I also catch some strange freeze in chromium-browser during video playback which leads to complete UI freeze of chromium-browser, app restart didn't helped and unable to kill chromium-browser process with timeout during attempt to reboot. But I'm not sure it was related with this issue, because it was happened on some other video, I didn't notice which one cause it.

ghollingworth, 2024-05-07:
Can you confirm hardware and software configuration here?  

I'm unsure whether the reference to MMAL is just a mistake in the driver or whether this is actually using the MMAL driver (in which case this is not supported in Bookworm)

6by9, 2024-05-07:
> I'm unsure whether the reference to MMAL is just a mistake in the driver or whether this is actually using the MMAL driver (in which case this is not supported in Bookworm)

The V4L2 codec driver is sitting on top of MMAL in the firmware.
The kernel has computed what it believes is the correct buffer size, but when it is passed to the firmware it got back a different number. The question is why kernel and firmware have a mismatch, and that is likely to be down to exact stream geometry. I'm downloading it now to analyse.

qrp73, 2024-05-07:
hardware is RPI4 4GB v1.5
```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Apr 17 2024 17:27:09 
Copyright (c) 2012 Broadcom
version 86ccc427f35fdc604edc511881cdf579df945fb4 (clean) (release) (start)

$ uname -a
Linux raspi 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
```

6by9, 2024-05-07:
360x290 or 328x240 would appear to be the potential answers.
 
Adding alignment means we're most likely looking at 328x240, aligned to 352x240 and RGBA giving 314880 bytes. I suspect the firmware is looking for 64 byte alignment, so https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L266 is incorrect declaring 32 (also line 259).

6by9, 2024-05-07:
There's something different in your configuration than mine.

Asking my Pi4 running Wayfire to play that URL in Chromium, and it works fine. It also asks the decoder for YUV420, not RGBA, so my buffer sizes end up as 126720 bytes for the 328x240 stream.

@jc-kynesim is there some magic for how Chromium chooses the pixel format?

qrp73, 2024-05-07:
I still can reproduce it, here is Stats for nerds:
[image]

I also found that when I switch video quality it leads to different buffer size in error:

for 240p:
```
[26130.084047] bcm2835-codec bcm2835-codec: vidioc_s_fmt: Current buffer size of 314880 < min buf size 322560 - driver mismatch to MMAL
```

for 144p:
```
[26156.030180] bcm2835-codec bcm2835-codec: vidioc_s_fmt: Current buffer size of 115200 < min buf size 119808 - driver mismatch to MMAL
```

jc-kynesim, 2024-05-10:
I can reproduce the fail. The h/w accel path we use in chrome changed in v124 as the old (functional, efficient) video decode accelerator (VDA) path went away and was replaced by the direct(?) decoder. The latter will typically end up asking for RGBX from V4L2. When it works it works well, but it looks like I need some init patches similar to what I had to do in VDA. Not sure (yet) what is different here to working video. Will fix.

6by9, 2024-05-10:
@jc-kynesim if you can reproduce then as per my earlier comment I think it's just that https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L266 and https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L259 should be reading 64 across the board.
I hadn't managed to reproduce, hence wasn't going to totally guess on it.

jc-kynesim, 2024-05-13:
OK - I'll try that and report back

jc-kynesim, 2024-05-13:
That might be necessary but it isn't sufficient - I think the issue is that chrome(?) ends up doing a S_FMT based on crop size rather than picture size. Still investigating.

jc-kynesim, 2024-05-13:
Scratch that - I'm an idiot - I wasn't running the kernel I thought I was running. Fix works fine - do you want to PR it or should I?

6by9, 2024-05-13:
Done - https://github.com/raspberrypi/linux/pull/6162

qrp73, 2024-05-14:
it looks like it also helps to fix some issues with video playback on other video hosting like telegram

6by9, 2024-05-14:
> When it will be available with apt upgrade?

It's now merged into our main kernel branch, so `sudo rpi-update` will get it from now on without having to specify the PR.
Apt updates are done every few months, or as and when anything really useful is merged into the kernel/firmware. I couldn't say when the next one will be.

qrp73, 2024-05-14:
I'm not sure if it was fixed in pull/6162 or it just include some other fixes, but when I installed this pulls/6162 I notice very significant improvement in video playback in browsers (both - firefox and chromium-browser). Now it plays much more smooth than before. Thanks a lot. :)

jc-kynesim, 2024-05-15:
Honestly I think it must be something else, that fix would affect a minority of videos - croppped_width & height are going to a multiple of 16 in most cases (which will work fine both before & after the patch) and that is the only thing that fix will change.
Subjectively I thought chrome v124 was smoother than v122 but I could easily have been kidding myself. If there are useful fixes they are in Wayland.