raspberrypi/bookworm-feedback #248: chromium-browser freezes and its process become zombie with no way to terminate it

qrp73, 2024-05-07:
Sometimes chromium-browser freezes during video playback and after that there is chromium-browse process (parent is systemd) which become zombie with no way to terminate it. During reboot it fails to kill chromium-browse process with timeout.

I catch it two times, but I'm not sure what is the sequence to reproduce it.
It starts to happens with some of latest updates. Previously I never seen such behavior.

Here is the log messages when it happens:
```
<...a lot of the same messages with timed out and -62...>
[119735.383258] bcm2835_mmal_vchiq: timed out waiting for sync completion
[119735.383275] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[119735.383282] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[119782.744094] bcm2835_mmal_vchiq: timed out waiting for sync completion
[119782.744130] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[119782.744140] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[119844.442079] bcm2835_mmal_vchiq: timed out waiting for sync completion
[119844.442121] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[119844.442138] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[119847.513288] bcm2835_mmal_vchiq: timed out waiting for sync completion
[119847.513316] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[119847.513329] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[119903.066240] bcm2835_mmal_vchiq: timed out waiting for sync completion
[119903.066259] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[119903.066266] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[119918.170519] bcm2835_mmal_vchiq: timed out waiting for sync completion
[119918.170544] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[119918.170552] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[120106.334182] bcm2835_mmal_vchiq: timed out waiting for sync completion
[120106.334238] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[120106.334262] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[120428.787145] bcm2835-codec bcm2835-codec: Mutex fail
```

There is possibility that this issue may be related with  #247, but this one has different behavior - in this case it leads to some kernel failure with zombie process and I didn't find how to stable reproduce it, while #247 is stable reproduced and don't leads to zombie process.

qrp73, 2024-05-14:
I catch it again and again. Here is first error appearance in the log from boot:
```May 13 13:40:38 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:40:38 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:41:09 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:41:09 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:47:55 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:47:55 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:58:40 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:58:40 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:58:40 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:58:40 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:59:02 raspi xdg-desktop-por[171676]: Backend call failed: Cannot invoke method; proxy is for the well-known name “org.freedesktop.ScreenSaver” with>
May 13 13:59:04 raspi xdg-desktop-por[171676]: Backend call failed: Cannot invoke method; proxy is for the well-known name “org.freedesktop.ScreenSaver” with>
May 13 13:59:18 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 13:59:18 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:49 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:49 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:49 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:49 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:51 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:51 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:03:51 raspi rtkit-daemon[171435]: Successfully made thread 197719 of process 196925 owned by '1000' RT at priority 10.
May 13 14:03:51 raspi rtkit-daemon[171435]: Supervising 8 threads of 5 processes of 1 users.
May 13 14:03:51 raspi rtkit-daemon[171435]: Supervising 8 threads of 5 processes of 1 users.
May 13 14:03:51 raspi rtkit-daemon[171435]: Supervising 8 threads of 5 processes of 1 users.
May 13 14:04:06 raspi rtkit-daemon[171435]: Supervising 8 threads of 5 processes of 1 users.
May 13 14:04:06 raspi rtkit-daemon[171435]: Supervising 8 threads of 5 processes of 1 users.
May 13 14:04:25 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:04:25 raspi rtkit-daemon[171435]: Supervising 7 threads of 4 processes of 1 users.
May 13 14:05:02 raspi xdg-desktop-por[171656]: A backend call failed: Inhibiting other than idle not supported
May 13 14:06:57 raspi dbus-daemon[171238]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.76' (uid=>
May 13 14:06:57 raspi (uetoothd)[198384]: ConfigurationDirectory 'bluetooth' already exists but the mode is different. (File system: 755 ConfigurationDirecto>
May 13 14:06:57 raspi systemd[1]: Starting bluetooth.service - Bluetooth service...
May 13 14:06:58 raspi bluetoothd[198384]: Bluetooth daemon 5.66
May 13 14:06:58 raspi dbus-daemon[171238]: [system] Successfully activated service 'org.bluez'
May 13 14:06:58 raspi systemd[1]: Started bluetooth.service - Bluetooth service.
May 13 14:06:58 raspi bluetoothd[198384]: Starting SDP server
May 13 14:06:58 raspi bluetoothd[198384]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
May 13 14:06:58 raspi bluetoothd[198384]: src/plugin.c:plugin_init() Failed to init vcp plugin
May 13 14:06:58 raspi bluetoothd[198384]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
May 13 14:06:58 raspi bluetoothd[198384]: src/plugin.c:plugin_init() Failed to init mcp plugin
May 13 14:06:58 raspi bluetoothd[198384]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
May 13 14:06:58 raspi bluetoothd[198384]: src/plugin.c:plugin_init() Failed to init bap plugin
May 13 14:06:58 raspi dbus-daemon[171238]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.ser>
May 13 14:06:58 raspi bluetoothd[198384]: Bluetooth management interface 1.22 initialized
May 13 14:06:58 raspi bluetoothd[198384]: src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
May 13 14:06:58 raspi bluetoothd[198384]: sap-server: Operation not permitted (1)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to clear UUIDs: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to add UUID: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to add UUID: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to add UUID: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to set mode: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to add UUID: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to add UUID: Failed (0x03)
May 13 14:06:58 raspi bluetoothd[198384]: Failed to add UUID: Failed (0x03)
[... cut ...]
May 14 06:07:29 raspi systemd[1]: Stopping user@1000.service - User Manager for UID 1000...
May 14 06:07:29 raspi systemd[171290]: Activating special unit exit.target...
May 14 06:07:29 raspi systemd[171290]: Stopped target default.target - Main User Target.
May 14 06:07:29 raspi org.freedesktop.secrets[171695]: discover_other_daemon: 1
May 14 06:07:29 raspi systemd-logind[171280]: Removed session 40.
...
```

I suspect it may be related with out of memory wayfire crash which happens during this boot session.

6by9, 2024-05-14:
`timed out waiting for sync completion` is indicating that the firmware isn't responding for some reason. Once gone it's unlikely to recover.
Is there anything logged from `sudo vclog -m`?

Ideally add `start_debug=1` to config.txt, reboot, and reproduce. `sudo vclog -a` should then report any asserts that have fired in the firmware which may give us a better idea of what is going on.