raspberrypi/bookworm-feedback #249: fresh install with US keyboard actually behaves like UK

dotysan, 2024-05-10:
I can reproduce this with both the downloaded imager on Windows and remote/HTTP images native on the Pi.

Raspberry Pi 5 with latest Raspberry Pi OS 64-bit 2024-03-15. Customize settings. Changed locale time zone but left the keyboard at the default "us".

After boot, open an editor and attempt to type # (Shift-3) instead you will get £.

If I use raspi-config to change they keyboard layout to the exact same "us" value, the bug goes away!

Happy to collect additional info if needed.

dotysan, 2024-05-11:
So I did a before/after and even though /etc/default/keyboard and some files in /etc/console-setup are edited, nothing appears to have changed.

However, after running raspi-config, a new /usr/share/greeter.ini appears and the bug is gone.

This symptom only occurs in the default GUI. Wayfire? If I exit to console mode the normal keyboard is loaded.

qrp73, 2024-05-12:
I suggest you to check these 3 files (in my case there are two layout in use: US and RU):

1) This one is used for keyboard layout in terminal
```sudo nano /etc/default/keyboard```
```
XKBMODEL="pc105"
XKBLAYOUT="us,ru"
XKBVARIANT=""
XKBOPTIONS="grp:alt_shift_toggle"
```

2) This one is used for keyboard layout for login screen
```sudo nano /usr/share/greeter.ini```
```
[input]
xkb_options = grp:alt_shift_toggle
xkb_model = pc105
xkb_layout = us,ru
xkb_variant =
```

3) This one is used for keyboard layout for Wayland desktop:
```sudo nano /usr/share/greeter.ini```
```
[input]
xkb_options = grp:alt_shift_toggle
xkb_model = pc105
xkb_layout = us,ru
xkb_variant =
```

lurch, 2024-05-12:
This sounds very similar to #60 , which we already fixed. Can you check whether you're using the latest version of [Raspberry Pi Imager](https://www.raspberrypi.com/software/) please? (which is v1.8.5 at the time of writing)

nguinasso, 2024-12-07:
I fixed this by changing gb to us in wayfire.ini

lurch, 2024-12-08:
Since this issue was opened we've now switched from wayfire to labwc (see https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/ for more details) so I'm going to close this issue.

closed 2024-12-08 by lurch